AllskyTeam/allsky #955: Image capture - Files are not deleted in the tmp folder

jimjackii, 2022-01-21:
Hi,

With the current file version of the file "saveImage.sh" I found a new error.

If the images are saved - at night and also during the day - the created images "image-xxx.jpg" in the tmp folder are not deleted.
The folder fills up and after a while no more images can be stored in it, e.g. if the tmp folder is a tmpfs with a limited size.

The problem is the last statement in the saveImage.sh file:

**# If it's daytime and we didn't save the image, delete it.
[ "${DAYTIME_SAVE}" = "false" -a "${DAY_OR_NIGHT}" = "DAY" ] && rm -f "${IMAGE_TO_USE}"**

This means that the files in the tmp folder are only deleted if day mode is active and the images are not to be saved.

The problem is not noticeable at first if the tmp folder is on the SD card as usual, but after a while the SD card will also be full.

Dentonknifeworks, 2022-01-21:
I also am seeing the image not being removed in the Temp dir.

EricClaeys, 2022-01-23:
@Dentonknifeworks, @jimjackii,

Would you please try the attached `scripts/saveImage.sh` file and let me know if it works better?
I tested it with DAYTIME_SAVE=true and false, and RESIZE_UPLOADS=true and false.
Other than the initial camera image save to allsky/tmp, there's only one copy to allsky/images/DATE, and if resizing uploads, one copy to a temporary file name.  This will save on wear to the SD card, ESPECIALLY if using a tmpfs for allsky/tmp (in which case there's only one write to the card per picture).
**Please rename your current saveImage.sh to saveImage.sh-OLD, then copy the new one in**, renaming it of course.

@AndreasLMeg FYI

[saveImage.sh.txt](https://github.com/thomasjacquin/allsky/files/7919587/saveImage.sh.txt)

Dentonknifeworks, 2022-01-23:
Put in place I will report back.

AndreasLMeg, 2022-01-23:
@EricClaeys now it works for me (since 6:52 no more remaining images) The older files I have to remove.
-rw-r--r--  1 pi pi  324705 Jän 23 06:46 image-20220123064510.jpg
-rw-r--r--  1 pi pi  316394 Jän 23 06:47 image-20220123064630.jpg
-rw-r--r--  1 pi pi  318861 Jän 23 06:48 image-20220123064744.jpg
-rw-r--r--  1 pi pi  314222 Jän 23 06:50 image-20220123064903.jpg
-rw-r--r--  1 pi pi  320346 Jän 23 06:51 image-20220123065017.jpg
-rw-r--r--  1 pi pi 1390716 Jän 23 07:06 image-20220123070607.jpg
-rw-r--r--  1 pi pi  373554 Jän 23 07:06 image.jpg

Dentonknifeworks, 2022-01-23:
No older images now.
Everything looks to be working now.
Did not test dark images as this is a Pi Cam only need one anyway LOL

jimjackii, 2022-01-24:
New script tested - it looks good at first. No more files in the tmp directory.
Test first in day mode. I'll report night mode tomorrow.

I don't use dark images.

jimjackii, 2022-01-25:
Hi,

the overnight test also worked -
everything's o.k.

Startrails, Keogram and Timelapse were created, no unnecessary files in the tmp directory.

Problem solved - closing this thread.

Many Thanks.

closed 2022-01-25 by jimjackii