AllskyTeam/allsky #967: New install (v0.8.3) - RPiHQ camera - images very flat

jcauthen78, 2022-01-22:
Started off with a fresh install, on a Raspberry Pi Zero 2W & RPiHQ Camera, trying a slimmed down setup to take camping with me. I've gone through the settings that logically go together, but I'm feeling I missed a step. Didn't see anything in the previous issues The images are coming out very flat in both Day & Night captures. It's not weather-ready, so I can't put it outside just yet for full testing.   Any thoughts or suggestions to fix the Flat images? Below is the info from my system and a tail of the allsky.log with the settings and output, and an example of the flat image as well. I can provide whatever other data/logs, just wasn't sure what all to include.

Y'all have done a great job with the updates & upgrades, keep doin what you do! 

I'm also noticing the Remove Bad Images is running every capture, but its failing?

Pi Zero 2W running BullsEye
Raspberry Pi HQ Camera
64gb Card. 
```
pi@allskyzero:~/allsky $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
```

GUI is installed, page file @ default _(soon to be increased to avoid timelapse kill)_. Also will alias the allsky/temp folder to reduce SD writes _(when i can find the instructions again)_

tail-f /var/log/allsky.log with Debug @ 4.  
-----
```
Jan 22 11:31:53 allskyzero allsky.sh[1693]:      ***** Stopping AllSky *****
Jan 22 11:32:04 allskyzero allsky.sh[2272]:      ***** Starting AllSky *****
Jan 22 11:32:06 allskyzero allsky.sh[2272]: CAMERA: RPiHQ
Jan 22 11:32:06 allskyzero allsky.sh[2272]: CAMERA_SETTINGS: /etc/raspap/settings_RPiHQ.json
Jan 22 11:32:21 allskyzero allsky.sh[2392]: *** WARNING: Could not set locale to en_US.UTF-8 ***
Jan 22 11:32:21 allskyzero allsky.sh[2392]:  ********************************************
Jan 22 11:32:21 allskyzero allsky.sh[2392]:  *** Allsky Camera Software v0.8.3 | 2021 ***
Jan 22 11:32:21 allskyzero allsky.sh[2392]:  ********************************************
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Capture images of the sky with a Raspberry Pi and a RPi HQ camera
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Add -h or --help for available options
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Author: Thomas Jacquin - <jacquin.thomas@gmail.com>
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Contributors:
Jan 22 11:32:21 allskyzero allsky.sh[2392]: -Knut Olav Klo
Jan 22 11:32:21 allskyzero allsky.sh[2392]: -Daniel Johnsen
Jan 22 11:32:21 allskyzero allsky.sh[2392]: -Robert Wagner
Jan 22 11:32:21 allskyzero allsky.sh[2392]: -Michael J. Kidd - <linuxkidd@gmail.com>
Jan 22 11:32:21 allskyzero allsky.sh[2392]: -Rob Musquetier
Jan 22 11:32:21 allskyzero allsky.sh[2392]: -Eric Claeys
Jan 22 11:32:21 allskyzero allsky.sh[2392]: -Andreas Lindinger
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument  5: -tty
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument  7: -alwaysshowadvanced
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument  8: 1
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument  9: -angle
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument 11: -awb
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument 13: -background
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument 15: -darkframe
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument 17: -dayautoexposure
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument 19: -dayautogain
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument 21: -daybin
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument 23: -daybrightness
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument 25: -daydelay
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument 27: -dayexposure
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument 29: -daygain
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument 31: -debuglevel
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument 33: -filename
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument 35: -flip
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument 37: -fontcolor
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument 39: -fontline
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument 41: -fontname
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument 43: -fontsize
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument 45: -fonttype
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument 47: -height
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument 49: -latitude
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument 51: -locale
Jan 22 11:32:21 allskyzero allsky.sh[2392]: Processing argument 53: -longitude
[... 92 lines cut ...]
Jan 22 11:32:38 allskyzero allsky.sh[2392]:   > Focus: 4.544234
Jan 22 11:32:41 allskyzero allsky.sh[2392]:   > Saving DAY image 'image-20220122113221.jpg'
Jan 22 11:32:42 allskyzero allsky.sh[2392]: Sleeping 30.0 seconds...
Jan 22 11:32:42 allskyzero allsky.sh[2417]: DEBUG: saveImage.sh DAY /home/pi/allsky/tmp/image-20220122113221.jpg EXPOSURE_US=32000 BRIGHTNESS=70 MEAN=-1.000 AUTOEXPOSURE=1 AUTOGAIN=1 AUTOWB=0 WBR=2.60 WBB=2.10 GAIN=5.00 GAINDB=140 BIN=1 FLIP=2 BIT_DEPTH=08 FOCUS= 5
Jan 22 11:32:46 allskyzero allsky.sh[2437]: /home/pi/allsky/scripts/removeBadImages.sh: line 126: bc: command not found
Jan 22 11:32:46 allskyzero allsky.sh[2440]: /home/pi/allsky/scripts/removeBadImages.sh: line 130: bc: command not found
Jan 22 11:32:46 allskyzero allsky.sh[2424]: /home/pi/allsky/scripts/removeBadImages.sh: line 131: [: -eq: unary operator expected
Jan 22 11:32:46 allskyzero allsky.sh[2443]: /home/pi/allsky/scripts/removeBadImages.sh: line 140: bc: command not found
Jan 22 11:32:46 allskyzero allsky.sh[2424]: /home/pi/allsky/scripts/removeBadImages.sh: line 141: [: -eq: unary operator expected
Jan 22 11:32:46 allskyzero allsky.sh[2417]: saveImage.sh: Resizing '/home/pi/allsky/tmp/image-20220122113221.jpg' to 1640x1232
Jan 22 11:33:12 allskyzero allsky.sh[2392]: STARTING EXPOSURE at: 2022-01-22 11:33:12   @ 32.00 ms (0.03 sec)
Jan 22 11:33:12 allskyzero allsky.sh[2392]:  > Kill command: pgrep 'libcamera-still' | xargs kill -9 2> /dev/null
Jan 22 11:33:12 allskyzero allsky.sh[2392]:   > Capture command: LIBCAMERA_LOG_LEVELS='ERROR,FATAL' libcamera-still --output '/home/pi/allsky/tmp/image-20220122113312.jpg' --timeout 2000 --nopreview --width 4056 --height 3040 --shutter 0 --gain 1 --awbgains 2.6,2.1 --vflip --saturation 2 --brightness 0.7 --quality 70 2> /dev/null
Jan 22 11:33:24 allskyzero allsky.sh[2392]:   > Focus: 4.291109
Jan 22 11:33:26 allskyzero allsky.sh[2392]:   > Saving DAY image 'image-20220122113312.jpg'
Jan 22 11:33:27 allskyzero allsky.sh[2392]: Sleeping 30.0 seconds...
Jan 22 11:33:27 allskyzero allsky.sh[2480]: DEBUG: saveImage.sh DAY /home/pi/allsky/tmp/image-20220122113312.jpg EXPOSURE_US=32000 BRIGHTNESS=70 MEAN=-1.000 AUTOEXPOSURE=1 AUTOGAIN=1 AUTOWB=0 WBR=2.60 WBB=2.10 GAIN=5.00 GAINDB=140 BIN=1 FLIP=2 BIT_DEPTH=08 FOCUS= 4
Jan 22 11:33:31 allskyzero allsky.sh[2499]: /home/pi/allsky/scripts/removeBadImages.sh: line 126: bc: command not found
Jan 22 11:33:31 allskyzero allsky.sh[2502]: /home/pi/allsky/scripts/removeBadImages.sh: line 130: bc: command not found
Jan 22 11:33:31 allskyzero allsky.sh[2486]: /home/pi/allsky/scripts/removeBadImages.sh: line 131: [: -eq: unary operator expected
Jan 22 11:33:31 allskyzero allsky.sh[2505]: /home/pi/allsky/scripts/removeBadImages.sh: line 140: bc: command not found
Jan 22 11:33:31 allskyzero allsky.sh[2486]: /home/pi/allsky/scripts/removeBadImages.sh: line 141: [: -eq: unary operator expected
Jan 22 11:33:31 allskyzero allsky.sh[2480]: saveImage.sh: Resizing '/home/pi/allsky/tmp/image-20220122113312.jpg' to 1640x1232
```
[image]

AndreasLMeg, 2022-01-22:
@jcauthen78 
hopefully the saturation value is the reason. Bullseye uses a different programm to capture the images. some values you have to define different to buster

You are using "Saturation: 60.0" but for bullseye there is a limit (0...2) - please change it to 1.0
Here you can see the command for the capture: (max limit is used) 
Capture command: LIBCAMERA_LOG_LEVELS='ERROR,FATAL' libcamera-still --output '/home/pi/allsky/tmp/image-20220122113312.jpg' --timeout 2000 --nopreview --width 4056 --height 3040 --shutter 0 --gain 1 --awbgains 2.6,2.1 --vflip --saturation 2 --brightness 0.7

brightness - a little bit the same - please use 0.0
(User enters -100 to 100.  Convert to -1.0 to 1.0.)

[image]

AndreasLMeg, 2022-01-22:
we will normalize the values ​​in the future, so that there is no dependency on cameras and capture programs

AndreasLMeg, 2022-01-22:
about removeBadImage: maybe this can help: https://unix.stackexchange.com/questions/623672/i-keep-receiving-command-not-found-when-i-try-to-run-my-code-in-bash

at the prompt in putty, type `which bc`
it should respond with` /usr/bin/bc`

jcauthen78, 2022-01-22:
BOOM!  you were 100% correct - taking them down to the new lower ranges worked beautifully, both Brightness & Saturation needed changed up to get the image happy.

I ran the `which bc` and it came back completely blank. 
`pi@allskyzero:~ $ which bc`
`pi@allskyzero:~ $ which bc`
`pi@allskyzero:~ $ ` 

I'll dig through StackExchange a little more and see what I can find out, see if maybe it's a PiZero specific or a BullsEye thing.

I can close this out as resolved for the Flat image issue unless ya want some resolution to the bash failing thing

[image]

jcauthen78, 2022-01-23:
> about removeBadImage: maybe this can help: https://unix.stackexchange.com/questions/623672/i-keep-receiving-command-not-found-when-i-try-to-run-my-code-in-bash
> 
> at the prompt in putty, type `which bc` it should respond with` /usr/bin/bc`

What is 'bc' ? maybe I can do a manual install of whatever it is? Line 126 : `MEAN=$(echo "${MEAN} * 100" | bc)`. It's being piped out to it, but I'm not sure what it is that BC does.

EricClaeys, 2022-01-23:
@jcauthen78   Long time no talk.  How you're doing well.
`bc` is a Linux calculator - I think it stands for Binary Calculator.  Anyhow, `bash` can do arithmetic with integers but not floating point numbers (like the mean brightness value of an image) so we have to use `bc` for calculations.  And it's not installed by default on Bullseye hence the "command not found" message.  All the other errors are because the `bc` command didn't work.

In 0.8.3 removeBadImages.sh is called on every image because it only takes a couple seconds and catches corrupted images (not common) and too bright or too dark images (common if using fixed exposure times around day/night transitions).

@linuxkidd - Someone else had this issue on Bullseye where `bc` wasn't installed.  What was the solution?  Are we (you!) going to change the installation script to install it if needed?

@AndreasLMeg FYI

linuxkidd, 2022-01-23:
Strange... `bc` is already in the [deps list](https://github.com/thomasjacquin/allsky/blob/master/src/Makefile#L95)
```
deps:
	@echo `date +%F\ %R:%S` Installing build dependencies...
	@apt update && apt -y install libopencv-dev libusb-dev libusb-1.0-0-dev ffmpeg gawk lftp jq imagemagick bc
```

Appears it was added by PR #902 2 weeks back.  :)   Be sure you're up to date on the latest code and run `install.sh`.

jcauthen78, 2022-01-23:
For giggles, did a Dist Upgrade & Firmware on the Zero W2, renamed the original allsky folder, re-git & install, reinstall of Gui too.   BC is recognized.  It looks like I did the first install a little more than 2'ish weeks ago, so missed that #902 PR. 
`pi@allskyzero:~ $ which bc`
`/usr/bin/bc`
`pi@allskyzero:~ $ `

This quick fix, and the previous Brightness & Saturation fix - y'all got me fixed up.  Thank you. Next on my list is the swap file, as well as the virtual temp folder to reduce card writes. 

@EricClaeys good to see ya still cranking hard on the updates, keep up the hard work.  Any luck with the day/night transition, or did yall have to scrap the idea?

linuxkidd, 2022-01-23:
@jcauthen78 Great to hear it's all fixed up for ya!

If you run into issues with the swap setup, etc.. please don't hesitate to reach out via a new issue.  I'll close this one since it's all wrapped up.

Clear skies!  o/

closed 2022-01-23 by linuxkidd

EricClaeys, 2022-01-24:
@jcauthen78 Jim, I improved what I could with the transition but there is only so much that's possible as long as we don't use our algorithm at night.
However, @AndreasLMeg has a nice auto exposure algorithm for the PRiHQ camera that works at night.  After he's done tweaking it we'll port it to ZWO.  Can't wait - the ZWO algorithm produces pulses when the moon is out, like last night.