AllskyTeam/allsky #968: Timelapse not generated

CleverJay, 2022-01-22:
Configuration is:
Rasp Pi 3B+, OS Bullseye, Allsky downloaded and installed 5 days ago, updated saveImage.sh yesterday. Increased swapfile size to 1024 per the Wiki. also set the Time-lapse width and height  to 2048X1534.
Allsky is generating a keogram and a startrail image, the images are saving into the images folder. But when I run either generateForDay.sh or time-lapse.sh from the terminal I get the message as seen in the attached term.txt file. I have run the scripts  both with the warning and info settings for the log file. I've attached the latest timelapseTMP.txt file. there only seems to be the  "deprecated pixel format used, make sure you did set range correctly" error massage that the wiki states should be ignored.
Thanks for looking into this
[term.txt](https://github.com/thomasjacquin/allsky/files/7919132/term.txt)
[timelapseTMP.txt](https://github.com/thomasjacquin/allsky/files/7919133/timelapseTMP.txt)
.

AndreasLMeg, 2022-01-22:
please look at /var/log/messages - maybe you can see a message like "OOM-Killer" Out of Memory Killer
Unfortunately, we cannot foresee how much memory is still available, which processes are still running, which CODEC is set,...
In any case, ffmpeg tries to get everything to the end of memory and then just gets killed (OOM)

Please try reducing the image size
But be careful, all images must have the same resolution one day (please delete old images)

# Resize images before cropping, stretching, and saving.
# Adjust IMG_WIDTH and IMG_HEIGHT according to your camera's sensor ratio.
IMG_RESIZE="true"
IMG_WIDTH=2028
IMG_HEIGHT=1520

CleverJay, 2022-01-22:
@AndreasLMeg  Thanks so much for your prompt response. I am indeed getting the OOM-killer error in var/log/messages. 
When I checked memory in the terminal, It showed zero swap memory. I edited the swapfile to 1024. and when I tried to turn swap back on, I got : 

> swapon: /var/swap: insecure permissions 0644, 0600 suggested.
> swapon: /var/swap: read swap header failed

see the attached terminal transcript
[swap memory error 2.txt](https://github.com/thomasjacquin/allsky/files/7919356/swap.memory.error.2.txt)

I could not change the permissions on the swapfile either. Any ideas on getting it recognize the swapfile?

EricClaeys, 2022-01-23:
@CleverJay The "Killed" message is almost always do to not enough RAM+swap.  The Wiki says to start with 1024 MB.  It looks like you need more (or when you tried to increase it, it didn't work).
Do the following to check how much swap you have:
~~~
grep CONF_SWAPSIZE /etc/dphys-swapfile
~~~
Also, did you follow [these instructions](https://github.com/thomasjacquin/allsky/wiki/Increase-swap-space) exactly?  It's possible the swap file isn't enabled, and that's why you're seeing a size of 0.

My /var/swap file  is mode 600, owned and grouped to root.  You should be able to:
~~~
sudo chmod 600 /var/swap
~~~
But that's just a security issue and isn't impacting your problem.

CleverJay, 2022-01-23:
SOLVED!
@EricClaeys  I had to do a couple additional steps than was shown in the Swapfile tutorial instructions. 
in addition to the chmod, I had to do:
` sudo mkswap
sudo swapon`
I also did the image resize that @AndreasLMeg suggested.
So now my swapfile is 1024 and the time-lapse for last night was created.
We may want to note the additional steps in the Wiki and note in the FAQ, the minimal RAM requirements for the RPiHQ camera.
Thanks!

EricClaeys, 2022-01-24:
@CleverJay Glad it's working now.  Would you please close the issue?

Where in the sequence of steps [here](https://github.com/thomasjacquin/allsky/wiki/Increase-swap-space) did you do the two extra commands?

Did you not have a swap file when you started?

Doesn't the `mkswap` command take an argument?

CleverJay, 2022-01-24:
@EricClaeys I ended up using this tutorial: [https://itsfoss.com/create-swap-file-linux/](url) that seemed to work better for my use case.

closed 2022-01-24 by CleverJay