AllskyTeam/allsky #969: Check for size of RAM+swap during installation

EricClaeys, 2022-01-23:
A LOT of people have problems due to not enough RAM+swap.
We should use check how much total RAM+swap the user has during installation, and either warn if not "much" (to be determined what this means, but probably at least 2 GB), or ask the user if they want to increase swap, and prompt for the amount, the increase for them.

We could use `free -m` looking in the second columns of the first and second lines (memory and swap), and total them, basing our decision as what to do on the total.

ShedAstro, 2022-02-04:
Trust me I use 4gb of swap memory and it makes no difference whatsoever ever to whether the camera creates the Timelapse and star trail images or not, some days it works some days it does not, I just don’t think that the PI Zero is up to the job at all, and would like to here from other Zero users…☹️

EricClaeys, 2022-02-05:
@ShedAstro The Allsky Facebook group would be a better place to get input from other Pi Zero users.  Not many people look at Git Hub, and when they do it's usually to report a problem.

AmiSMB, 2022-06-24:
I changed from a Pi Zero to a Pi Zero 2 for the same reason. It now works reasonably. If you configure hardware encoding it works even better!

EricClaeys, 2022-06-29:
Closing - this request was added to the Projects list - see the `Projects` link at the top of any Allsky GitHub page.
Project Title: **Check for size of RAM+swap during installation #969**

closed 2022-06-29 by EricClaeys