AllskyTeam/allsky #970: compile error,

ON2ON-HAM, 2022-01-23:
Hi,

I tried an install both on Buster and Bulldog, using a Raspi Zero, and HQ Camera.
After installing, all is working, camera, capture etc.  but ..
During installation following compile errors show up , related to Startrail, etc... :+1: 
[image]

Thanks for taking a look at this, if possible.

Luke 
ON2ON

AndreasLMeg, 2022-01-23:
@ON2ON-HAM no error, just warning. sunwait is external. Sorry, we can't change this.

For startrails, we have to look....

ON2ON-HAM, 2022-01-23:
Ok, Thanks for fast reply !
73
Luke

AndreasLMeg, 2022-01-23:
@ON2ON-HAM 

Thanks for the finding.
https://github.com/thomasjacquin/allsky/pull/971

Andreas

linuxkidd, 2022-01-23:
Thanks @ON2ON-HAM for finding and reporting the warnings... Thanks @AndreasLMeg for the PR to fix the startrails unused variables! :)   All merged now.

Note, @ON2ON-HAM  -- you should be able to use the code as compiled without issue, even though it complained about the unused variables.

Clear skies! o/

closed 2022-01-23 by linuxkidd