AllskyTeam/allsky #974: Timelapse and startrails stopped

uzelessknowledge, 2022-01-27:
I’ve had my build running on the roof for 2 years. I noticed the last Timelapse that built was 1/1 and the last startrails was 1/18. 
I tried to manually build a Timelapse and get this error. I saw similar errors in the logs. 
[image]

EricClaeys, 2022-01-27:
@uzelessknowledge It looks like you have an older version of Allsky.  I'm pretty sure we fixed the Illegal instruction bug a while ago.

uzelessknowledge, 2022-01-27:
Let me find some time to see how to update. 
strange it’s been running fine this long.

linuxkidd, 2022-01-28:
HI @uzelessknowledge ... 

Just picking up from the facebook comments.
- Did you upgrade the allsky software since the above posting?
- Can you provide the output of:
```
$ cat /etc/os-release
```

It's likely just a few commands needed to fully upgrade the OS and Allsky, so no expectation to need to re-image the sd card or anything like that.

uzelessknowledge, 2022-01-29:
I was going to follow the update wiki as I see I don't have to do an SD card image.
https://github.com/thomasjacquin/allsky/wiki/How-to-update-the-software
[image]

uzelessknowledge, 2022-01-29:
So I’ve done the update but need to get into the pi so I can backup and restore some files. Problem is I can’t SFTP now. Connection closes right after authentication. Ugh. 
Really don’t want to reimage.

EricClaeys, 2022-01-29:
@uzelessknowledge Is the only thing you updated the Allsky software?  It doesn't touch the SFTP configuration as far as I know.  In fact,  it doesn't do anything to the network.
Can you get to the WebUI or the website if you have it installed?

@linuxkidd Michael, any ideas?

linuxkidd, 2022-01-29:
I agree.. the upgrade doesn't run anything that would mess with non-related packages.

@uzelessknowledge Can you still `ssh`, and just not `sftp` ?  If so, you may try just issuing a `reboot` and try again once the Pi comes back up.  Otherwise, if you cannot `ssh` at all, I'd power cycle it, and retry your `sftp` connection after that.

uzelessknowledge, 2022-01-29:
Oh sorry. Not correlating the 2 together. Just struggling since I can’t SFTP to move files. 

Yes, I can ssh with putty but not SFTP. I had issues reboot after the updates. I’ll try a full power cycle.

EricClaeys, 2022-01-29:
@uzelessknowledge sorry to hear you are having issues logging in, but am glad they are not Allsky related.

Let us know the stus of the original postem item after you get loggrd in, and if things work, please close the Issue.
Thanks

uzelessknowledge, 2022-01-29:
Yes sir. I need to get all my settings back so I can do that.

linuxkidd, 2022-01-29:
@uzelessknowledge -- if you're still having SFTP issues, you can use `rsync` which runs directly over SSH ( vs SFTP )... 
- For Mac OS, it should be built in, and accessible from the terminal.
- For Windows, you'd need to enable the 'Windows Subsystem for Linux', then reboot, then install a Linux distro from the Microsoft Store ( Debian works well for me ) -- THEN, you can use `rsync` after installing it.

A sample rsync command:
`rsync -rP pi@192.168.1.5:/home/pi/allsky/  ./`

This would transfer the `allsky` directory with all sub-directories, to the current directory on your local computer, assuming `192.168.1.5` is the IP of the Pi.

( you can reverse the order of the paths to copy from local to the remote IP ).  Do keep us updated on your progress.

EricClaeys, 2022-01-29:
@uzelessknowledge One of our "to do" items is to add a "backup configuration" button in the WebUI to make it quicker to reinstall Allsky on newly imaged Pi's.

uzelessknowledge, 2022-01-30:
Well, I got it updated, but I can't get the FTP to my website to work now. I'm not understanding how the new configs are used vs the old ones.

uzelessknowledge, 2022-01-30:
When I try to manually run the upload script I see an error. It's trying to give the file a temp name. 

pi@allsky:~ $ bash allsky/scripts/upload.sh /home/pi/allsky/images/20220129/image-20220129190809.jpg /allsky/ image.jpg
upload.sh: FTP'ing /home/pi/allsky/images/20220129/image-20220129190809.jpg to /allsky/image.jpg, TEMP_NAME=x-27037
*** upload.sh: ERROR:
FILE_TO_UPLOAD='/home/pi/allsky/images/20220129/image-20220129190809.jpg'
REMOTE_HOST='<redacted>'
REMOTE_DIR='/allsky/'
TEMP_NAME='x-27037'
DESTINATION_FILE='image.jpg'

put: /home/pi/allsky/images/20220129/image-20220129190809.jpg: Access failed: 553 Can't open that file: No such file or directory (/allsky/x-27037)
put of /home/pi/allsky/images/20220129/image-20220129190809.jpg failed!

linuxkidd, 2022-01-30:
@uzelessknowledge 

If you manually FTP to your hosting site, can you confirm that the `allsky` directory is off the top level of the visible filesystem?
- This may well be correct, assuming your hosting provider locks you into your home directory.

However, it's possible that it should be more like `/home/myuser/www/allsky`  or similar.

Hope this helps!

EricClaeys, 2022-01-30:
@uzelessknowledge Try the steps at the [troubleshooting uploads Wiki page](https://github.com/thomasjacquin/allsky/wiki/Troubleshooting:-uploads).
The key is what's in the lftp_cmds.txt file and what happens when you do an lftp manually based on the contents of the file.

uzelessknowledge, 2022-01-31:
I reimages the SD. Using the current build for Allsky and the website I rebuilt my entire setup. Images are being captured, startrails and timelapse are building. Closing issue. 
Thank you all for your help. I still have some other kinks to work out, but this issue is considered resolved.

closed 2022-01-31 by uzelessknowledge

EricClaeys, 2022-02-01:
@uzelessknowledge Glad you got things working, but I hate seeing people having to resort to wiping their SD card.

uzelessknowledge, 2022-02-01:
> @uzelessknowledge Glad you got things working, but I hate seeing people having to resort to wiping their SD card.

1. There were too many changes between allsky and website where I was out of sync
2. With my pi refusing ftp connections and I couldn’t get it working.