AllskyTeam/allsky #978: View image with rpiHQ

Kilis71, 2022-01-28:
I have a problem, I have installed the latest version of allskycam with rpiHQ camera, but the image is not displayed it tells me software is starting up

AndreasLMeg, 2022-01-28:
@Kilis71 Please see this Wiki page and attach your log file.
Looks like the Allsky system is having trouble starting up.
Unfortunately, there are many reasons for this and without logs we cannot help.

Kilis71, 2022-01-28:
I solved it by setting REMOVE_BAD_IMAGES to False
thanks anyway

AndreasLMeg, 2022-01-29:
Please close this issue when your problem is solved or you have now fixed the problem yourself.

EricClaeys, 2022-01-29:
@Kilis71 Having to disable REMOVE_BAD_IMAGES usually means you are using manual exposure at night, and at the day/night transitions the pictures are initially too bright until it gets darker.  The too bright images are discarded which is why you see the "starting up" message.

However, there are other reasons you can see the message.  Would you please re-enable the option then follow the instructions at the "reporting problems" Wiki page?  I would like to make sure it's not a bug.
Thanks

closed 2022-02-13 by Kilis71