AllskyTeam/allsky #979: Keogram black line near labels

loloderu, 2022-01-29:
Found a but appearing on some keogram having black horizontale line in the foreground of labels

replace
               cv::putText(*acc, text,
                    cv::Point(ann->at<int>(r, 0) - textSize.width, 
                    acc->rows - (textSize.height)),
                    cf->fontFace, cf->fontScale,
                    cv::Scalar(0, 0, 0), cf->lineWidth+2,
                    cf->fontType);

with

               cv::putText(*acc, text,
                    cv::Point(ann->at<int>(r, 0) - textSize.width-1, 
                    acc->rows - (textSize.height)-1),
                    cf->fontFace, cf->fontScale,
                    cv::Scalar(0, 0, 0), cf->lineWidth,
                    cf->fontType);

AndreasLMeg, 2022-01-29:
@loloderu Please send us images (before and after your change)

loloderu, 2022-01-29:
before
[image]
after
[image]

EricClaeys, 2022-01-29:
@loloderu Thanks for finding this bug.  Can you explain why the lines appeared on some images but not most of them?  Is this the "off by one" bug?

@AndreasLMeg Someyimes these lines appear through images with text.  It may be the same bug.

loloderu, 2022-01-29:
I was not able to explain clearly the origins of the bug but it appens only if you keep an extra width for the black text (cf->lineWidth+2)

AndreasLMeg, 2022-01-30:
@loloderu Looks very interesting!

I have never seen such behavior. The cause is still unknown to me. The only cause of horizontal lines up to now has only been image elements in the original image (exactly in the middle of the image) which were then disruptive in the keogram. (raindrops, lampposts, text, ...)

**Absolutely necessary for further investigations: Can you provide me with the original images? (possibly a partial amount is sufficient)**

If you disable the text completely, does this problem still occur?
Are you using the current version of keogram, or have you already made other changes?

thank you

AndreasLMeg, 2022-01-30:
> @loloderu Thanks for finding this bug. Can you explain why the lines appeared on some images but not most of them? Is this the "off by one" bug?
> 
> @AndreasLMeg Someyimes these lines appear through images with text. It may be the same bug.
I have found this issue (https://github.com/thomasjacquin/allsky/issues/810) Yes, I think this is the same reason. (maybe openCV ?, which version ?, which cpu ?, which OS ?,...)

loloderu, 2022-01-30:
I am using the very last version. There was no problem with previous version because there was no black text behind the normal text I believe.
If label are diseable ther is no black line: this only when you execute the black texte with width+2 that lines are apearing

loloderu, 2022-01-30:
This raspberry pi 4 with Raspbian GNU/Linux 10 (buster) OS. Opencv 3.2.0

loloderu, 2022-01-30:
@AndreasLMeg : what  original images do you want? do you mean the keogram without the labels?

AndreasLMeg, 2022-01-30:
> @AndreasLMeg : what original images do you want? do you mean the keogram without the labels?
the images directory from this day would be nice. (e.g. ~/allsky/images/20220129)

AndreasLMeg, 2022-01-30:
> This raspberry pi 4 with Raspbian GNU/Linux 10 (buster) OS. Opencv 3.2.0

I have found this issue: https://github.com/opencv/opencv_contrib/issues/2959
But I have to understand that....

AndreasLMeg, 2022-01-30:
My system tells me cv2 **Version 4.5.3** is in use
```
pi@allsky:~/allsky/images/20220129 $ python3
Python 3.7.3 (default, Jan 22 2021, 20:04:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv2
>>> cv2.__version__
'4.5.3'
```

AndreasLMeg, 2022-01-30:
maybe one solution (opencv/opencv#16301): When using a higher thickness (e.g., 5) the problem goes away.

@loloderu could you test this, please ?

loloderu, 2022-01-30:
for me
Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv2
>>> cv2.__version__
'4.5.4'
>>> 
pkg-config --modversion opencv
3.2.0

pi@Rapi42Scope:~/allsky $ apt list "*opencv*"
En train de lister... Fait
cl-opencv-apps/stable 1.12.0-2 all
gstreamer1.0-opencv/stable 1.14.4-1+rvt+deb10u1 armhf
libgstreamer-opencv1.0-0/stable 1.14.4-1+rvt+deb10u1 armhf
libopencv-apps-dev/stable 1.12.0-2 armhf
libopencv-apps1d/stable 1.12.0-2 armhf
libopencv-calib3d-dev/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-calib3d3.2/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-contrib-dev/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-contrib3.2/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-core-dev/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-core3.2/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-dev/stable,now 3.2.0+dfsg-6 armhf  [installé]
libopencv-features2d-dev/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-features2d3.2/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-flann-dev/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-flann3.2/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-highgui-dev/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-highgui3.2/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-imgcodecs-dev/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-imgcodecs3.2/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-imgproc-dev/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-imgproc3.2/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-ml-dev/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-ml3.2/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-objdetect-dev/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-objdetect3.2/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-photo-dev/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-photo3.2/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-shape-dev/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-shape3.2/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-stitching-dev/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-stitching3.2/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-superres-dev/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-superres3.2/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-ts-dev/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-video-dev/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-video3.2/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-videoio-dev/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-videoio3.2/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-videostab-dev/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-videostab3.2/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-viz-dev/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv-viz3.2/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
libopencv3.2-java/stable,now 3.2.0+dfsg-6 all  [installé, automatique]
libopencv3.2-jni/stable,now 3.2.0+dfsg-6 armhf  [installé, automatique]
node-opencv/stable 6.0.0+git20180416.cfc96ba0-3 armhf
opencv-data/stable,now 3.2.0+dfsg-6 all  [installé, automatique]
opencv-doc/stable 3.2.0+dfsg-6 all
python-opencv-apps/stable 1.12.0-2 all
python-opencv/stable 3.2.0+dfsg-6 armhf
python3-opencv-apps/stable 1.12.0-2 all
python3-opencv/stable 3.2.0+dfsg-6 armhf
ros-opencv-apps/stable 1.12.0-2 armhf

loloderu, 2022-01-30:
For information the keogram option I used for both images are:
-font-color ffffff --font-size 0.5 --font-line 0.5

loloderu, 2022-01-30:
> > @AndreasLMeg : what original images do you want? do you mean the keogram without the labels?
> 
> the images directory from this day would be nice. (e.g. ~/allsky/images/20220129)
This is about 300MB, how can I send them to you?

loloderu, 2022-01-30:
The image with

EricClaeys, 2022-01-31:
@loloderu Do you get the lines if you only use about 10 images?

You can test by creating allsky/images/test and copying 10 images from yesterday to the `test` directory. Then run `scripts/generateForDay.sh -k test`

EricClaeys, 2022-01-31:
If you get the same error with 10 images that would be easy to upload.

AndreasLMeg, 2022-01-31:
@loloderu: 

I think the problem is "--font-line 0.5" the openCV function only supports "int" values (initial value=1). 
I have to check the keogram parameter -> maybe 0.5 will change to 0

work in progress....

AndreasLMeg, 2022-01-31:
@EricClaeys @loloderu @linuxkidd 
I'm not sure if restricting font-line (min=1) and font-size(min=1) will solve this problem for all users.
Or do we leave it as it is and just write a wiki entry?
How would you decide?

loloderu, 2022-01-31:
@AndreasLMeg  --font-size 0.5 --font-line 1" is ok too ! => --font-line 0.5 is 0 in reality as only integer are permited 
[image]

loloderu, 2022-01-31:
@AndreasLMeg : finally I would say that adding a comment in keogram help saying font-line has to be an integer >=1 would solve the problem for now :)

EricClaeys, 2022-02-01:
@loloderu, @AndreasLMeg, @linuxkidd 
I think we should fix keogram.cpp to only accept an integer for --font-line.  IMHO it's a bug to accept anything else.  When I started working on Allsky last year I was surprised how little error checking was done anywhere.  Every time I hit one of those conditions it took forever to determine what was wrong, whereas an error check and message would have made it easy to fix.

Will that fix the issue or do we need to upgrade Allsky to use opencv 3.2?

loloderu, 2022-02-02:
For my part I use opencv 3.2

AndreasLMeg, 2022-02-02:
> Will that fix the issue or do we need to upgrade Allsky to use opencv 3.2?

I think opencv 3.2 is included in Raspi debian buster. And to install a "better" version would bring other problems (bugs, memory usage, diskspace usage,....)

EricClaeys, 2022-02-11:
@loloderu, @AndreasLMeg, @linuxkidd
Where are we with this issue?  Did the change Andreas put in fix the Issue?  If so, @loloderu would you please close this?
Thanks

loloderu, 2022-02-11:
I confirm the bug was due to the use of a floating point value less than 1 causing font-line to be szt to 0. Thanks to Andreas and to you for solving this!

closed 2022-02-11 by loloderu