AllskyTeam/allsky #986: sFTP Gandi ?

Frankastro85, 2022-01-31:
Hello,
Does someone use the host Gandi ?
I can't upload my files of my my Allsky : http://www.villageduciel.fr/allsky/

My parameters of my ftp-settings.sh :

*****************************************************************************
#!/bin/bash

# FTP/SFTP settings
PROTOCOL='fstp'
USER='******'
PASSWORD='********'
HOST='sftp.sd6.gpaas.net'
IMGDIR='/allsky/'
MP4DIR='/allsky/videos/'
KEOGRAM_DIR='/allsky/keograms/'
STARTRAILS_DIR='/allsky/startrails/'
***************************************************************************

I can access to this page with Filezilla :
[image]
[image]

An idea ?
Thank you

EricClaeys, 2022-02-01:
@Frankastro85 Did you look at the Wiki page for troubleshooting uploads?

I never heard of a PROTOCOL of "fstp".  I'm guessing it's a typo.

linuxkidd, 2022-02-02:
Also, note that your path shows '/allsky/', but the full path is likely required:
```
/lamp0/web/vhosts/www.villageduciel.fr/htdocs/allsky
```

The other paths will likely need to be adjusted as well.

Frankastro85, 2022-02-02:
Hello @linuxkidd,
Thank you, I'll try that asap.
Frank

Frankastro85, 2022-02-02:
@EricClaeys Hello, Yes, I saw it and I have no problem for my site ftp. But I need to change and my new site use sftp.

Frankastro85, 2022-02-06:
Hello,
Please help me to understand what I do wrong. I can't upload my pictures/videos etc. to my website on Gandi.

With Filezilla, I have this address :
sftp://71422@sftp.sd6.gpaas.net/lamp0/web/vhosts/www.villageduciel.fr/htdocs/allsky/index.html

So, my settings ftp-settings are :

IMAGE_DIR=""

VIDEOS_DIR="/lamp0/web/vhosts/www.villageduciel.fr/htdocs/allsky/videos"
VIDEOS_DESTINATION_NAME=""
WEB_VIDEOS_DIR=""

KEOGRAM_DIR="/lamp0/web/vhosts/www.villageduciel.fr/htdocs/allsky/keograms"
KEOGRAM_DESTINATION_NAME=""
WEB_KEOGRAM_DIR=""

STARTRAILS_DIR="/lamp0/web/vhosts/www.villageduciel.fr/htdocs/allsky/startrails"
STARTRAILS_DESTINATION_NAME=""
WEB_STARTRAILS_DIR=""

IMGDIR="${IMAGE_DIR}"
MP4DIR="${VIDEOS_DIR}"

REMOTE_HOST="sftp.sd6.gpaas.net"

REMOTE_USER="*********"

REMOTE_PASSWORD="*********"

A screenshot of Filezilla (I can access to the address) :

[image]

I don't understand my problem.
Thank you.

EricClaeys, 2022-02-07:
@Frankastro85 I would have expected IMAGE_DIR="/lamp0/web/vhosts/www.villageduciel.fr/htdocs/allsky"
But that doesn't explain why you can not upload videos.  Look at the Wiki page on troubleshooting uploads.

Frankastro85, 2022-02-07:
Ok, thank you !

EricClaeys, 2022-02-11:
@Frankastro85 Any update?  Almost everyone who has followed the instructions on the Wiki page have gotten uploads to work.

Frankastro85, 2022-02-11:
Hello @EricClaeys ,
I'm probably stupid but no, I can't access to my website...
So I've the possibility to have a second website. I've tried with it and I've the same problem.
If I send you (MP) my password, can you try for me ?

EricClaeys, 2022-02-11:
@Frankastro85 Sure, be happy to help.  Send it to AstroEric _at_ eccssw _dot_ com.
If your Pi is accessible via the Internet and you are comfortable, you can send me login info and I can see if I can get it working.  I did this with someone else and it worked well.
Eric

EricClaeys, 2022-02-11:
@Frankastro85 You may be inexperienced with Linux but that doesn't mean you're stupid.  If you can't get something to work after following the documentation it could indicate that the documentation needs clarification.

Frankastro85, 2022-02-11:
Thank you :-)
I've sent you a message MP

Frankastro85, 2022-02-11:
I would like to try this command to understand...

'__Try this from the command line and see what messages it produces (replace IMAGE_DIR with what's in your ftp-settings.sh file):
echo hi > /tmp/test.txt
scripts/upload.sh /tmp/test.txt IMAGE_DIR__'

So, I dit that :
cmd Allsky
echo hi > /tmp/test.txt
scripts/upload.sh /tmp/test.txt "/lamp0/web/vhosts/www.villageduciel.fr/htdocs/allsky"
But of course, it's not correct...
I've this message :

pi@allsky:~/allsky $ scripts/upload.sh /tmp/test.txt "/lamp0/web/vhosts/www.villageduciel.fr/htdocs/allsky"
*** Usage: upload.sh [--silent]  file_to_upload  directory  destination_file_name [unique_name] [local_directory]
Where:
   '--silent' doesn't display any status messages
   'file_to_upload' is the path name of the file you want to upload.
   'directory' is the directory ON THE SERVER the file should be uploaded to.
   'destination_file_name' is the name the file should be called ON THE SERVER.
   'unique_name' is an optional, temporary name to use when uploading the file.
   'local_directory' is the name of an optional local directory the file should be
        copied to in addition to being uploaded.

EricClaeys, 2022-02-11:
@Frankastro85 My bad.  I just fixed the Wiki page.  The "upload.sh" script takes 3 arguments, not 2, so you should execute:
~~~
scripts/upload.sh /tmp/test.txt "/lamp0/web/vhosts/www.villageduciel.fr/htdocs/allsky" test.txt
~~~
The last argument ("text.txt") tells it what name the remote file should be called.

Sorry about that.

Frankastro85, 2022-02-11:
Ok, thank you !

Frankastro85, 2022-02-12:
@EricClaeys Thank you so much !!!
[](http://www.villageduciel.fr/allsky/)

EricClaeys, 2022-02-12:
Closing - issue resolved per user.

closed 2022-02-12 by EricClaeys