AllskyTeam/allsky #989: allsky website cannot handle PNG images

gilesco, 2022-02-01:
Allsky website doesn't handle PNG image output from the allsky application.

It is hardcoded to expect mp4 for timelapse and jpg for anything else.

linuxkidd, 2022-02-02:
Hi @gilesco, Thanks for finding this and reporting it.  Would you mind opening an issue reporting this under the [allsky-website](https://github.com/thomasjacquin/allsky-website/issues/new/choose) repo?

The way the software installs, it's not easy to know that the website is under a separate repo, but it'd be great to track this there.

linuxkidd, 2022-02-02:
btw, I already submitted a [PR](https://github.com/thomasjacquin/allsky-website/pull/85) for [allsky-website](https://github.com/thomasjacquin/allsky-website) to fix this issue.  Once it's reviewed and merged, you should be good to go.

closed 2022-02-04 by EricClaeys