AllskyTeam/allsky #993: Hangs after creating timelapse etc

skipperbilly, 2022-02-01:
### Discussed in https://github.com/thomasjacquin/allsky/discussions/991

<div type='discussions-op-text'>

<sup>Originally posted by **skipperbilly** February  1, 2022</sup>
My ASC with TJ software worked perfectly for months until a major upset caused by a Windows driver installation.

I replace my Rasp Pi and SD card and installed Bullseye and the latest version of TJ's software, Samba etc etc and after some teething problems I got it working.

It now runs all night and in the morning it produces a timelapse, keogram and star trail.

BUT

It then hangs - and shows the attached screen saying exp is 60 seconds (but it isnt) and displays a very underexposed image.

[image]

Any suggestions???

The camera is indoors at the moment for testing. It has a more than adequate power supply and the SD card is an ultra high speed Sandisk 64GB

I don't know how to access the fault log.

Please help - its driving me insane!!
</div>

EricClaeys, 2022-02-01:
@skipperbilly That's a bummer.  Please see the Wiki page on reporting problems and follow those instructions. It describes how to produce a log file qith the infi we need to help you.

skipperbilly, 2022-02-01:
Thanks

The files from this morning are attached. I have reset the log folder and
if the same issue happens again tomorrow I will send them to you.

I am not used to how this site works - should I email them to you or just
attach them to the message on the github page??

Many thanks for any help you are able to give me.

David (Skipper Billy)

skipperbilly, 2022-02-01:
Just seen your request to add the *.txt suffix to files - attached.

-- 
*Click here for my Astro-Imaging website.
<https://sites.google.com/view/astro-imaging/deep-space-objects>*

#!/bin/bash

# NOTES:
#  - Variable names must start at the beginning of a line.
#  - There should be nothing after a variable's value, i.e., no comments on the same line.
#  - Values can optionally be quoted with double quotes, NOT single quotes.
#  - If a variable occurs multiple times, only the last value is used.

# Choose between "ZWO" or "RPiHQ".  Note that "auto" is no longer allowed.
CAMERA="ZWO"

########## Images
# Set to "true" to upload the current image to your website.
IMG_UPLOAD="false"

# If IMG_UPLOAD is "true", upload images every IMG_UPLOAD_FREQUENCY frames, e.g., every 5 frames.
# 1 uploades every frame.
IMG_UPLOAD_FREQUENCY=1

# The websites look in IMG_DIR for the current image.
# When using the Allsky website from the "allsky-website" packge,
# "current" is an alias for "/home/pi/allsky".
# If you use the default IMG_DIR and have the Allsky website, set "imageName" in
# /var/www/html/allsky/config.js to:
#     imageName: "/current/tmp/image.jpg",
# This avoids copying the image to the website.
# Only change IMG_DIR if you know what you are doing.
IMG_DIR="current/tmp"

# Resize images before cropping, stretching, and saving.
# Adjust IMG_WIDTH and IMG_HEIGHT according to your camera's sensor ratio.
IMG_RESIZE="false"
IMG_WIDTH=2028
IMG_HEIGHT=1520

# Crop images before stretching and saving.
# This is useful to remove some of the black border when using a fisheye lens.
# If you crop an image you may need to change the "Text X" and/or "Text Y" settings in the WebUI.
CROP_IMAGE="false"
CROP_WIDTH=640
CROP_HEIGHT=480
CROP_OFFSET_X=0
CROP_OFFSET_Y=0

# Auto stretch images saved at night.
AUTO_STRETCH="false"
AUTO_STRETCH_AMOUNT=10
AUTO_STRETCH_MID_POINT="10%"

# Resize uploaded images.  Change the size to fit your sensor.
RESIZE_UPLOADS="false"
RESIZE_UPLOADS_SIZE="962x720"

# Create thumbnails of images.  If you are not running the WebUI consider changing this to "false".
IMG_CREATE_THUMBNAILS="true"

# Remove corrupt images before generating keograms, startrails, and timelapse videos.
# This only adds a few seconds per image as it's being processed.
REMOVE_BAD_IMAGES="false"

# If REMOVE_BAD_IMAGES is "true", images whose mean brightness is
# less than THRESHOLD_LOW or greater than THRESHOLD_HIGH percent (max: 100) will be removed.
# Set to 0 to disable the brightness checks.
REMOVE_BAD_IMAGES_THRESHOLD_LOW=1
[... 833 lines cut ...]
Jan 31 16:23:26 allsky allsky.sh[3419]:  Max Auto-Exposure (night): 30,000ms (30.0s)
Jan 31 16:23:26 allsky allsky.sh[3419]:  Delay (day): 5,000ms
Jan 31 16:23:26 allsky allsky.sh[3419]:  Delay (night): 1,000ms
Jan 31 16:23:26 allsky allsky.sh[3419]:  Gain (night only): 50, Auto: No, max: 200
Jan 31 16:23:26 allsky allsky.sh[3419]:  Gain Transition Time: 15.0 minutes
Jan 31 16:23:26 allsky allsky.sh[3419]:  Brightness (day): 100
Jan 31 16:23:26 allsky allsky.sh[3419]:  Brightness (night): 100
Jan 31 16:23:26 allsky allsky.sh[3419]:  Skip Frames (day): 5
Jan 31 16:23:26 allsky allsky.sh[3419]:  Skip Frames (night): 1
Jan 31 16:23:26 allsky allsky.sh[3419]:  Aggression: 50%
Jan 31 16:23:26 allsky allsky.sh[3419]:  Gamma: 50
Jan 31 16:23:26 allsky allsky.sh[3419]:  WB Red: 53, Blue: 110, Auto: No
Jan 31 16:23:26 allsky allsky.sh[3419]:  Binning (day): 1
Jan 31 16:23:26 allsky allsky.sh[3419]:  Binning (night): 1
Jan 31 16:23:26 allsky allsky.sh[3419]:  USB Speed: 80, auto: Yes
Jan 31 16:23:26 allsky allsky.sh[3419]:  Text Overlay: [none]
Jan 31 16:23:26 allsky allsky.sh[3419]:  Text Extra File: [none], Age: 600 seconds
Jan 31 16:23:26 allsky allsky.sh[3419]:  Text Line Height 60px
Jan 31 16:23:26 allsky allsky.sh[3419]:  Text Position: 15px from left, 30px from top
Jan 31 16:23:26 allsky allsky.sh[3419]:  Font Name:  SIMPLEX (0)
Jan 31 16:23:26 allsky allsky.sh[3419]:  Font Color: 255, 255, 255
Jan 31 16:23:26 allsky allsky.sh[3419]:  Small Font Color: 0, 0, 255
Jan 31 16:23:26 allsky allsky.sh[3419]:  Font Line Type: 16
Jan 31 16:23:26 allsky allsky.sh[3419]:  Font Size: 7.0
Jan

linuxkidd, 2022-02-01:
HI @skipperbilly unfortunately, it appears that git doesn't process email attachments.  Instead, you can open this issue in your browser, and as long as you're signed in, drag + drop the log file into the comment box.  This will attach it directly to the 'issue' so we can view and try to diagnose the issue.

Thanks!

skipperbilly, 2022-02-01:
Thanks - the files are attached here
[allsky.log.txt](https://github.com/thomasjacquin/allsky/files/7982576/allsky.log.txt)
[image]
[config.sh.txt](https://github.com/thomasjacquin/allsky/files/7982577/config.sh.txt)
.

skipperbilly, 2022-02-01:
Thank you.

I have attached the files to issue #993

Many thanks.

David (SkipperBilly)

EricClaeys, 2022-02-02:
@skipperbilly I see part of what is happening but the log file doesn't have all the information I need.  It's important to follow the instructions on the "reporting problems " Wiki page **exactly**.  There are a lot of starts and stops of Allsky due to not finding the camera that I don't think are related to your issue.

If you can start Allsky with a clean log file around 6 am that would be ideal; next best is to restart it right before going to bed then attach the log file when you wake up.  That way it has entries from the issue you are seeing plus little extraneous entries.

Did something happen around 1 am?  Exposure times went from consistently less than a second to 30 seconds (your nighttime max).  For example, someone turned off an outside light?

skipperbilly, 2022-02-02:
Thanks Eric

I have now followed the instructions to the letter and will send the next
clean level 4log this evening.

The unit is set up indoors for testing and about 1am I went to bed and
turned the lights off.

All the best

David.

skipperbilly, 2022-02-03:
Hello Eric

Last night I put the camera outside and did as you suggested to get a clean log file (attached) .

It was all OK until about 07.36 this morning when it stopped taking night images, switched to day mode and then started reporting 60 second exposures again but the images are very dark/black.

[allsky.log.txt](https://github.com/thomasjacquin/allsky/files/7993743/allsky.log.txt)

Jonk2, 2022-02-03:
Hi David,

My 178MC occasionally causes a 'starting up' and 'not running' loop on the allsky software.

Sometimes it's less than 24 hours, sometimes it's 5+ days. I just have to reboot.

I'm not sure that the allsky software is at fault, but would try and replicate the results, i.e. leve it running for a week and log what happens youself.

I suspect it's a similar issue to mine, which I don't think we can do much about.

I have the latest rpi 4b - what are you using?

Jon

EricClaeys, 2022-02-03:
@skipperbilly kind creepy that I could tell something happened around 1 am just by looking at tha log file, eh?  Makes you wonder what else I know about you :-)
Thanks for the log file.  I'll take a look when I have a chance.

skipperbilly, 2022-02-03:
LOL !!!

Thanks

David

EricClaeys, 2022-02-04:
@skipperbilly Something strange is going on with the camera.  At 7:51:21 it takes a 60 second exposure and at 7:52:16 (only 55 seconds later) it finishes with a mean brightness of 254 which is basically pure white.
It then tries 28 seconds which finishes 1 second later (huh?) with a mean brightness of 78 which is considered dark for daytime ("normal" daytime brightness is 120 to 134).
So then it tries a 49 second exposure which returns instantly (huh?) with a mean brightness of 39, which is darker than the prior picture with a _shorter_ exposure.
Allsky then tries 60 seconds again and still gets 39 brightness so it gives up, which is why you see a "60 second" exposure but a dark picture.  From what I can tell this isn't Allsky - it's the camera or the ZWO driver.

I have seen this with other ZWO cameras - a short exposure gives a brighter picture than a long exposure.

Didn't you say in Facebook you rebooted and the problem went away?  Did the reboot turn the camera off?

@Jonk2   FYI

skipperbilly, 2022-02-04:
@EricClaeys Thanks for having a look. It does seem odd doesn't it??

When I rebooted it, it was full power down reboot so power to the camera would have been lost then reconnected.

I assume that the driver for the camera is built in to the TJ software??  I haven't installed anything other than the TJ software (except SAMBA).

EricClaeys, 2022-02-05:
@skipperbilly we downloaded the ZWO Linux driver and include it with Allsky and the the capture.cpp program links to the driver.

Let's keep this Issue open a little longer in case the problem reapears.  If it doesn't we'll close the Issue.

skipperbilly, 2022-02-05:
OK thanks Eric.

Just checked and it did exactly the same thing yesterday and again today.

Weird!

EricClaeys, 2022-02-05:
@skipperbilly Did a full power down reboot help those times as well?

skipperbilly, 2022-02-05:
I left it running and after about 40 minutes it sorted itself out - same both days.

EricClaeys, 2022-02-05:
@skipperbilly can you include the log file from when it wasn’t working and then macically started working?  And let me know the appropriate times for when it changed?

skipperbilly, 2022-02-05:
Hello Eric

I have attached a segment of the log of when is started going weird and magically fixed itself - the rest of the log was all normal. (saves you trawling through hundreds of pages).

All the best.

David
[segmentofallsky.log](https://github.com/thomasjacquin/allsky/files/8009023/segmentofallsky.log)
.

EricClaeys, 2022-02-10:
@skipperbilly I can't make any sense of what's happening.  Did you start chanting when if magically started working, or switch from standing on you left foot to standing on your right foot? :-)  If I had to guess I'd say there is a hardware problem.  Do you have a Windows PC you can attach it to and run ZWO's Windows software?  Or know someone else who has a ZWO camera you can borrow?
Has the camera EVER worked, and if so, when and what software?

@linuxkidd, @AndreasLMeg,  Can you guys take a look at the "segmentofallsky.log" file @skipperbilly attached to see if you can make any sense of it?  60 second exposures appear to be taking a second according to the timestamps, and the mean brightness is all over the map at the same exposure.

skipperbilly, 2022-02-10:
Hello Eric

I may have been looking sideways whilst the wind direction changed ;-)

The camera works just fine with other software - like ZWO's own and
AllSkEye etc.

I do have another identical camera - I will swap them over when I get a
chance and see if it makes any difference.

It's a weird one isn't it!!???

AndreasLMeg, 2022-02-11:
@EricClaeys the first I would do, is to calculate the correct exposure for the gain change:
I have no ZWO, so I can't test this.
```
                // If we went from Night to Day, then current_exposure_us will be the last night
                // exposure so leave it if we're using auto-exposure so there's a seamless change from
                // Night to Day, i.e., if the exposure was fine a minute ago it will likely be fine now.
                // On the other hand, if this program just started or we're using manual exposures,
                // use what the user specified.
                if (numExposures == 0 || ! asiDayAutoExposure)
                {
					if (asiDayAutoExposure && asi_day_exposure_us > asi_day_max_autoexposure_ms*US_IN_MS)
					{
						snprintf(bufTemp, sizeof(bufTemp), "%s", length_in_units(asi_day_exposure_us, true));
						Log(0, "*** WARNING: daytime Manual Exposure [%s] > Max Auto-Exposure [%s]; user smaller number.\n", bufTemp, length_in_units(asi_day_max_autoexposure_ms*US_IN_MS, true));
						asi_day_exposure_us = asi_day_max_autoexposure_ms * US_IN_MS;
					}
                    current_exposure_us = asi_day_exposure_us;
                }
                else
                {
                    // If gain changes, we have to change the exposuretime to get an equal exposed image, we have to change
                    // ZWO gain has unite 0.1dB, so we have to convert the gain values to a factor first
                    // newExp = (oldExp * oldGain) / newGain
                    // e.g.
                    //   20s = (10s * 2.0) / (1.0) 

                    // current values are her the last night values
                    current_exposure_us = (current_exposure_us * pow(10, (float)currentGain / 10.0 / 20.0)) / pow(10, (float)asiDayGain / 10.0 / 20.0);
                    Log(2, "Using the last night exposure, old and new Gain to calculate new exposure of %s\n", length_in_units(current_exposure_us, true));
                }
```

AndreasLMeg, 2022-02-11:
@skipperbilly I have no ZWO, so this is only **brain storming**:

After the night/day change, allsky tries a few test exposures to determine a suitable exposure time. According to the logging, this goes wrong, because the camera seems to deliver images that are underexposed when the exposure times are too high (strange behavior, maybe a security setting, or an allsky bug)
With the above code adjustment, it could now work to get a suitable exposure time right away - we may then no longer get into this problem area

EricClaeys, 2022-02-11:
@AndreasLMeg I will add code to log the mode.  It's possible this camera is in a strange mode which is causing problems.  I will then add code to put the camera in the mode we want.
In the log file, note the timestamps. A 60-second exposure isn't taking 60 seconds, etc.  Very strange.

@skipperbilly FYI

AndreasLMeg, 2022-02-11:
@EricClaeys could also you make a test run with the code change (calculate the exposure time for night/day change) ?

EricClaeys, 2022-02-12:
@skipperbilly, would you please try the attached file?  I'm hoping it will give us some information on what's happening.
~~~
# Set "Debug Level" to 4 in the WebUI.
cd allsky/src
mv capture.cpp capture.cpp-ORIGINAL
# copy the attached file to this directory, removing the ".txt" from the name.
make capture
sudo systemctl stop allsky
cp capture ..
sudo truncate -s 0 /var/log/allsky.log
sudo systemctl start allsky
# Let it go for a couple minutes, then attach /var/log/allsky.log, appending ".txt" to the file name
~~~

@AndreasLMeg FYI

[capture.cpp.txt](https://github.com/thomasjacquin/allsky/files/8053667/capture.cpp.txt)

EricClaeys, 2022-02-14:
@skipperbilly First, I am very happy you found the issue and can now start enjoying your working camera.  Second, I  don't feel my time was wasted at all - I view this as a learning experience for all of us and plan to add something to the Wiki for when it eventually happens to someone else, so we can get them up and running much quicker than with you.
If you do what @Jonk2 mentioned, would you please take a few pictures that show the cable and whatever else you adjust?  That might make it easier for others to do.

Also, would you do me a favor and summarize the issue from your perspective?  I can fill in details from the log file.  I will then add this to the Wiki.

Thanks

skipperbilly, 2022-02-19:
Ok, I'll check it out.

David.

EricClaeys, 2022-02-22:
@skipperbilly, have you had a chance to check for moisture per @Jonk2 suggestion?  Someone else is having the same weird camera behavior as you and when he looked at the camera "it was full of water".

@AndreasLMeg, @linuxkidd, FYI

skipperbilly, 2022-02-22:
Yes - I did - it is bone dry. I also have coloured desiccant sachets in the
housing that turn from green to orange if they absorb any moisture. They
are all green.

I am just waiting to hear back from ZWO support to see if they can suggest
anything.

David.

EricClaeys, 2022-02-22:
@Jonk2 do you remember any other symptoms when you found moisture in your camera, other than 123C temp?  Exposures all over the place, 20 second exposures that only took 2 secomds, etc?

Jonk2, 2022-02-28:
@EricClaeys sorry for the late reply, I don't login much - when I noticed the incorrect temperature reading I wasn't at home at the time, I was away from the camera, so all I did was send a shutdown to the Pi and therefore the camera remotely.
When I got home and removed the camera for inspection after unsuccessfully trying to get it to operate correctly, I had to take it down to inspect which is when I found a fair amount of moisture inside - it turned out I hadn't sealed the chamber correctly and as it's a cooled version, a lot of condensation formed inside the camera due to the tec - lesson learnt!

EricClaeys, 2022-04-12:
Closing issue - the problem was due to moisture in the camera.  The Wiki mentions this when troubleshooting issues.

closed 2022-04-12 by EricClaeys