AllskyTeam/allsky #995: Error showing Live View after fresh install

Jooppie, 2022-02-02:
Hi Thomas,

I installed allsky on a raspberry pi zero 2 w with as RpiHQ. I used the buster release. All seems fine also the webui. But I do not het images. The error log is below.  When I excute the:

raspistill --output '/home/pi/allsky/tmp/image-20220202152706.jpg' --thumb none --burst -st --timeout 2000 --nopreview --mode 3 --exposure auto --analoggain 1 --awb off --awbgains 2.8,2 --vflip --saturation 50 --brightness 50 --quality 95 

It works fine, and a picture is saved. So that does not seem to be the problem.

Best, Ron

Feb  2 15:27:06 allskyron allsky.sh[8444]: ==========
Feb  2 15:27:06 allskyron allsky.sh[8444]: === Starting daytime capture ===
Feb  2 15:27:06 allskyron allsky.sh[8444]: ==========
Feb  2 15:27:06 allskyron allsky.sh[8444]: Stop the allsky service to end this process.
Feb  2 15:27:06 allskyron allsky.sh[8444]: STARTING EXPOSURE at: 2022-02-02 15:27:06   @ 32.00 ms (0.03 sec)
Feb  2 15:27:06 allskyron allsky.sh[8444]:   > Capture command: raspistill --output '/home/pi/allsky/tmp/image-20220202152706.jpg' --thumb none --burst -st --timeout 2000 --nopreview --mode 3 --exposure auto --analoggain 1 --awb off --awbgains 2.8,2 --vflip --saturation 50 --brightness 50 --quality 95
Feb  2 15:27:31 allskyron allsky.sh[8444]: /home/pi/allsky/allsky.sh: line 225:  8543 Killed                  "${ALLSKY_HOME}/${CAPTURE}" "${ARGUMENTS[@]}"
Feb  2 15:27:31 allskyron allsky.sh[8444]: 'capture_RPiHQ' exited with RETCODE=137
Feb  2 15:27:31 allskyron allsky.sh[8444]: ***
Feb  2 15:27:31 allskyron allsky.sh[8444]: *** After fixing, restart the allsky service. ***
Feb  2 15:27:31 allskyron allsky.sh[8444]: ***

EricClaeys, 2022-02-02:
@Jooppie Ron, the Wiki is your friend and workarounds to most of the common issues are there.  This is not an Allsky problem but an issue with the small amount of memory on your Pi.  Please see the Wiki page on RETCODE =137 to see what to do.

If that fixes your problem, please close the Issue.

Jooppie, 2022-02-02:
Hi Thomas,

Thanks, that works. Thank you for the quick response and the nice software!!

Best, Ron

EricClaeys, 2022-02-04:
Closing since the user reported the issue was resolved after following the instructions in the Wiki.

closed 2022-02-04 by EricClaeys