AllskyTeam/allsky #998: Captures 1 image and crashes.

mpinnuck, 2022-02-04:
Running on pi 3a+
1 image is saved.
Start on command line or via Web ui same results
Any suggestions on a fix appreciated

Feb  4 17:07:48 allsky allsky.sh[1115]: CAMERA: RPiHQ
Feb  4 17:07:48 allsky allsky.sh[1115]: CAMERA_SETTINGS: /etc/raspap/settings_RPiHQ.json
Feb  4 17:07:56 allsky allsky.sh[1115]: *** WARNING: Could not set locale to en_US.UTF-8 ***
Feb  4 17:07:56 allsky allsky.sh[1115]:  ********************************************
Feb  4 17:07:56 allsky allsky.sh[1115]:  *** Allsky Camera Software v0.8.3 | 2021 ***
Feb  4 17:07:56 allsky allsky.sh[1115]:  ********************************************
Feb  4 17:07:56 allsky allsky.sh[1115]: Capture images of the sky with a Raspberry Pi and a RPi HQ camera
Feb  4 17:07:56 allsky allsky.sh[1115]: Add -h or --help for available options
Feb  4 17:07:56 allsky allsky.sh[1115]: Author: Thomas Jacquin - <jacquin.thomas@gmail.com>
Feb  4 17:07:56 allsky allsky.sh[1115]: Contributors:
Feb  4 17:07:56 allsky allsky.sh[1115]: -Knut Olav Klo
Feb  4 17:07:56 allsky allsky.sh[1115]: -Daniel Johnsen
Feb  4 17:07:56 allsky allsky.sh[1115]: -Robert Wagner
Feb  4 17:07:56 allsky allsky.sh[1115]: -Michael J. Kidd - <linuxkidd@gmail.com>
Feb  4 17:07:56 allsky allsky.sh[1115]: -Rob Musquetier
Feb  4 17:07:56 allsky allsky.sh[1115]: -Eric Claeys
Feb  4 17:07:56 allsky allsky.sh[1115]: -Andreas Lindinger
Feb  4 17:07:56 allsky allsky.sh[1115]:  Camera: Raspberry Pi HQ camera
Feb  4 17:07:56 allsky allsky.sh[1115]:   - Resolution: 4096x3040
Feb  4 17:07:56 allsky allsky.sh[1115]:   - Pixel Size: 1.55microns
Feb  4 17:07:56 allsky allsky.sh[1115]:   - Supported Bins: 1x, 2x and 3x
Feb  4 17:07:56 allsky allsky.sh[1115]: Capture Settings:
Feb  4 17:07:56 allsky allsky.sh[1115]:  Command: raspistill
Feb  4 17:07:56 allsky allsky.sh[1115]:  Image Type: RGB24
Feb  4 17:07:56 allsky allsky.sh[1115]:  Resolution (before any binning): 4096x3040
Feb  4 17:07:56 allsky allsky.sh[1115]:  Quality: 100
Feb  4 17:07:56 allsky allsky.sh[1115]:  Daytime capture: 1 (yes)
Feb  4 17:07:56 allsky allsky.sh[1115]:  Exposure (night): 60000ms
Feb  4 17:07:56 allsky allsky.sh[1115]:  Auto Exposure (night): 1 (yes)
Feb  4 17:07:56 allsky allsky.sh[1115]:  Gain (night): 4.00
Feb  4 17:07:56 allsky allsky.sh[1115]:  Auto Gain (night): 0 (no)
Feb  4 17:07:56 allsky allsky.sh[1115]:  Brightness (day): 50
Feb  4 17:07:56 allsky allsky.sh[1115]:  Brightness (night): 50
Feb  4 17:07:56 allsky allsky.sh[1115]:  Saturation: 50.0
Feb  4 17:07:56 allsky allsky.sh[1115]:  Auto White Balance: 0 (no)
Feb  4 17:07:56 allsky allsky.sh[1115]:  WB Red: 2.80
Feb  4 17:07:56 allsky allsky.sh[1115]:  WB Blue: 2.00
Feb  4 17:07:56 allsky allsky.sh[1115]:  Binning (day): 1
Feb  4 17:07:56 allsky allsky.sh[1115]:  Binning (night): 1
Feb  4 17:07:56 allsky allsky.sh[1115]:  Delay (day): 30000ms
Feb  4 17:07:56 allsky allsky.sh[1115]:  Delay (night): 10ms
Feb  4 17:07:56 allsky allsky.sh[1115]:  Text Overlay:
Feb  4 17:07:56 allsky allsky.sh[1115]:  Text Position: 15px left, 75px top
Feb  4 17:07:56 allsky allsky.sh[1115]:  Font Name:  0
Feb  4 17:07:56 allsky allsky.sh[1115]:  Font Color: 255, 0, 0
Feb  4 17:07:56 allsky allsky.sh[1115]:  Font Background Color: 0
Feb  4 17:07:56 allsky allsky.sh[1115]:  Small Font Color: 0, 0, 255
Feb  4 17:07:56 allsky allsky.sh[1115]:  Font Line Type: 16
Feb  4 17:07:56 allsky allsky.sh[1115]:  Font Size: 32.0
Feb  4 17:07:56 allsky allsky.sh[1115]:  Font Line: 1
Feb  4 17:07:56 allsky allsky.sh[1115]:  Outline Font : 0 (no)
Feb  4 17:07:56 allsky allsky.sh[1115]:  Rotation: 0
Feb  4 17:07:56 allsky allsky.sh[1115]:  Flip Image: 0
Feb  4 17:07:56 allsky allsky.sh[1115]:  Filename: allsky.jpg
Feb  4 17:07:56 allsky allsky.sh[1115]:  Filename Save Directory: /home/pi/allsky/tmp
Feb  4 17:07:56 allsky allsky.sh[1115]:  Latitude: 33.71S
Feb  4 17:07:56 allsky allsky.sh[1115]:  Longitude: 151.08E
Feb  4 17:07:56 allsky allsky.sh[1115]:  Sun Elevation: -9
Feb  4 17:07:56 allsky allsky.sh[1115]:  Locale: en_US.UTF-8
Feb  4 17:07:56 allsky allsky.sh[1115]:  Notification Images: 1 (yes)
Feb  4 17:07:56 allsky allsky.sh[1115]:  Show Time: 1 (yes) (format: %Y%m%d %H:%M:%S)
Feb  4 17:07:56 allsky allsky.sh[1115]:  Show Exposure: 1 (yes)
Feb  4 17:07:56 allsky allsky.sh[1115]:  Show Gain: 0 (no)
Feb  4 17:07:56 allsky allsky.sh[1115]:  Show Brightness: 0 (no)
Feb  4 17:07:56 allsky allsky.sh[1115]:  Show Focus Metric: 0 (no)
Feb  4 17:07:56 allsky allsky.sh[1115]:  Preview: 0 (no)
Feb  4 17:07:56 allsky allsky.sh[1115]:  Taking Dark Frames: 0 (no)
Feb  4 17:07:56 allsky allsky.sh[1115]:  Debug Level: 1
Feb  4 17:07:56 allsky allsky.sh[1115]:  On TTY: No
Feb  4 17:07:56 allsky allsky.sh[1115]:  Mode Mean: 0 (no)
Feb  4 17:07:56 allsky allsky.sh[1115]: ==========
Feb  4 17:07:56 allsky allsky.sh[1115]: === Starting daytime capture ===
Feb  4 17:07:56 allsky allsky.sh[1115]: ==========
Feb  4 17:07:56 allsky allsky.sh[1115]: Stop the allsky service to end this process.
Feb  4 17:07:56 allsky allsky.sh[1115]: STARTING EXPOSURE at: 2022-02-04 17:07:56   @ 32.00 ms (0.03 sec)
Feb  4 17:07:56 allsky allsky.sh[1115]:   > Capture command: raspistill --output '/home/pi/allsky/tmp/allsky-20220204170756.jpg' --thumb none --burst -st --timeout 2000 --nopreview --mode 3 --exposure auto --analoggain 1 --awb off --awbgains 2.8,2 --saturation 50 --brightness 50 --quality 100
Feb  4 17:08:07 allsky allsky.sh[1115]: /home/pi/allsky/allsky.sh: line 225:  1201 Killed                  "${ALLSKY_HOME}/${CAPTURE}" "${ARGUMENTS[@]}"
Feb  4 17:08:07 allsky allsky.sh[1115]: 'capture_RPiHQ' exited with RETCODE=137
Feb  4 17:08:07 allsky allsky.sh[1115]: ***
Feb  4 17:08:07 allsky allsky.sh[1115]: *** After fixing, restart the allsky service. ***
Feb  4 17:08:07 allsky allsky.sh[1115]: ***

AndreasLMeg, 2022-02-04:
@mpinnuck 
Good morning, I have found this https://github.com/thomasjacquin/allsky/wiki/RETCODE=137-error
Maybe the solution

EricClaeys, 2022-02-04:
@mpinnuck   The Wiki is your friend.  Most of the known issues have troubleshooting pages that help you work around the issue, which usually isn't a problem with Allsky by with something else - in this case, a lack of memory is causing the issue.  The Wiki page has instructions to resolve it.  If that works for you, please let us know then close this Issue.  Thanks much!

mpinnuck, 2022-02-05:
Hi Andreas,
I found this article as well in the wiki.
Increasing the swap file size to 1024 did fix the problem.
Sorry for wasting your time.
Congratulations to Thomas and all contributors on a great piece of software.

Regards
Mark

mpinnuck, 2022-02-05:
Thank you for the prompt replies

closed 2022-02-05 by mpinnuck

AndreasLMeg, 2022-02-05:
@mpinnuck I'm sorry if I got the impression of wasting my time.
If I read an issue and I feel like providing first aid, I do so without wasting too much time on additional explanations.
If it's a new or big problem, I'll take more time. 
Besides Allsky, I have other hobbies (job, family,...)