AllskyTeam/allsky #999: Nighttime captures: autogain goes wrong when option Version 0.8 Exposure is set

vnawrath12159, 2022-02-05:
With the option „Version 0.8 exposure“ set to true, I’m experiencing a major problem with the nighttime capture, when „auto gain“ is set to true. Exposure is  set to manual with 6 seconds. When nighttime capture begins, allsky starts with the manual gain value, given in the camera settings. And then the autogain goes to the max autogain value (for me: 200), instead of adjusting the gain according the current situation.

For example: if the sky has a lot of clouds, the autogain should adjust from 10 to around 100, depended on the amount of sky, the clouds are covering. If the clouds go away, autogain raise to the max. autogain setting (it’s set to 200). This is the normal behaviour when option „Version 0.8 exposure“ is disabled (= „false“). But when this option is enabled, the gain value rises in any case over several captures up to the maximum value of autogain (= 200). This means, if the sky is cloudy, the pictures get overexposed, as allsky is taking the maximum gain value instead of the right one.

Short background, why I’m using „Version 0.8 expsure“:
With activating this option, there are no „ASI_TIMEOUT“ anymore during daytime capture. When I’m disabling this option, this error occurs several times per hour. This is also interrupting the „endOfNight“ actions after changing from nighttime to daytime captures. That means, latest when timelapse is created, the ASI_TIMEOUT orrurs five times, which will restart allsky. The restart interrupts the endOfNight creations.

My setting:
- Raspberry Pi 4B (4G)
- ASI ZWO 462MC Color
- Connected via USB3
- Allsky Version 0,83

Attached: log with debug level 4.

[allsky.log](https://github.com/thomasjacquin/allsky/files/8007844/allsky.log)

EricClaeys, 2022-02-05:
@vnawrath12159 I will have to look into this

AndreasLMeg, 2022-02-05:
@EricClaeys 
I have no experience with ZWO cameras.
Because it interests me, I would like to know the following:
1. What is meant by "Version 0.8 expsure"? (newexposure ? - I found it in allsky-portal)
2. Is this a first installation, a new installation or a problem that has come into the running system since an update?
3. Brightness (night): 0 - is that OK?
4. Gain Transition Time: 0.0 minutes - is that OK?
5. the timeout problem is not in the allsky.log - is this correct ?
6. do you have a logging (level 4)  for/with the day->night change  ?
7. USB bandwidth.  Values between 40-100. If you get ASI_ERROR_TIMEOUT errors in the log file, try chaning this value. - does this value change the behaviour ?

Yes maybe nohup is a soultion for endOfNight dying.

vnawrath12159, 2022-02-05:
@AndreasLMeg 

1. It is the newexposure. Parameter in camera settings is called "version 0.8 exposure"
2. Before I noticed the problem, I didn't use the newexposure feature. It was version 0.82. But after a new installation according the instructions in the wiki, this problem remained.
3. yes
4. yes
5. yes. The allsky.log I provided here is only the time period, when I switched "version 0.8 exposure" on.
6. No, I will do this logging tomorrow.
7. The ASI_ERROR_TIMEOUT occures with every USB bandwidth value between 40-100. And also with the parameter "Auto bandwidth" = yes. The ASI_ERROR_TIMEOUT does occur only, when "version 0.8 exposure" is set to false.

EricClaeys, 2022-02-06:
@AndreasLMeg 

1. A poor naming choice but it meant a lot to us developers. In version 0.7 the ZWO capture command turned on video mode when it started and only turned it off when it exited.  This meant the camera wes always doing something whih warmed the sensor.  In version 0.8 the capture command would turn video mode on, take a picture,  and turn video mode off.  Some people saw a 15 degree decrease in temperature. This had the unfortunate side effect of producing ASI_ERROR_TIMEOUT errors with some cameras so we added an option ( "Version 0.8 Exposure Mode") to disable the new mode.  In the log file it is called something like "Turn off video between pictures".
2. We changed the behavior in 0.8.
3. Brightness value is up to the user, but the default is 50 so 0 at night will producer a darker image.  I have not been able to determine how ZWO implements "brightness".  In their code they also call it an "offset" or a "auto exposure target brightness".  It somehow lightens the exposure and I believe only works with auto exposure.  Since I don't know how it works, during the day when our histogram code runs, it adds or subtracts 12% to the brightness for every 50 points.  So a valu of 0, which is 50 belos optw the default will be 12 darker, and 150 will be 24% brighte⁶r.  I got the number 12 from playing around with it.
4. If day and night gain are different I would set this to 10 or so minutes otherwise you'll have a big change in gain at the day/ night transition and a corresponding big change in brightness.  This option tries to smooth out those changes.
5. What timeout problem? ASI_ERROR_TIMEOUT?  If it's not in the log that means you don’t have the problem.
6.

EricClaeys, 2022-06-30:
Closing - this request was added to the Projects list - see the `Projects` link at the top of any Allsky GitHub page.
Project Title: **Port RPiHQ modeMean to ZWO**

Our plan is to port the Allsky auto-exposure algorithm from the RPi camera to ZWO.  Hopefully this will eliminate the nighttime flicker and ASI_ERROR_TIMEOUT's some users see.  That exposure algorithm also handles auto gain.

closed 2022-06-30 by EricClaeys