AllskyTeam/allsky #1005: new install of Allsky yesterday , issue with manual timelapse

ON2ON-HAM, 2022-02-07:
I needed to create a new SD card, and installed as before Allsky
All works as normal except for some reason when I perform the Timelapse script manually it comes with following error : 
Before I never had this kind of message,, swap file is also set to 1024
Running on a Bullseye.

I did a raspi update and update before installing Allsky

[image]

ON2ON-HAM, 2022-02-07:
Thanks Eric,
Solved : Difference between a Raspi Zero (used that last time with working TimeLapse) and the Raspi 3B+ I have now is that the 3B+ needs a different setting for Width and Height of the TimeLapseFrames.
Then it works !

Best Rgds
Luke

closed 2022-02-07 by ON2ON-HAM

EricClaeys, 2022-02-07:
@ON2ON-HAM With the same camera the 3b needs different settings?
What width and height did both cameras use?  And were both on Bullseye?