AllskyTeam/allsky #1011: upload image failed

Kilis71, 2022-02-10:
I have a problem uploading images to the website, it creates this error:

put: /home/pi/allsky/tmp/image-20220210231341.jpg: Errore fatale: Certificate verification: certificate common name doesn't match requested host name ‘ftp.astrofilipisani.it’ (D8:83:D7:51:A5:6B:BD:E7:90:BF:F5:B7:EB:CD:8C:BB:43:65:43:7A)
put of /home/pi/allsky/tmp/image-20220210231341.jpg failed!

EricClaeys, 2022-02-11:
@Kilis71 please take a look at the Wiki page on troubleshooting uploads.  If that fixes your Issue, please close it.
Eric

EricClaeys, 2022-02-11:
@Kilis71  What steps in the Wiki did you try, and what was in the `lftp_commands.txt` file afterwards?
The step for fixing the "certificate common name doesn't match requested host name" message should have worked.

Kilis71, 2022-02-11:
I solved it by inserting the string "set ssl: verify-certificate false" I had also made another mistake, I had entered the password of the site encrypted in base 64 instead of in clear text. now everything works, thanks for your interest, I spent a few sleepless nights there before solving :)

EricClaeys, 2022-02-11:
@Kilis71 Great!  Glad to hear the Wiki worked.
Would you please close this Issue now that it's been resolved?  Thanks

closed 2022-02-11 by Kilis71