AllskyTeam/allsky #1015: changing rotation causes image capture to fail

ssagerian, 2022-02-12:
[git_head.txt](https://github.com/thomasjacquin/allsky/files/8052458/git_head.txt)
[allsky.log.txt](https://github.com/thomasjacquin/allsky/files/8052460/allsky.log.txt)

EricClaeys, 2022-02-12:
@ssagerian Steve,  I just had the same problem after I updated my Pi.  The newest libcamera package requires that you update the firmware via `sudo rpi-update`.  I think the Raspberry team has released the libcamera software for the camera before it was ready for prime time.

I assume you also updated the software on your Pi?

ssagerian, 2022-02-12:
[allsky.log.txt](https://github.com/thomasjacquin/allsky/files/8053656/allsky.log.txt)

EricClaeys, 2022-02-12:
@ssagerian Steve, the libcamera software which is what Bullseye uses only supports rotation of 0 and 180 degrees.  I know that stinks.  Are you ok if I change the title of this Issue to "Support rotation of 90 and 270 degrees in libcamera" and then make it a feature request?

ssagerian, 2022-02-12:
so the project manager said to the developer I want you to rewrite the code to rotate this image zero degrees? How do I get THAT assignment?

ssagerian, 2022-02-13:
Well the bug would be then that the GUI has the option for 90 and 270 rotations, Do what you like.

EricClaeys, 2022-02-14:
@ssagerian Steve, I agree.  We need to change the WebUi.  There are also several options we need to add, and some other changes for Bullseye.

EricClaeys, 2022-04-12:
Closing issue.  The fixes have been made and will be in the next release.

closed 2022-04-12 by EricClaeys