AllskyTeam/allsky #1021: config.js files changes not updating

Skippyinspace, 2022-02-13:
Ive changed the config.js file in the website directory  on my webhost but other than removing the xxx title goes in here message, it just reads title now, 

No other title have changed such as location, computer, etc even though they have been added to the config.js page

any ideas, seems to be something simple

EricClaeys, 2022-02-14:
@Skippyinspace It's probably your browser caching the old file.  In your browser press F12, then click on the Network link on the right.  Under that check the box to disable the cache, then refresh the browser.
You may need to click in `>>` in order to see the Network link.

What is the URL to your website?

Skippyinspace, 2022-02-14:
The url to the website is www.skippyinspace.com/allsky

I may also need help in filling out the FTP setting js file
Thanks again

Ian

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Skippyinspace, 2022-02-17:
have manually edited the index file, which has sorted it for now.

closed 2022-02-17 by Skippyinspace