AllskyTeam/allsky #1025: No thumbnail and strange name of the videos on Webstie

Frankastro85, 2022-02-14:
Hello,
I haven't a thumbnail for the video and the name is strange...
[image]

What I do wrong ?
Thank you.
Frank

EricClaeys, 2022-02-14:
@Frankastro85 You didn't do anything wrong.  The strange name is a bug that we fixed and then broke.  I will fix again in a couple days.

The lack of thumbnail is because your website is on a remote server that has disabled the ability for tha Allsky website to run the command that creates the thumbnails. That's fairly common for security reasons. 

On our "to do" list is to allow the thumbnail to be created on the Pi and uploaded to the server.

Frankastro85, 2022-02-14:
Ok, thank you !  It's not dramatic ;-)
And about the name ?

uzelessknowledge, 2022-02-17:
@Frankastro85 I uploaded an image called 'Timelapse' and pointed the path to it. I prefer this rather than 'No Thumbnail'.
https://funrices.net/allsky/videos/

EricClaeys, 2022-02-17:
@uzelessknowledge, @Frankastro85 
The "No thumbnail" message is simply a .png file in the "allsky" directory.  Until we allow the actual thumbnail to be created on the Pi and uploaded, you can replace NoThumbnail.png with any image you want, and if you use the same filename then no code changes are needed.

Frankastro85, 2022-02-17:
Ok  good idea.
But the most important for me is the name (date).

mpinnuck, 2022-02-22:
The problem is the video filename prefix created by Allsky generateForDay.sh is "allsky" and the web script functions.php and the ./videos/index.php assume the prefix is "Timelapse". In the functions.php 
function display_thumbnails($image_type)
{
  if( $image_type == "Timelapse"){
      ... 
     $image_type_len = strlen(TIMELAPSE_PREFIX); 
  } else {
     ...
At the top of functions.php define('TIMELAPSE_PREFIX','allsky');
or simply put "allsky" in the code above. I do this so I will find it easier to maintain the prefix if it changes in the future.
This will give a correct date in div.day-text on the videos thumbnail page.

mpinnuck, 2022-02-22:
The most likely reason you have no thumbnails is the version of ffmpeg does not support H264 Decoder.
Type ffmpeg on the command line and it will list its configuration.
I run my server on a QNAP NAS with a version if ffmpeg that did not support a H264 decoder. This resulted was no videos thumbnails.
I upgraded ffmpeg on the NAS and thumbnails are generated ok.
There web scripts are all ok.

EricClaeys, 2022-02-22:
@mpinnuck Mark, there is a fix for the name issue that is being reviewed.  It runs fine on my two Pi's.

I believe @Frankastro85 is using a hosted server for his website, and hosted solutions often disable the exec() function which calls ffmpeg, for security reasons. I have a version of functions.php that detects when exec() is disabled, and doesn't use it.
The ultimate fix for no video (and less frequently startrails and keogram thumbnails) is to have them created on the Pi and uploaded along with the full-size version.

mpinnuck, 2022-02-22:
Thanks for the feed back just letting you know my experience for what it’s worth.
Yes, ISPs’ typically disable server side scripting on domestic personal use web hostings that come included with internet access package.
Uploading html from the pi will work in all environments.
You guys have done a great job on the software and works well day night.

Frankastro85, 2022-02-22:
@EricClaeys 
Hello,  sorry  for this delay but I'm on holiday.
I'll be back next week and try your solution. 
Thanks 😊

EricClaeys, 2022-02-22:
@mpinnuck Mark, thanks for the kind words.  We enjoy improving Allsky and comments like yours help keep us going.

I will create a new Wiki page describing "no thubnails" on the website, and will mention your and my experiences.  If the remote website is under control of the user, they can likely fix the issue.

EricClaeys, 2022-02-25:
Closing this issue.  It's the same as [this one](https://github.com/thomasjacquin/allsky-website/issues/84) which is correctly against the allsky-website package.  It's not an issue with the "allsky" package.

@mpinnuck, @Frankastro85, @uzelessknowledge - FYI

@mpinnuck I update the [allsky website Wiki page](https://github.com/thomasjacquin/allsky/wiki/Troubleshooting:-Website-(allsky-website)-Issues) to include your comments about the version of ffmpeg.  Thanks for letting me know, and if you have any feedback on that page, let me know.

closed 2022-02-25 by EricClaeys