AllskyTeam/allsky #1032: QUESTION: Make timelapse slower.

jmodrako, 2022-02-17:
Hey team! How can I make a timelapse a little bit slower as right now it's so fast that I don't like.

EricClaeys, 2022-02-17:
@jmodrako In allsky/config/config.sh there's a setting called "FPS" (Frames Per Second).  Lower it.
To see what your change looks like without waiting until tomorrow, run:
~~~
allsky/scripts/generateForDay.sh -t 20220216
~~~
then look in allsky/images/20220216 for the allsky-20220216.mp4 file.

Note that if you lower the FPS a lot, for example, to 10, you'll notice the video gets jerky.

Going forward, when you have a question, comment, or want to suggest a change, please do so in the Discussions link above.  We'd like to keep "Issues" limited to actual Issues.

If this answers your question, please close the Issue.
Thanks.

EricClaeys, 2022-02-25:
@jmodrako Did it help?

aaronwmorris, 2022-03-01:
The other option is to increase the number of frames.  You can crank your gain and lower your exposure time to increase the total number of exposures taken over the night.

closed 2022-06-03 by EricClaeys