AllskyTeam/allsky #1036: Capture error

KaktuMAN, 2022-02-17:
Hello,
After updating to the new version, allsky worked for one day but since this day it stopped working.
I tried to go back to the previous version but without any improvement. So I re-installed the new version and the bug still here. Even by trying to debug the values causing the bug, I didn't find the place where the values where.
What happen : the service starts but after ~5sec. it shutdown, then restart (And it's basically a loop)
Here is the full log of allsky. 
[allsky.log](https://github.com/thomasjacquin/allsky/files/8091203/allsky.log)

EricClaeys, 2022-02-17:
@KaktuMAN Please see the Wiki page on Troubleshooting ASI_ERROR_TIMEOUTs.

Did you stop allsky, or did it end on its own?

KaktuMAN, 2022-02-17:
> @KaktuMAN Please see the Wiki page on Troubleshooting ASI_ERROR_TIMEOUTs.
> 
> Did you stop allsky, or did it end on its own?

Ho, yes I forgot about that and only made 1st step (I'm going to test that)

Allsky end on its own

KaktuMAN, 2022-02-18:
So, the first command isn't working (Command syntax) and the list of steps to follow aren't solving the problem do you want the log again ?

EricClaeys, 2022-02-18:
@KaktuMAN Is the first command the `uhubctl` command?  I just updated the command line in the Wiki based on your feedback.  However, it only works if you have `uhubctl` installed on your Pi.

- Did you unplug the camera from the Pi?
- What USB bandwidth settings did you try?
- What did the WebUI System page say about throttling?
- Were you able to move the USB cable to a different port?
- Did you set "Version 0.8 exposure mode" to Yes and No?
- Do you have a powered USB hub you can try?  Or a more powerful power adapter?  Or shorter/thicker cables?
- If you have a Windows PC, plug the camera into it and use the ZWO software that came with the camera to see if it also has problems.  If it does, it's likely a hardware issue.

I don't think another log file will tell us anything we don't already know.

KaktuMAN, 2022-02-18:
Now the result is : No compatible smart hubs detected at location 1-1! 

- I cannot, I don't have access to the PI
- 40-80-100 (Auto ON/OFF)
- No throttling
- I cannot, I don't have access to the PI
- Yes, (And tried to change bandwith with each state)
- No, I don't have access to the PI
- And I can't I don't have access to the PI

The problem appeared only after updating to the new version. Even by downgrading it didn't restarted the camera

EricClaeys, 2022-02-18:
@KaktuMAN To get the version look in /var/log/allsky.log for "Camera Software".  The version in the log file you attached yesterday was v0.8.3 which is the latest.

EricClaeys, 2022-02-18:
@KaktuMAN Another thing to try:

Enable `DebugPrint` in ~/.ZWO/ASIconfig.xml.  It will add a LOT of lines to /var/log/allsky.log so you'll want to disable it after Allsky stops due to errors.  Attach the log file when done.

KaktuMAN, 2022-02-18:
> @KaktuMAN To get the version look in /var/log/allsky.log for "Camera Software". The version in the log file you attached yesterday was v0.8.3 which is the latest.

V6

KaktuMAN, 2022-02-18:
[image]
What do I need to put ?

KaktuMAN, 2022-02-18:
And also I decided to update to the 8.3 version because I started to get corrupted timelapse

EricClaeys, 2022-02-18:
@KaktuMAN on the `debugPrint line change the `00` to `01`.  You will need to restart Allsky.

KaktuMAN, 2022-02-18:
[allsky.log](https://github.com/thomasjacquin/allsky/files/8100534/allsky.log)
Here you go @EricClaeys

KaktuMAN, 2022-02-19:
I'll bring the raspberry back tomorrow to do some testing.

EricClaeys, 2022-02-20:
@KaktuMAN   Good.  I don't know what else to try other than resetting the USB bus, checking power, etc.

EricClaeys, 2022-02-22:
@KaktuMAN Yes. Please keep me informed.   Someone else recently had the same issue as you and had a lot of moisture in the camera, and drying it out solved their problem.

KaktuMAN, 2022-03-12:
Update : The errors are back ! 👌
Allsky "worked" for 5 days, after that I brought it back to where it was before the problem. Then 2 days without any issue on the third day no images was taken by allsky (As of now I don't know why, but rebooting 3 times the pi and the pictures were back).
It worked again for one night but today no images, no timelapse. I've got a blank image from the camera at 6:26am nothing saved. (ASI_ERROR_TIMEOUT #1) So I'm trying all the things of the wiki but I think I just have to get the camera again 😢

EricClaeys, 2022-03-12:
@KaktuMAN Sorry to hear that.  Let me know what you find when you get the camera

KaktuMAN, 2022-04-02:
Hello,
Little update ! 
What I have tried : 

- Fresh install of allsky 👎
- Fresh OS install 👎
- New OS 👎
- Change SD card 👎
- Change raspberry (And do all the steps again :) ) 👎 

Currently I'm trying to let the camera run for a day on a windows computer to see if the communication crash (Aka, camera dead). If not I will try with a new cable between the camera and the raspberry
It's the same error as previously (`ASI_ERROR_TIMEOUT`) but with `#X` changing. I'll update you when I have more information ✌️

EricClaeys, 2022-04-02:
@KaktuMAN the Windows machine may be supplying more power to the camera on the USB bus, so you may get different results.
If it works on Windows but not the Pi, try a powered USB hub or a different power adapter and cable to the Pi.  And use the shortest USB cable you can.

EricClaeys, 2022-04-11:
@KaktuMAN    Any update?

KaktuMAN, 2022-04-12:
Yes, update ! With the last soft update you've made we have seen the real issues with the raspberry
We did a lot of experiments from changing the power adapter (it worked for a day) to changing the whole system (PSU and Raspberry) and sunday we tried to power the camera and the pi separately. Let me explain,
We wired the raspberry normally to the PSU and we used the PSU also to power the camera (So the raspberry isn't powering the camera = no more power issues, the raspberry is just getting the data from the camera). With this configuration everything is working properly now the only problem is the raspberry pi overheating, we are working ont it.
So I assume that the error was coming from the camera getting underpowered and was stopping all communication to the raspberry. We are doing some last tests to ensure that it was this error, tonight the camera will be outside. @EricClaeys