AllskyTeam/allsky #1040: keogram and startrails 'std::out_of_range'

clint70, 2022-02-19:
Running any scrip - endofnight (in this case) or any other that creates keogram, startrails fails.  I have not been able to track down where the error is.  Logs have not indication...  

Output is attached. 
[allsky_error_endofNight.txt](https://github.com/thomasjacquin/allsky/files/8101250/allsky_error_endofNight.txt)

clint70, 2022-02-19:
@EricClaeys  thank you, I will run that here in a little bit and thx, I will set that end of night to 'false'.  Be nice to have it running with no warning(s).  Once I get the information, I will post here.

EricClaeys, 2022-02-19:
@clint70   Sorry about the wrong name.  You were correct.  I was on my phone so going by memory.

**It takes a couple seconds per picture.**  When run after every picture is taken, it's not noticeable.  In version 0.8.3 the default is to do that - see REMOVE_BAD_IMAGES (?) variable in config/config.sh.

 I just tried removeBadImages.sh on a small directory of mine and didn't get the error you got, and looking at the code I'm not sure how it's possible to get that error.  Do you have the newest version of removeBadImages.sh?

Does `/home/pi/allsky/tmp/badError.txt` exist?  If so, what's in it?
Does `/home/pi/allsky/tmp/removeBadImages.log` exist, and if so, what's currently in it?

clint70, 2022-02-19:
@EricClaeys, Ya, not sure what is going on, both of those files are there, but zero bytes.  For your info, the next day 20220218 also had badImages that had to be removed.  I decided to run the script against that day to be safe.  

Also, once the bad images were removed all was good.  The keogram, startrails and timelapse all generated and had no errors.  OTHER than this in the TimeLapseErr.txt
[timelapseTMP.txt](https://github.com/thomasjacquin/allsky/files/8103147/timelapseTMP.txt)

I think now I know how to 'fix' it, I'll keep an eye on it and see if things continue.  BUT for now, I believe that this issue is fixed for me.  LET me know if there is more information you would like to see...  thank you for the time / help

EricClaeys, 2022-02-20:
@clint70   **Would you please upload one of the bad files** so I can see why keogram and startrails don't gracefully detect it?
After that, if you set REMOVE_BAD_IMAGES=true in config.sh, you won't have to run it manually.

The timelapseTMP.txt file you attached is exactly what should be there. One "ln -s" line for every file, and the last message about pixel format that you can safely ignore per the Wike page on troubleshooting timelapse issues.

EricClaeys, 2022-04-11:
Closing Issue - no response from user in almost 2 months.

closed 2022-04-11 by EricClaeys