AllskyTeam/allsky #1055: v0.8.3 startrails & crop question

jcauthen78, 2022-02-27:
brand new install - 26 Feb 2022
Camera: ASI178MC (native resolution 3096x2080)

Found this when wondering why my timelapse for last night didn't work (manual output line is below).

`CROP_IMAGE="true"`
`CROP_WIDTH=2080`
`CROP_HEIGHT=2080`
`CROP_OFFSET_X=15`
`CROP_OFFSET_Y=-40`

The resulting image is coming out 2080x2040 and i'm not sure why. I've looked through the config.sh and can't see why it's trimming 40px off the height. Tested and set it to 2040x2040, resulting image is 2040x2020 (20px missing).

I tried running the manual timelapse this morning. 
`./startrails -b 0.26 -d /home/pi/allsky/images/20220226 -e jpg -v -o 20220226.jpg >> /tmp/startrails.txt`

The output (for every image): 
`/home/pi/allsky/images/20220226/image-20220227012910.jpg: image size 2080x2040 does not match expected size 2080x2070; ignoring`

I'm not finding where that image size/ratio is being called out... feels like it is hard-coded somewhere? Not seeing it in the config.sh, I may be missing it somewhere else. I tried declaring the size with the -s flag: 
`./startrails -b 0.26 -d /home/pi/allsky/images/20220226 -e jpg -v -s 2080x2040 -o 20220226.jpg >> /tmp/startrails.txt`
result-
`/home/pi/allsky/images/20220226/image-20220226201752.jpg: image size 2080x2070 does not match expected size 2080x2040; ignoring
ERROR: Unable to read sample file '/home/pi/allsky/images/20220226/image-20220226201752.jpg'; quitting`

I wasn't running into this error on my previous v0.8.3 install, unfortunately i can't go snag the older startrails file because the memory card decided it didn't like to work any more. 

There's a high chance I'm completely missing something, but I think I'm on track.

EricClaeys, 2022-02-28:
@jcauthen78 Jim, after reading your post I spent some time trying to understand the crop.  I tried it once a year ago and couldn't figure it out (I didn't know it cropped from the **center**), so I gave up.

I'm pretty sure I understand it now.  Your sensor is 2080 px high and you're telling Allsky you want a 2080 px high crop, starting at CROP_OFFSET_Y of -40.  Since the crop height and the sensor height are the same, you are trying to move the crop rectangle above the top of the sensor by 40 px, and the bottom of the crop rectangle should therefore be 40 px up from the bottom of the sensor (2040).  There are no pixels on the sensor at -40, which is above the first row of pixels.  Instead of giving an error you're only getting the 2040 pixel height that's on the sensor.

Unfortunately saveImage.sh (which does the crop) doesn't know the sensor size so can't check for invalid conditions like this.

It's not an issue with the width because a 2040 crop moved to the right by 15 pixels puts the right edge at 2056 which is less than the sensor width of 3096.

For the 2040x2040 crop giving 2040x2020 the width isn't a problem for the same reason as above.
The center of the 2080 tall sensor is at 1040 pixels (2080/2).
Half the crop height is 1020 (2040/2).
Since crop rectangles are centered on the center of the sensor, half the crop (1020) is above and half (1020) is below the center.  So the top of the crop rectangle is 20 (1040 - 1020), and the bottom of the crop rectangle is 2060 (1040 + 1020).  OK so far, BUT the CROP_OFFSET_Y is -40 which moves the crop rectangle up 40 px.  The top of the box is now **-20** (20-40) and the bottom of the box is 2020 (2060-40).

The top 20 px are cut off since there are no pixels on the sensor at -20.

I will add a page to the Wiki about troubleshooting/explaining crops.

I will reply to the timelapse issue in the next post.

jcauthen78, 2022-02-28:
@EricClaeys - Ahh, that makes complete sense. Was using the offset to center up the visual 180° circle while maintaining the full circle - didn't take into mind that there weren't pixels over there to maintain the image resolution I was calling out. I was fighting it and it was telling me '`im doing exactly what you're telling me!`'  haha.  dang computers being smarter than me. 

I think I just figured out the Timelapse issue (screenshot attached). Now that I understand the crop offset and how it can change image size, I pulled up the folder in Adobe Bridge to look a little deeper. I changed the crop to center up a little bit after I re-mounted it outside, which changed the image size part way through the run.  Went from 2080x2070 to 2080x2040 earlier in the night and didn't realize its implications till this morning when I saw that the run last night went just fine and gave me the correct Keogram & Startrails. 

So, I'm chalking these both up to impatient user error. 😅
[image]

EricClaeys, 2022-02-28:
@jcauthen78 So both problems are fixed?
I'd like to keep this Issue open since I'm testing a change to saveImage.sh to check for invalid CROP_WIDTH, CROP_HEIGHT, CROP_OFFSET_X, and CROP_OFFSET_Y values.  saveImage.sh can run the `identify` command to get the image resolution and then do the math I did in my last post.  I've tested it with good values, with each of the four CROP_* variables bad, and with a combination of CROP_* variables bad.  All tests passed.

I am going to have it output an ERROR to the log file, then stop Allsky which will force the user to fix the bad number(s) before it'll run Allsky again.

jcauthen78, 2022-02-28:
sounds like a plan. I like the idea of doing the `identify`.

EricClaeys, 2022-04-11:
@jcauthen78   I'm closing this Issue.  I added sanity checks to the CROP_* variables in saveImage.sh (the problem bit me as well, so I got off my butt and made the changes).  **They changes will appear in the next release.**
The code checks for crop areas that are partially or fully off the image, numerical values that aren't numbers, e.g., CROP_WIDTH="20%", and some other things.

closed 2022-04-11 by EricClaeys