AllskyTeam/allsky #1061: Exposure time <1ms

Bra1nsen, 2022-03-02:
Hello, I can't thank you enough for your service. I would be interested to know if it is possible to further minimize the exposure time. i used the skycam for the cloud observation. To avoid overexposure it would be great, if one can choose the shortest possible exposure time. Raspberry HQ Cam ~ 0.3ms

Many greetings,
Pau
[image]
l

EricClaeys, 2022-03-02:
@Bra1nsen Not directly until the next release.
In the meantime, there are a couple ways you can do it:
### Edit the code.
In `src/capture_RPiHQ.cpp`, change the line
~~~cpp
asiDayExposure_us = atoi(argv[++i]) * US_IN_MS;
~~~
to
~~~cpp
asiDayExposure_us = atof(argv[++i]) * US_IN_MS; 
~~~
In the most current version this is line 985.
Then do:
~~~shell
cd allsky/src
make capture_RPiHQ
sudo systemctl stop allsky
cp capture_RPiHQ ..
sudo systemctl start allsky
~~~

Now change the WebUI to allow you to enter a fractional ms:
In `/etc/raspap/camera_options_RPiHQ.json`, look for `"name":"dayexposure",`, then change "number" 4 lines below it to "text".  That line should now look like `"type":"text"`.  
Go to the WebUI's "Camera settings" page and you should be able to enter a fraction of daytime exposure in ms.

### Use auto-exposure mode
Our new auto-exposure mode will use whatever exposure produces the best image, all the way down to 1 microsecond.  To use it, in `config/config.sh`, set `CAPTURE_EXTRA_PARAMETERS="-nightmean 0.3"` then restart Allsky.

Bra1nsen, 2022-03-02:
awesome, thanks eric - I'll let you know once I've made it.

how does the new auto exposure mode work? HDRImaging?

EricClaeys, 2022-03-03:
@Bra1nsen   The auto-exposure gets the mean value of a portion of the image (I think the center 1/3) and tries to match what you used for "-nightmean" (0.3 in the example above), where 0 is black and 1 is pure white.  It keeps a history of the last few images so it doesn't ping pong between too bright and too dark.  It's not perfect but is pretty good.  We plan to port it to the ZWO since it does both day and night, and our ZWO auto exposure algorithm only does daytime.

Bra1nsen, 2022-03-05:
Thank you Eric it worked perfectly
![0 1ms](https://user-images.githubusercontent.com/58213050/156879559-646c
[image]
5fed-3ca8-45f9-9223-69e8e9f8eefb.png)
[image]

Bra1nsen, 2022-03-05:
Would you give me your e-mail, I have another request for you. It's about the integration of an HDR algorithm. Maybe you are interested, I would also compensate you financially for your help.

EricClaeys, 2022-03-05:
@Bra1nsen Please post any feature requests to the Discussions page so all the Allsky developers will see them.
I believe the libcamera command has an HDR mode, but we aren't using it because most people are still on Buster which uses `raspistill` to take pictures,  and it doesn't have an HDR mode.  This doesn't mean we can't use libcamera's HDR though.

EricClaeys, 2022-03-05:
@Bra1nsen if your original issue is resolved would you please close this issue?
Thanks

closed 2022-03-05 by Bra1nsen