AllskyTeam/allsky #1065: Special Character in FTP Pwd

CleverJay, 2022-03-06:
### Discussed in https://github.com/thomasjacquin/allsky/discussions/1064

<div type='discussions-op-text'>

<sup>Originally posted by **CleverJay** March  6, 2022</sup>
I have a special character "$" in my FTP password that is causing  files not to upload to my web server. is there a way to escape the $ in my password?
</div>

CleverJay, 2022-03-06:
Using backslash "\$" doesn't seem to work as an escape. I have other LFTP scripts that are working with the non-escaped "$"

EricClaeys, 2022-03-07:
@CleverJay Can you attach one of the working scripts, changing the password as necessary?

EricClaeys, 2022-03-07:
@CleverJay  Glad you got it working.  In ftp-settings.sh did you use a single backslash ("\") to escape it?

I'm reopening this Issue since I want to fix the code so users don't have to escape special characters.  There are several of them and most people won't know what they are until uploads fail.  Once I have a fix I'll close the Issue.
In you scripts, if your password had a ":" or "@" you'd need to escape those as well.

CleverJay, 2022-03-07:
@EricClaeys yes, a single backslash before the dollar sign `\$` Thanks again for all of your hard work on this project.

linuxkidd, 2022-07-17:
The need to escape the `$` is due to the use of double quotes to enclose the [password variable](https://github.com/thomasjacquin/allsky/blob/master/config_repo/ftp-settings.sh.repo#L77).  Double quotes allows bash to expand the `$` as if it were the start of a shell variable.

We can switch to using single quotes instead, which prevents bash expansion inside them, but the issue then would be the need to escape single quotes in your password.   As far as I know, there's no "good" way to not need to escape at least some character, although single quote enclosure seems to be the lesser of all evils.

EricClaeys, 2022-12-05:
Issue resolved in next release.

PR #1982 (2022-12-05): Update upload.sh
Fixes #1065 
This eliminates the problem of "special" characters in the password.
commit 4b694a85 (2022-12-05): Update upload.sh

Fixes #1065 
This eliminates the problem of "special" characters in the password.
closed 2022-12-05 by EricClaeys