AllskyTeam/allsky #1066: Fresh Install on Pi Zero W with 120MC-S stuck on "software is starting up"

DamoclesSword, 2022-03-08:
I posted about this months ago,  but gave up on the issue. I decided to reinstall everything on a Pi Zero W with a 120MC-S. The USB 3 vs 2 issue seems to not affect this camera. I have everything initialized and working with the GUI, but it gets stuck on "Allsky software is starting up". No matter what I do, nothing gets it past this screen. Any solution?

DamoclesSword, 2022-03-08:
I am not sure how to attach logs, but I ran this 
~/allsky $ tail /var/log/allsky.log

and go: 
Mar  7 21:29:39 allsky allsky.sh[1248]: === Starting nighttime capture ===
Mar  7 21:29:39 allsky allsky.sh[1248]: ==========
Mar  7 21:29:39 allsky allsky.sh[1248]: WARNING: Value of 200 greater than max value allowed (100) for control 'AutoExpMaxGain' (#10).
Mar  7 21:29:39 allsky allsky.sh[1248]: WARNING: Value of 50 greater than max value allowed (20) for control 'Offset' (#5).
Mar  7 21:29:39 allsky allsky.sh[1248]: STARTING EXPOSURE at: 2022-03-07 21:29:39   @ 10.0 sec
Mar  7 21:29:50 allsky allsky.sh[1248]: STARTING EXPOSURE at: 2022-03-07 21:29:50   @ 6.1 sec
Mar  7 21:29:56 allsky allsky.sh[1248]:   > Saving NIGHT image 'image-20220307212950.jpg'
Mar  7 21:29:56 allsky allsky.sh[1248]:   > Sleeping: 443.9 sec
Mar  7 21:30:05 allsky allsky.sh[1276]: removeBadImages.sh: File is bad: removed 'image-20220307212950.jpg' (above threshold: MEAN=99.920900, threshold = 90)
Mar  7 21:31:05 allsky allsky.sh[1248]:      ***** Stopping AllSky *****

I am not sure if this is related to the issue or not

EricClaeys, 2022-03-08:
@DamoclesSword Please read the Wiki page on reporting issues.  It describes how to set up the log file to contain the necessary information.  You can attach the log file by appending a ".txt" to a copy of it, then simply dragging and dropping it to the area at the bottom of your reply.

Sleeping 443.9 seconds is unusual.  What is your night delay?

You can resolve the two WARNING messages in the WebUI's "Camera Settings" page.  The first one is for nighttime "Max Gain" and the second one is for "Brightness (offset)".  Your values of 200 and 50 are greater than the maximum of 100 and 20, respectively.

The "removeBadImages.sh" message is saying that the image is brighter than the threshold of 90.  Its mean brightness is 99.92 which is almost pure white, so it's deleting the picture.  I suspect all your pictures are too bright so it never saves a valid picture,  which is why the "Allsky software is starting up" message stays there - it only goes away once the first valid image is saved.

EricClaeys, 2022-03-08:
@DamoclesSword   All commands are run from the command line in a terminal window.  You can get to the Pi via ssh, which only gives you one window, or via VNC, which presents a remote desktop,  which IMHO is much nicer since you can open many terminal windows and other programs.

If you followed the instructions on the "Reporting Problems" Wiki page, please attach /var/log/allsky.log.  I am pretty sure all the images before 11 pm were extremely bright so were being discarded, and then around 11 pm they got dark enough (average brightness less than .9, where 0 is pure black and 1 is pure white) so the images were kept and hence displayed. 
In the "Editor" page of the WebUI, try changing the REMOVE_BAD_IMAGES variable to "false", then in the "Camera Settings" page turn on the "Show Histogram" option to show the mean brightness.  This should give you an idea of what's going on. 

On the bottom of the WebUI, click on the "Show advanced options" button.  You should then see the `Brightness (offset)` and `Max Gain` options.

What commands are you running when you "get nothing"?

For the timelapse, keogram, and startrails, you can look in the log file for "Starting startrails", etc, and then look for error messages.
Or you can run
~~~shell
cd allsky/scripts
./generateForDay.sh  -s 20220307
~~~
to generate a startrails for yesterday, and see what it says.
"-k" will do a keogram and "-t" will do a timelapse.

DamoclesSword, 2022-03-09:
@EricClaeys Maybe there is a step I am missing, but once I SSH into the pi, I run the sudo systemctl stop allsky then I run the truncate and nothing happens after that. I try typing /var/log/allsky.log but that returns -bash: /var/log/allsky.log: Permission Denied. If there's something else I need to do, I dont see it on the wiki. I can look into the VNC but I have a feeling that might stress the little Pi Zero too much. 

You right about the images, I think something happened with the gain where it was forcing a gain of 200 in that hidden menu. I changed it after I found it later. 

As for the commands, they are what I mentioned above.

Lastly, do the timelapse, keogram, and startrails not generate automatically? I can attempt to generate a timelapse but I know it is going to take awhile.

EricClaeys, 2022-03-09:
@DamoclesSword `/var/log/allsky.log` is the name of the log file.  Typing `/var/log/allsky.log` by itself will tell Linux you want to execute the file, but it's not a program so it's not executable, hence the "Permission Denied" message.
If you want to **view** the file, you have to use a text editor like `nano /var/log/allsky.log` or to view the last 50 lines of the file you can `tail -50 /var/log/allsky.log`.  You can change "50" to another number. The command I use the most is `tail -f /var/log/allsky.log` which will continually display whatever is written to the file, in real time, until you enter CTRL-C to exit the command.
To get a copy of the log file to upload to GitHub, do this:
~~~shell
cd
cp /var/log/allsky.log   allsky.log.txt
~~~
The "cd" command puts you in /home/pi.
The "cp" command copies the log file to a file called "allsky.log.txt" in /home/pi.  GitHub won't let you upload a file ending in ".log" so you're basically renaming the file to end in ".txt" which GitHub accepts.  You then need to transfer that file to a PC to upload to GitHub.  If you use VNC, you could run the web browser on the Pi and simply drag and drop (another advantage of using VNC).

Linux is case-sensitive, so `generateforDay.sh` is different than `generateForDay.sh` (capital "F").
Try again, making sure it's exactly as below:
~~~shell
cd /home/pi/allsky/scripts
./generateForDay.sh  -s 20220307
~~~

A gain of 200 shouldn't cause the pictures to be almost pure white.  However, if you had both auto gain and auto exposure on at the same time, that could cause the problem.  Some people have reported strange results when both are set to "auto".  I suggest turning auto gain **off** and auto exposure **on**, then play around with the gain setting.  Higher values produce brighter, but grainier images.   Yesterday I lowered my gain from 300 to 275 because there was too much grain.  Cameras have different maximum values for gain and other settings.  If you set the "Debug Level" to 4, then truncate the log file, when you restart Allsky, the top 50 or so lines will show all the capabilities of your camera including their minimum, maximum, and default values.  You can use `head -50 /var/log/allsky.log` to view the top 50 lines.  Change to another number to view those lines.

EricClaeys, 2022-03-09:
@DamoclesSword, The timelapse, keogram, and startrails should generate automatically at the end of night, assuming you have `TIMELAPSE="true"`, `KEOGRAM="true"`, and "STARTRAILS="true"` in allsky/config/config.sh.
If they aren't there is a problem.  If you haven't increased your swap space that's likely the cause.  There's a Wiki page for increasing swap under the "Troubleshooting" section.

DamoclesSword, 2022-03-09:
When I type nano /var/log/allsky.log, it pops up an error message saying it is unwritable.  I then exited out of that and re-ssh into the pi and typed 

cd
cp /var/log/allsky.log   allsky.log.txt

Which nothing popped up. If I dont have VNC setup, assuming we figure out this log issue, how would I get this file off of the Pi? 

As for the timelapse command, you are right, I missed the capital F before. 

As for that second message, I have all the features enable, but there's nothing on GUI. I ordered a Pi 4 so I might just wait for that to come in before worrying about the file generation. I do want to figure out the logs though.

EricClaeys, 2022-03-09:
@DamoclesSword   After you copy a file with the `cp` command, nothing should show up except your prompt.  If there's an error it will display a message.  If you do `ls -l allsky.log.txt` it will display information on the file, like the permissions, owner, group, date, and size.

By default the log file is not readable by the "pi" login.  As soon as I installed the software for the first time I did this:
~~~
sudo chgrp pi /var/log/allsky.log
sudo chmod 644 /var/log/allsky.log
~~~
That makes the file readable and writeable by the "pi" login.

From your PC you can try using `scp` (secure copy) to copy files from the Pi to your PC, or vice versa. Try
~~~
scp pi@allsky:allsky.log.txt  .
~~~
That _should_ copy the file to the current directory on your PC.

EricClaeys, 2022-03-11:
@DamoclesSword You aren't being dumb - you are inexperienced with Linux, which is totally different.  Every experienced Linux user was inexperienced at one time.

I agree I am probably assuming things you don't know.  I'll try to improve there.
The Wiki does assume basic knowledge and I will likely keep it that way since the pages will get really complex if they each have to explain the basics.  However, I just added an outline of a [Linux and Pi Basics](https://github.com/thomasjacquin/allsky/wiki/Linux-and-Pi-Basics) Wiki page for people like yourself.  When I have an initial version I'd appreciate your feedback.

This post is somewhat long since I included explanations that I hope will help you.  However, if you prefer I not do that in the future, let me know.

Some Linux basics:
Commands are executed by typing their name and optionally adding arguments, then pressing the "Enter" key.  A simple example: typing `echo` will output a blank line. Typing `echo hello world` will output "hello world".  Typing `scp` will produce the usage message you saw because the `scp` command expects at least two arguments and none were given.  If you want to learn about a command you can type `man scp` to get the "man"ual page for the `scp` command.  That will display the usage as well as a description of each argument; most manual pages are multiple pages.  In this example:
~~~
usage: scp [-346ABCpqrTv] [-c cipher] [-F ssh_config] [-i identity_file]
[-J destination] [-l limit] [-o ssh_option] [-P port]
[-S program] source ... target
~~~
everything between a left and right bracket is optional, so for example, you can optionally specify port 22 by adding `-P 22`.  In the above example, everything is optional except "source" and "target".  The "source ..." means you can specify multiple source files.  Most arguments typically start with a "-" and then some letter or number, and may be followed by a second argument.  When you see something like `[-346A]` it's the same as `[-3] [-4] [-6] [-A]`.  Remember that Linux is case-sensitive so `-A` and `-a` are different.
Typically the order of the arguments doesn't matter, although with something like `-P 22` you can't enter `22 -P`.
Options that begin with a "-" usually have a single letter or number after them.  Many commands also allow options that begin with "--" and are followed by a word.  For example, the `rm` (remove) command has both `-f` and `--force` options which both force the removal of a file.

For the `scp` command, are you sure you entered `scp pi@allsky:allsky.log.txt  .` exactly like this, with a space before the ending period?  The "usage" message only appears if the command is not called correctly.  The command I gave you has two arguments: 
1. `pi@allsky:allsky.log.txt` is the "source" file location  This tells scp to log in to the "allsky" machine as the "pi" user, then grab the "allsky.log.txt" file.  I am assuming your Pi is called "allsky" and you log in as "pi", since those are the defaults.  If it can't find the "allsky" computer you'll need to enter the Pi's IP address, like `pi@192.168.0.123:allsky.log.txt`.
Just like with email addresses, the "@" separates a user name and a computer name, ala "john@google.com".  scp uses the ":" to separate the user/machine from the file to get.
3. `.` is the target directory to put the file in on the machine you run the scp command on. ("." means the current directory in Linux).

The "-J" option to scp requires a "destination".  The manual page says that option tries to use an intermediate server for the transfer, which you don't want.  The `ssh: Could not resolve hostname c:` message says it couldn't find a computer named "**c**".  Did you type something like `scp -J c:\some_file`?  If so, that would explain that message.

Our goal with the "allsky.log.txt" file is for me to see it.  If I could log into your Pi I could look myself, but that requires that your Pi be accessible via the Internet.  The next best thing is to attach it to a GitHub post.  Our difficulty is figuring out how we get it from your Pi to your PC so you can attach it to a post.  How did you add the lines from the log file in your second post?  I assume you did a copy/paste from your ssh window on your PC to your GitHub post?  That only included 10 lines of the file whereas I need to see the whole thing which will be several hundred lines.
If you can't get `scp` working, you can try the following:
* ssh into the Pi
* Make your ssh window as large as possible.
* Create a .txt file using Notepad on your PC.
* Run `more /var/log/allsky.log` on the Pi.  This will display the file one page at a time.
* Copy/paste the first page into the file on your PC.
* Press the spacebar in the ssh window to go to the next page in the log file.
* Copy/paste the second page into the end of the file on your PC.
* Keep doing a copy/paste followed by the spacebar until you get to the end of the file.
* Type "q" to exit the `more` command.
* Attach the file from your PC to your GitHub post.

Going forward, when you get errors on the Pi, please copy and paste the command you typed as well as the error, or do a screen capture, so I can see **exactly** what you typed, including spaces, etc.  This will make troubleshooting much quicker.

We'll get this working, so please hang in there with me.  I know it's frustrating now, but once you get Allsky running it'll be worth the hassle.

Eric

DamoclesSword, 2022-03-11:
@EricClaeys The way you just handled that was so kind. I am blown away at that level of support for this program! I was able to do that last step to get all the logs from the first initialization on my new Pi4 2Gb. There's some errors at the start of the log that are from that setup process where I didnt have the camera plugged it. That install did go smooth though. 

[PiLog.txt](https://github.com/thomasjacquin/allsky/files/8232508/PiLog.txt)

I know I am replying out of order, which I feel like what matters most is above. It seems that the software has yet to try to start creating a timelapse. I did already configure the GUI so all the the values that need to be set to "true" are. 

When you say ` . ` (hopefully that inserts that correctly) is a target directory, where is that location? If I type 

`scp pi@allsky:allsky.log.txt .`

Where is that file going to be located? Or do I need to replace the `.` with something? 

When I type:
`pi@allsky:~ $ scp pi@allsky:allsky.log.txt .` (which this was pasted from my command line) I get:

pi@allsky's password:
scp: allsky.log.txt: No such file or directory

So, I redid all the steps above 

> pi@allsky:~ $ cd
pi@allsky:~ $ cp /var/log/allsky.log.txt
cp: missing destination file operand after '/var/log/allsky.log.txt'
Try 'cp --help' for more information.

That did nothing. If I understand you message from the other day correctly _"The "cd" command puts you in /home/pi.
The "cp" command copies the log file to a file called "allsky.log.txt" in /home/pi. GitHub won't let you upload a file ending in ".log" so you're basically renaming the file to end in ".txt" which GitHub accepts."_ That means that when I run the cd and cp command, that file is still on the Pi correct? The only way to get it off is via VNC or something like that? 

I have tried researching the SCP command on google, but their syntax seems different that what you're saying because after where we have `pi@allsky:allsky.log.txt` they seem to add some [user@]DEST_HOST:]file2. 

I then tried the whole thing:

> pi@allsky:~ $ sudo systemctl stop allsky
pi@allsky:~ $ sudo truncate -s 0 /var/log/allsky.log
pi@allsky:~ $ scp pi@allsky:allsky.log.txt .
pi@allsky's password:
scp: allsky.log.txt: No such file or directory
pi@allsky:~ $

I know this is a lot, but I am just at a lost. I have the software running and the Pi4 seems great, I just dont understand why the timelapses and others are not being generated, and do not understand why I cannot seem to figure out these log files.

EricClaeys, 2022-03-11:
@DamoclesSword The cool thing is, I meant it.  When people think they're doing dumb stuff they often become frustrated, get pissed off, and lose interest, whereas if they think they're inexperienced, they realize they can increase their experience which gives them hope.  You were able to upload the log file which is a great first step.  I can't do much without it.
You are trying to learn a lot of things at once and I assume you are not a programmer so I'm not surprised you are having challenges.

Anyhow, so much for my philosophy.

Our first goal is to copy the log file on your Pi to your PC so you can easily attach it to your GitHub reply.  There are multiple ways to copy the file - I suggested `scp` because under the rugs it uses ssh to do the copy, and you are already using ssh so we know it works.  VNC is another way but you'll need to enable it on the Pi, then install a VNC viewer onto your PC.  That's not that hard, but requires you learn the Pi desktop.  I use a third way to copy files which is to mount my /home/pi directory onto my PC (I call it the Y: drive) so I can simply drag and drop files like I do with any other Windows file.  Setting that up is much more advanced but works great.

=== More Linux info: ===
Linux and Windows are pretty similar in concept - they just use different syntax.  Directory names in Linux are separated by `/` and Windows uses `\`.  In Linux the highest level directory (called "root") is `/`; in Windows it's "C:\".  `/home` in Linux is a subdirectory of the root, and `/home/pi` is a subdirectory of `/home`.  In Windows, `C:\users` is a subdirectory of the root, and `C:\users\eric` is a subdirectory of `C:\users`.  Linux is case-sensitive; Windows is not.  When you first login as the "pi" user on Linux, you are put in the "pi" user's home directory, which is `/home/pi`.  When I log into Windows I'm put in my home directory which is `C:\users\eric`.

`.` stands for the current directory, whatever that is.  `..` is its parent directory.
You move to different directories using `cd` (change directory).  With no arguments, `cd` puts you in the home directory. The tilde character `~` stands for the home directory, so `cd ~` also puts you in the home directory.  Typing `cd` is easier, BUT, let's say you are in `/var/log` and want to go to the `/home/pi/allsky` directory.  You could type `cd /home/pi/allsky` or `cd ~/allsky` which is easier.  I mention this because some of the commands I suggest contain a tilda.

Linux prompts you with `pi@allsky:~ $ ` which is your login (pi) followed by "@" followed by your Pi's name (allsky) followed by `:` followed by the current directory (~), so you are in pi's home directory.  (The ` $ ` is just there for historical reasons - many years ago the prompt was simply `$ `).  If you `cd allsky` from with the home directory you will be in `/home/pi/allsky` (also called `~/allsky`) so your prompt changes to `pi@allsky:~/allsky $ `.   Windows does something very similar if you use the `cmd` window.

The `cp` (copy) command has two required arguments - a source file and a destination.  When you ran `cp /var/log/allsky.log.txt` you didn't specify the destination so got the `cp: missing destination file operand after '/var/log/allsky.log.txt'` error message.  Note that had you specified a destination it still would have failed because `/var/log/allsky.log.txt` doesn't exist - the log file is `/var/log/allsky.log`.  What you wanted to execute was `cp /var/log/allsky.log allsky.log.txt`. which would put the copy of the log file in the current directory.
> Hint: when executing commands I suggest, try copying/pasting them rather than typing them - it's less error prone.

[BTW, I just found out today that GitHub DOES accept files ending in .log - sorry about that. You still want to make a copy of the log file otherwise by the time you upload it, it would contain a lot of extra lines which makes it harder for me to view.  And you still want the attached file to end in ".txt" because I often read emails on my phone, and it can't display ".log" files but CAN display ".txt" files.]

If you are in the home directory and run `scp pi@allsky:allsky.log.txt .` it will try to copy the "allsky.log.txt" file to `.` which is the current directory which is `/home/pi`.  (If you want to copy the file to somewhere else, for example, the "allsky" directory, you could replace the period with `~/allsky`, but I suggested copying it to `.` because it was less typing hence less error prone.) I'm not sure why that command failed - are you sure "allsky.log.txt" is in /home/pi? You can verify by running:
~~~
cd
ls
~~~
That will list all the files and directories in your home directory.

The `scp` command is like `cp` but is used to copy files from one machine to another.  It requires the same two arguments (source and destination).  One difference in those commands is how you specify the source and destination.  With `cp` you simply give it two files or directories on the Pi.  With `scp`, at least one of the files should be on a remote machine and you use the `login_name@machine_name:file_name` syntax, such as `pi@allsky:allsky.log.txt` - "pi" is the login_name, "allsky" is the machine name, and "allsky.log.txt" is the file_name you want to copy.    And `.` is the 2nd argument so is the destination location.  `pi@allsky` is the pi login's home director on the "allsky" computer (which is your Pi), so `scp` will look in that home directory for the "allsky.log.txt" file.  Y [... cut ...] he problem).
* Did you do a firmware upgrade on your ASI120?  The README.md file on the GitHub home page mentions this:

> **NOTE:** Owners of USB2.0 cameras such as ASI120MC and ASI120MM may need to do a [firmware upgrade](https://astronomy-imaging-camera.com/software-drivers). This changes the camera to use 512 byte packets instead of 1024 which makes it more compatible with most hardware.

I suggest looking into that, and checking for moisture in the camera, and changing the Debug Level to 4.

DamoclesSword, 2022-03-11:
@EricClaeys So, when I started following the steps from above, I didnt mix your directions with the original ones so that is why I was at the wrong log level. I will be traveling out of town today for 4 days, so I think this might all have to wait. I read your post at at a fairly high level, but will need to dig more into it later. The core of it is over my head but I still have the same root question: When run the scp command, where is it going to copy the file to? 

Assuming I have ssh into the pi, could you just type all the commands needed to do this? It is getting hard to merge the information in the wiki with the stuff I am reading from you here. 

Also yes, the camera is inside in my office for testing right now. I have already updated the firmware of my 120MC-S so all that should be in order. I will check the lat long to make sure they are correct and see what happens when I return from my work trip.

EricClaeys, 2022-03-11:
@DamoclesSword Waiting is no problem.  I expected that you might need to read my post more than once since it has a lot of details.  I will use some of what I wrote in the "Linux basics" Wiki page.

I don't mean this in a snarky way, but the scp command will copy the file to where ever the second argument points to.  If you run scp on your Pi and use `.` or anything else that doesn't have a `:` in it, like `/home/pi`, it will copy the file to your Pi.
If you run scp on your PC and the second argument is `.`, it will copy the file to your PC.
In both cases, if the second argument is `.`, the file will be copied to whatever directory you are in when you execute the command on your Pi or PC.

Kinda creepy that I can tell when your lights went out by looking at the log file, heh?
By any chance was the brightness of the office changing the first few minutes you ran the test?  If so, what would explain the weird ping pong behavior.

Be sure to also check "angle" when you check latitude and longitude.

Enjoy your work trip.

EricClaeys, 2022-03-11:
@DamoclesSword Sorry, forgot to give you the commands.

On the Pi, do this.  You should be able to copy this whole block to the clipboard on your PC, then paste it into an ssh window on your Pi:
~~~shell
sudo systemctl stop allsky
sudo truncate -s 0 /var/log/allsky.log
sudo systemctl start allsky
# let it run until you have the problem, however long that takes
sudo systemctl stop allsky
cp /var/log/allsky.log /home/pi/allsky.log.txt    # note the more explicit destination
~~~

On your PC, do this in a "cmd" window in whatever directory you want the log file to be copied to.  You can copy/paste this on your PC:
~~~
scp pi@allsky:allsky.log.txt  .
~~~
the "allsky.log.txt" file should now be in the directory you are in on the PC.  You can then use the Windows file explorer to drag and drop it to the bottom of your GitHub post.

It's good to know the camera was in the office since that explains the extremely bright images at night when it was taking long exposures with the lights on.  And you turned off the lights the first time around 11 pm when the images got dark.

BTW, when you googled the `scp` command and it showed `[user@]DEST_HOST:]file2`, there was another `[` before the first one separated by a bunch of space - that's poor, confusing formatting of the manual page.  That string is simply a more accurate way of saying `destination`, because the destination (or source or both) can be a file on the current machine or on a remote machine.  You can specify
* a destination host name followed by `:` and a file name (for a remote file: `DEST_HOST:file2`)
* a user name followed by `@` followed by a destination host name followed by `:` followed by a file name (for a remote file: `user@DEST_HOST:file2`)
* just a file name (for a local file: `file2`).

Remember that in manual pages, `[` and `]` indicate optional arguments.

DamoclesSword, 2022-03-11:
@EricClaeys 

So I ran that entire string above and it did finally do something it seems and responded with this: 

allsky.log.txt                                                                                                                                                                                                                                         100%   74   135.8KB/s   00:00

However, I still dont get where this file is supposed to be located on my PC? It clearly downloaded something but is not located anywhere that I can find

EricClaeys, 2022-03-11:
@DamoclesSword. Great - that's good progress.

Did you run `scp` on your PC in a `cmd` window?  If so, the file should be in whater directory you were in when you ran the command.

EricClaeys, 2022-03-23:
@DamoclesSword   Any update?  If the problem is resolved, please close this Issue.  Thanks.

EricClaeys, 2022-04-11:
Closing Issue - no reply from user in a month.

closed 2022-04-11 by EricClaeys