AllskyTeam/allsky #1067: Allsky start errors

SM4IVE, 2022-03-08:
I am geting several errors  like acces denided  se inkluded screen shots
have installed the latest sw versions after sd card crash  the old sw   was no problem to set up but this new stuff are worse

[image]
[image]

EricClaeys, 2022-03-08:
@SM4IVE is allsky/tmp a normal directory or did you make it a tmpfs (memory filesystem)?

Run `ls -ld /home/pi/allsky/tmp` and post the results. 
The pi login should be the owner and group owner, and it should be mode 775 (drwxrwxr-x).
If it's not, then do this:
~~~shell
cd allsky
sudo chown pi tmp
chgrp pi tmp
chmod 775 tmp
~~~
This will set up the permissions correctly.

SM4IVE, 2022-03-09:
Did that no differens
[image]

EricClaeys, 2022-03-09:
@SM4IVE    Do an `ls -l /home/pi/allsky/tmp` and post the results please.

SM4IVE, 2022-03-10:
Got some help of a friend now it starts but some times it says camera not found 
last night it stoped at 22.00 
[image]

EricClaeys, 2022-03-11:
@SM4IVE,  I got this message in email, but don’t see it in GitHub. Did you delete it?

The `ls` command should have the lowercase letter L, not the number 1:   `ls -l /home/pi/allsky/tmp`.

Eric

 

 <https://user-images.githubusercontent.com/95401143/157600361-f65ff6d7-da99-466b-8eeb-b68b73d15ff9.gif>

EricClaeys, 2022-03-11:
@SM4IVE,  I got this message in email, but don’t see it in GitHub. Did you delete it?

The `ls` command should have the lowercase letter L, not the number 1:
~~~
ls -l /home/pi/allsky/tmp
~~~
That will display the permissions, date, time, size, and other information on the files, in addition to their names.

Eric

 

From: SM4IVE  
Sent: Thursday, March 10, 2022 12:13 AM
To: thomasjacquin/allsky 
Cc: EricClaeys 
Subject: Re: [thomasjacquin/allsky] Allsky start errors (Issue #1067)

Got some help of a friend now it starts but some times it says camera not found
last night it stopped at 22.00

[image]

SM4IVE, 2022-03-13:
There is still errors. it seems to work for approx 2 hours then it crashes. i start allsky.sh in terminal not in webgui. I also noticed that if i look on the images by surfing in on ip to webgui it seems it starts a new seesion. This i am not sure about.
[image]

EricClaeys, 2022-03-13:
@SM4IVE There is a Wiki page for ASI_ERROR_TIMEOUT that will give you ideas of what to try.

Also, please see the Wiki page in Reporting Issues and upload the log file.  I would loke to see what it's doing before it stops working.

SM4IVE, 2022-03-14:
I changed the usb settings lets see if it gets better
[image]

EricClaeys, 2022-03-14:
@SM4IVE for testing you may want to turn daytime shots on, so you don't have to wait until night.

SM4IVE, 2022-03-14:
> @SM4IVE for testing you may want to turn daytime shots on, so you don't have to wait until night.

But it seems to be working now. i have other issues with allsky web s

EricClaeys, 2022-04-11:
Closing issue - reported problem has been resolved.