AllskyTeam/allsky #1069: HDR-Muli-Exposure-Image-Fusion

Bra1nsen, 2022-03-10:
Hey All,

my goal is to save images in different exposure times. for example: 

1 Image: 0.2ms
2 Image: 0.5ms
3 Image 0.8ms
4 Image 1ms
5 Image 1.5ms

by combining these images it will be possible to generate a HDR Image.

https://github.com/zhiqinzhu123/HDR-Multi-exposure-image-Fusion-Source-Code-

It would be a great feature for sure!

EricClaeys, 2022-03-11:
@Bra1nsen The ZWO and RPiHQ cameras don't support this so we'd need to implement it in software, and it would be a pretty large effort and add a lot of complexity that IMHO few people would use.  A lot of people use their cameras to look for meteors and related items, so often don't even take daytime pictures.  To be honest, even if we wanted to implement this it wouldn't be for a couple years - we have a pretty big backlog of fixes and other enhancements.

In the future, if you have a feature request or question, would you please create a new Discussion item for them (use the link above) instead of creating an Issue?  We're trying to keep the Issue list to just bugs.  Thanks.

closed 2022-03-11 by EricClaeys

Bra1nsen, 2022-03-11:
I think you underestimate the tremendous added value of exposure fusion, especially for capturing astronomical objects ;)

EricClaeys, 2022-03-11:
@Bra1nsen You goal has very short exposure which I assume was for daytime images, which aren't generally astronomical objects.  Or am I missing something?  My concern isn't so much with the added value as much as the added cost and complexity.  Creating a nighttime HDR image would take minutes and likely result in a lot of missed meteors.

I can see where it would be nice to have HDR daytime images, especially when the Sun is in the picture.  I haven't tried this, but I wonder if some of the benefit could be achieved by underexposing daytime images and then stretching them.  Allsky can currently do this.

Bra1nsen, 2022-03-11:
watch from 1:40 - 3:00 [[HDR](https://www.youtube.com/watch?v=CUWHlke-yjc&ab_channel=LUCIDVisionLabs)]

Yes, I totally agree with you, that's definitely a lot of work. but I think this is the way to go. i will keep you up to date, i will test the hdr multi exposure fusion algorithm in the next few days. Maybe there will be an elegant way to integrate the function into the allsky software. I would be happy to contribute a share, maybe to further develop the allsky software in the future.

Many greetings

EricClaeys, 2022-03-11:
@Bra1nsen Interesting,  I didn't know there's a chip with different pixel sized to do HDR on the chip.  Does ZWO sell a camera with that chip?

Bra1nsen, 2022-09-10:
Hey Eric, how are you? 
Iam still working with my skycam :)
[image]
[image]

I need your help, could you please link me the part of the allsky code, where the automatic image saving process is taking place? I would like to know, how to organigram the image saving part while using timestamps and auto creating of folders.

kind regards
Paul

Bra1nsen, 2022-09-10:
and to your question, i can recommend the sony imx490 camera, that would beat anything for sure.

EricClaeys, 2022-09-13:
@Bra1nsen  the "capture_ZWO" and "capture_RPiHQ" programs (source in the allsky/src directory) save the initial image-YYYMMDDHHMMSS.jpg file in allsky/tmp, then call the scripts/saveImage.sh script, passing the name of the file as well as information about the file, like exposure, bin, and many others.  saveImage.sh then optionally crops, resizes, stretches, and uploads the image, then makes a copy of the file called allsky/tmp/image.jpg, then moves the original image to allsky/images/DATE.  There is never a chance to overwrite a file since each file is saved with a unique name.

The "capture_*" programs keep track of the time the image starts, then use that to create the name of the image.