AllskyTeam/allsky #1072: Where are "day" and "night" defined?  Mine are backwards...

ScottTheHippy, 2022-03-11:
I've done two installs - one Pi 3 with Buster and Zero 2 W with Bullseye, and what I can't seem to find is how to say at what time night (or day) begins and ends.  Both my instances are 180 degrees off.  Am I staring this in the face?  D'urr...   Cheers!

EricClaeys, 2022-03-11:
@ScottTheHippy 
Did you take a look at the Wiki page on [Setting: allsky](https://github.com/thomasjacquin/allsky/wiki/allsky-Settings), in particular the `angle` camera setting?
There's also a detailed page on [Angle and sunrise/sunset](https://github.com/thomasjacquin/allsky/wiki/Angle-and-sunrise---sunset) that will answer your question.  It's important that your longitude is correct.

In the future, when you have a question, please add it in the Discussions link above, rather than creating an Issue.  If the question points to a bug we can at that point create an Issue.
Thanks.

If the Wiki pages resolve your problem, would you please close this Issue?

ScottTheHippy, 2022-03-13:
Howdy Eric, 

Not sure how I managed to get my coordinates all borked, twice no less, but that helped me get right.  Thank you, and for sure I will use the wiki for any future questions like this.   Cheers!

closed 2022-03-13 by ScottTheHippy

EricClaeys, 2022-03-14:
@ScottTheHippy Maybe you were moving around a lot when you entered the coordinates, so you kept changing them?  -:)

But more seriously, I'm glad you got it working and the solution was easy once you knew the problem. And glad the Wiki helped.  If you ever have feedback on the Wiki, please let me know.  I'm trying to make it really useful for people.