AllskyTeam/allsky #1078: Keogram not generated

drakula1981, 2022-03-18:
Hello,

First of all, thank you so much for this fantastic software!
As not everything is rosy in the best of all worlds, I encounter a problem during the generation of the keogram of the day!
This is the message in the /var/log/allsky.log : _Unable to read first file (/home/pi/allsky/images/); quitting_

I tried the following command right in the terminal with the same error :
'/home/pi/allsky/keogram' -s 1920x1080 -d '/home/pi/allsky/images/20220317' -e jpg -o '/home/pi/allsky/images/20220317/keogram/keogram-20220317.jpg' --rotate 0 --font-size 1.0 --font-line 1 --font-color '255 255 255'

I specify that this bug has just appeared since today only!
The timelapse and startrail worked fine for the same day
What could have gone wrong?

Here an extract of the log file while calling the endofnight script :
_Mar 18 06:42:08 HyperRedAllSkyCam allsky.sh[840]: ==========
Mar 18 06:42:08 HyperRedAllSkyCam allsky.sh[840]: === Starting daytime capture ===
Mar 18 06:42:08 HyperRedAllSkyCam allsky.sh[840]: ==========
Mar 18 06:42:08 HyperRedAllSkyCam allsky.sh[840]: Stop the allsky service to end this process.
Mar 18 06:42:08 HyperRedAllSkyCam allsky.sh[840]: STARTING EXPOSURE at: 2022-03-18 06:42:08   @ 143.03 ms (0.14 sec)
Mar 18 06:42:08 HyperRedAllSkyCam allsky.sh[840]: endOfNight.sh: ===== Posting twilight data
Mar 18 06:42:08 HyperRedAllSkyCam allsky.sh[840]: endOfNight.sh: ===== Generating Keogram
Mar 18 06:42:11 HyperRedAllSkyCam allsky.sh[840]: Unable to read first file (/home/pi/allsky/images/20220317/image-20220317112604.jpg); quitting
Mar 18 06:42:11 HyperRedAllSkyCam allsky.sh[840]: generateForDay.sh: Command Failed: '/home/pi/allsky/keogram' -s 1920x1080 -d '/home/pi/allsky/images/20220317' -e jpg -o '/home/pi/allsky/images/20220317/keogram/keogram-20220317.jpg' --rotate 0 --font-size 1.0 --font-line 1 --font-color '255 255 255'
Mar 18 06:42:11 HyperRedAllSkyCam allsky.sh[840]: endOfNight.sh: ===== Keogram complete
Mar 18 06:42:11 HyperRedAllSkyCam allsky.sh[840]: endOfNight.sh: ===== Generating Startrails_

drakula1981, 2022-03-18:
@EricClaeys Thanks for your answers, Firstly, yes in the camera settings, there were Height and Width set to 1920 and 1080. I removed them and I will come back to you

EricClaeys, 2022-03-18:
@drakula1981 When the Width and Height are set in the WebUI, a "-s <width>x<height>" argument is added to the keogram and startrails command lines, e.g., "-s 1920-x1080".  However, I believe the Width and Height settings in the WebUI only use that portion of the sensor, but don't actually change the resolution of the image - instead, they are doing a "digital zoom". 
 So the resolution of the image is the sensor's resolution, but the keogram and startrails programs are expecting a smaller image, so they ignore any image that's not 1920x1080.

Please let me know the dimensions of the image it failed on.

drakula1981, 2022-03-18:
@EricClaeys The camera is a Raspberry HQ camera so the images size are 4056x3040

EricClaeys, 2022-03-19:
@drakula1981  Are you on Bullseye or Buster?

drakula1981, 2022-03-21:
@EricClaeys I'm still on Buster

EricClaeys, 2022-04-11:
Closing Issue - no response from user in 3 weeks, and reported problem was resolved.

closed 2022-04-11 by EricClaeys