AllskyTeam/allsky #1083: TMP folder fills up SD card after 1 month

ShedAstro, 2022-03-21:
I noticed my 64gb SD card was full after 1 month of use, and I have it set to auto delete the thumbnails, and keep 2 weeks of time lapse and star trails, don’t use keogram, I deleted all the images in the images folder, and still SD card was full, then found 58gb of files in the AllSky TMP folder….why….so I emptied that and got all the space back, next day another 2000 files in it 😮😮 what does this happen, all settings as they should be….??

EricClaeys, 2022-03-21:
@ShedAstro   Please update your allsky software.  We fixed that TMP folder bug a couple months ago.
If that solves your problem, please close this Issue.

Thanks.

ShedAstro, 2022-03-21:
> @ShedAstro Please update your allsky software. We fixed that TMP folder bug a couple months ago. If that solves your problem, please close this Issue.
> 
> Thanks.

Ok, but it’s took me months to get this thing working each night, and the update procedure, is a real long winded pain, for a non Linux person like me…what can it just be an update button in the software….
Oh well looks like I will have to just delete the folder every few weeks, and re create, no biggie….thanks…

EricClaeys, 2022-03-21:
@ShedAstro   I agree that upgrades are currently a pain in the butt.  I'm working on an "upgrade.sh" script that will do a non-destructive upgrade of /home/pi/allsky (the WebUI and website already have this).  The script should do everything including back up the existing software, install all new code, then restore settings and all images.  It'll leave the old code in case something goes wrong.

ShedAstro, 2022-03-21:
> @ShedAstro I agree that upgrades are currently a pain in the butt. I'm working on an "upgrade.sh" script that will do a non-destructive upgrade of /home/pi/allsky (the WebUI and website already have [...]

I use the web Ui….I think…so how do I update with that….??

EricClaeys, 2022-04-05:
@ShedAstro   The only way to upgrade the WebUI or Allsky Website is via the install.sh scripts in ~/allsky/gui and ~/allsky/website.  Our goal is to put an "Update" button in the WebUI that will update any or all of the packages with just a single click.  It'll also check if an update is needed.

You ok with closing this Issue?

EricClaeys, 2022-04-11:
Closing Issue.  The original problem has been resolved, and the feature requested added to our "to do" list.

closed 2022-04-11 by EricClaeys