AllskyTeam/allsky #1085: Log Files

maserowik, 2022-03-22:
I noticed that that the allsky.log file does not clear out on my system.  Is there a setting I need to activate to have the log file clear out.  
[image]

I also noticed it in the lightttpd folder
[image]

Granted the files are small.

EricClaeys, 2022-03-22:
@maserowik Are there other .gz files in /var/log?  On my system there are so I assume that's how Linux works and it's not related to Allsky.

@linuxkidd. Michael, comments?

jimjackii, 2022-03-23:
@maserowik 
That's an Linux mechanism. It's the logrotate function.

EricClaeys, 2022-03-23:
@jimjackii   Is there a limit to the number of old files it keeps?  If so, where is that limit set?

jimjackii, 2022-03-23:
@EricClaeys 
There is a config file for logrotate -> /etc/logrotate.conf.
It rotates "weekly" and you can config how many old files will be saved with the parameter "rotate x".
4 means it saves the current logfile and 4 old logfiles.

maserowik, 2022-03-23:
I am going to change on mine. Thanks for the information.

EricClaeys, 2022-03-23:
@maserowik   Are you going to change lighttpd or allsky's number of weeks?
12 for lighttpd seems way to high.  They could probably both be 2.

maserowik, 2022-03-24:
> @maserowik Are you going to change lighttpd or allsky's number of weeks? 12 for lighttpd seems way to high. They could probably both be 2.

I set to two weeks.
Mike

EricClaeys, 2022-03-24:
@maserowik   Mike, I'd like to keep this open until I submit a change. I'll also set both to 2 weeks.
Eric

EricClaeys, 2022-03-24:
Closing - just created PR 1090 to change allsky and lighttpd to save only 2 weeks of logs.

closed 2022-03-24 by EricClaeys