AllskyTeam/allsky #1089: 2x2 binning not working with rpi HQ camera?

pootle, 2022-03-24:
I've set 2x2 binning in the web interface, but the images saved are still full sensor size.

I then also tried setting resize in config.sh to get half size, but all the values I have tried just give a message that the resize failed and the image is not saved (running allsky from a shell prompt).

I'm running on an rpi zero2 W and otherwise it all works well - I also ran on a pi3B which was fine, but uses a lot more power!

EricClaeys, 2022-03-24:
@pootle Please follow the instructions on the Wiki page for Reporting Problems, and I will take a look at your log file.

pootle, 2022-03-25:
Here's a log file with the resize failing see from around line 90
tia
[alog.log](https://github.com/thomasjacquin/allsky/files/8348969/alog.log)

pootle, 2022-03-25:
I notice in the log file it reports separate entries for binning at night and by day, but I cannot see a way to set daytime binning in the config file or the webui. I can see that the nighttime images are binned as requested, but once it switches to day mode the image are full size.
`Mar 25 08:48:19 allsky allsky.sh[1680]:  Binning (day): 1
Mar 25 08:48:19 allsky allsky.sh[1680]:  Binning (night): 2`
How can I set binning for daytime?

EricClaeys, 2022-03-25:
@pootle On the "Camera Settings" page of the WebUI, go to the bottom and click on the "Show advanced options".  You'll then see day binning and other settings that are not used very often, hence considered "advanced".

EricClaeys, 2022-03-25:
@pootle Would you please attach your ~/allsky/config/config.sh file?  You'll need to append ".txt" to the name. And attach ~/allsky/images/20220325/image-20220325084819.jpg, with a ".txt" appended to its name?

I think some setting is incorrect.

pootle, 2022-03-25:
I had tinkered with the config file since I posted that, so I have rerun to get a consistent set of files....
[allsky.log](https://github.com/thomasjacquin/allsky/files/8349734/allsky.log)
[config.sh.txt](https://github.com/thomasjacquin/allsky/files/8349741/config.sh.txt)
[image-20220325104721.jpg.txt](https://github.com/thomasjacquin/allsky/files/8349749/image-20220325104721.jpg.txt)

pootle, 2022-03-25:
It looks like the problem was caused by day and night binning being different.
I have just run with both binning set to 1 and resize set to 1/2 the image size, and then crop to set the picture square, and all is working properly.

I have also run with both binning set to 2, no resize, but still with a crop and that works OK as well,

You can close the issue if you want.

EricClaeys, 2022-03-25:
@pootle I'm not sure how the problems in the log file can be due to different bin sizes since only one bin is used at a time when taking pictures.   Keogram and startrails creation will fail with different bins, but that's because they don't support having images with different resolutions. 

Can you try going back to different bins but remove the comment on the `REMOVE_BAD_IMAGES_THRESHOLD_HIGH` line?  There's an error occurring while checking for bad images (the "syntax error" messages).

pootle, 2022-03-26:
retrieved the config file uploaded a couple of posts ago and amended, set daytime bin back to 1 and enabled daytime image save. Here's the result:
[last_allsky.log](https://github.com/thomasjacquin/allsky/files/8354928/last_allsky.log)
[config.sh.txt](https://github.com/thomasjacquin/allsky/files/8354929/config.sh.txt)
[image-20220326074447.jpg.txt](https://github.com/thomasjacquin/allsky/files/8354932/image-20220326074447.jpg.txt)

EricClaeys, 2022-03-27:
@pootle The gray-ish picture is most likely caused by the **Brightness** being too high.
On Bullseye I believe valid values are -1.0 to 1.0, with 0 being the default.  You have it set to 50, which is the default on Buster.  **Please ignore the values listed in the WebUI** - they are for Buster and need to be updated.  Try setting it to 0.
What the Brightness does is add a set amount to every pixel's brightness.  Pixels usually range in brightness from 0 (pure black) to 255 (pure white).  If you notice what should normally be black where there is no image in your picture is gray - that's because instead of being 0, it's 0 plus some value, such as 50.  And 50 is a light gray.

You also have **Saturation** (may be called Gamma) at 50, but on Bullseye it ranges from 0 to 15.99, with a default of 1.  Try setting it to 1.

Did you turn off **Notification Images**?  If so, I'm curious why?

At the top of the log file is a warning message about not being able to set the locale to en_US.UTF-8.  Can you run the `locale` command in a terminal window and let me know what it returns?

Would you please copy the attached file to your Pi in /home/pi, then execute:
~~~shell
cd
mv removeBadImage.sh.txt  removeBadImages.sh
chmod 755 removeBadImages.sh
mv removeBadImages.sh allsky/scripts
~~~
Then let allsky run for a couple minutes and attach the last 100 or so lines from the log file?  I want to see if the new file fixed the errors from removeBadImages.sh.

For the "IMG_RESIZE failed" message, you may need to increase your swap space. See [this page](https://github.com/thomasjacquin/allsky/wiki/Increase-swap-space).

[removeBadImages.sh.txt](https://github.com/thomasjacquin/allsky/files/8356746/removeBadImages.sh.txt)

pootle, 2022-03-28:
Here's the results of locale - just about to run the script.......

`pi@allsky:~ $ locale
LANG=en_GB.UTF-8
LANGUAGE=
LC_CTYPE="en_GB.UTF-8"
LC_NUMERIC="en_GB.UTF-8"
LC_TIME="en_GB.UTF-8"
LC_COLLATE="en_GB.UTF-8"
LC_MONETARY="en_GB.UTF-8"
LC_MESSAGES="en_GB.UTF-8"
LC_PAPER="en_GB.UTF-8"
LC_NAME="en_GB.UTF-8"
LC_ADDRESS="en_GB.UTF-8"
LC_TELEPHONE="en_GB.UTF-8"
LC_MEASUREMENT="en_GB.UTF-8"
LC_IDENTIFICATION="en_GB.UTF-8"
LC_ALL=
`

pootle, 2022-03-28:
And here's the tail of log file

btw just to say with the brightness and saturation changes it is working much better!  Wasn't too bothered as I mostly just run it for the night time stuff, but with the better day time images I may keep it running daytime as well.

The only big thing I would like to improve is the liveview - it is really very slow and lags behind real time by several seconds. This makes it impractical to use for focus adjustment. With the old camera stuff I actually ran a simple web service based on one of the examples in libcamera to run an mjpeg stream and that could get the lag down to around 0.2 seconds which is ok for focussing. II'm not sure if the new camera software can do that sort of thing yet though. Shall I raise an enhancement request for that?
Log file tail attached....
Thanks for all your work and help on this.
[allsky.log](https://github.com/thomasjacquin/allsky/files/8362115/allsky.log)

pootle, 2022-03-28:
Oh yes - ram and swap - I'll try that later. I'm running on a pi zero 2, so ram is limited. 
Might work better on the headless raspberry pi build as well. 
I find that the script to make keograms, star trails and timelapse fails in the morning, but if I stop allsky running and run the script from shell command then it works fine.

I run it overnight each night on the pi zero 2 with a 5000mAh power bank, and there is plenty left in the morning. May rig it up with poe now daytime looks so much better.

EricClaeys, 2022-03-28:
@pootle If you haven't increased swap size that is most likely the reason startrails, keogram, and timelapse are failing.  You can look in the log file for messages, and look at the files in allsky/tmp for messages.

Don't enter an enhancement. It takes a couple seconds to launch a new libcamera-still for every picture, so we are limited in how quickly we can update pictures.

Are you running the Allsky Website on your Pi?

EricClaeys, 2022-03-30:
@pootle   Try changing the `Locale` setting in the WebUI to `en_GB.UTF-8`.  Then see if you still get a warning in the log file about not being able to set the locale.

EricClaeys, 2022-04-06:
@pootle    The warning in the log file about `Locale` is a bug.  It tries to use the en_US.UTF-8, then looks at what the user set, rather than the other way around.  It's fixed in the next release.

Any update on your issue(s)?  Can we close this ticket?

pootle, 2022-04-07:
Yep, you've done a great job. Thanks for all your help. The Locale setting seems to have fixed that final glitch.

closed 2022-04-07 by pootle