AllskyTeam/allsky #1094: Missing stars!

patdev1, 2022-03-25:
I have been running the allsky software on a Pi 4 and RPiHQ camera for some months.  A few weeks ago my PI crashed and I needed to do complete reinstall OS and Allsky. Since then I have unable to get any decent night time shots with stars. All the processes are running fine, images, timelapse, startrails etc. Just virtually no stars. This worked fine before moving to the latest release allsky  (0.8.3) and OS (bullseye).  I have tried every combination (I think) of exposure, gain, AWB etc but to no avail. Any suggestions I can try?

patdev1, 2022-03-25:
Thanks for the prompt response. 

As requested 

commit 7c9afca43bb9615c7d3247584a38de95b0770676
Merge: a9ab316 23ff18f
Author: Michael J. Kidd <linuxkidd@users.noreply.github.com>
Date:   Sun Mar 6 12:40:01 2022 -0800

    Merge pull request #1043 from thomasjacquin/keogram.cpp-minor-changes

    Keogram.cpp misc changes

commit a9ab316168c9f6a1cd925cfd0a5c087cea8cbcc
[config.sh.txt](https://github.com/thomasjacquin/allsky/files/8354229/config.sh.txt)
[allsky.log.txt](https://github.com/thomasjacquin/allsky/files/8354242/allsky.log.txt)
[image]
3

EricClaeys, 2022-03-25:
@patdev1 Thanks for the files.
When I zoom in I see some stars, but not many, and see black circles around other stars.  libcamera does some noise reduction that may be causing this problem.  First try setting the Brightness to 0.

If you want to play around with the settings, it might be quicker to stop allsky, then run the libcamera-still command from the command line, like:
~~~shell
export LIBCAMERA_LOG_LEVELS='ERROR,FATAL'     # only needed once
# Run this command, changing gain, shutter, and brightness:
libcamera-still --output /home/pi/allsky/tmp/image.jpg --timeout 1 --nopreview --width 2028 --height 1520 --shutter 60000000 --gain 4 --awbgains 2.5,2.5 --vflip --saturation 1 --brightness 0.25 --quality 95
~~~
"gain" is from 1 - 16.
This will keep overwriting the same file which should appear on the WebUI and/or website if you have it on your Pi.
Also, try different de-noise routines.  Add `--denoise xxx`, where "xxx" is one of:
* **auto** - This is the default. It always enables standard spatial denoise. It uses high quality colour denoise.
* **off** - Disables spatial and colour denoise.
* **cdn_off** - Disables colour denoise.
* **cdn_fast** - Uses fast color denoise.
* **cdn_hq** - Uses high quality colour denoise.

See https://www.raspberrypi.com/documentation/accessories/camera.html for more info.

On a separate note, your `ftp-settings.sh` file is causing an error:  `/home/pi/allsky/config/ftp-settings.sh: line 103: hi: command not found`.  It should be easy to fix.

patdev1, 2022-03-26:
Thanks.  Will give it a go the next clear night at report the results.

patdev1, 2022-03-31:
Looks like setting brightness to 0 has helped although not back to my previous results. Adjusting the denoise setting did not make much difference. I will continue to try various exposure,gain, etc settings to see if get an improvement. 

Should binning help?

Attached image from earlier this year (pre crash) and current best.
current  best 
 [image]

Pre crash 
[image]

EricClaeys, 2022-04-01:
@patdev1 Binning shouldn't matter, but you never know.
It's interesting to see the difference in the white building near the bottom - in the current best, you can see detail on the building but in the pre-crash it's a washed out light gray, as are the trees next to it.

Try changing the contrast to see if that helps:
~~~
CAPTURE_EXTRA_PARAMETERS="--contrast 1.5"
~~~
in config.sh.  The minimum is 0, default is 1, and max is 15.99.

Also, did you stretch your images before?  There are some STRETCH options in config.sh you can try.

patdev1, 2022-04-02:
Thanks.
I will give that a go.  We do not get that many clear nights so may be a few days before I can report back.
Regards
Pat

EricClaeys, 2022-04-06:
@patdev1   Pat, any update?

patdev1, 2022-04-08:
Cloud free nights have been in rather short supply over the last week or so and as a result testing has been limited.

I have set parameters in the "capture_extra_parameters" variable in config.sh.  The effect seems limited.  I do not seem to see these reflected in the allsky.log file (example below) - debug level was set to 4. Is this expected?

***@***.***
Apr  7 05:21:22 allsky allsky.sh[24171]: DEBUG: saveImage.sh NIGHT /home/pi/allsky/tmp/image-20220407052019.jpg EXPOSURE_US=60000000 AUTOEXPOSURE=0 AUTOGAIN=0 AUTOWB=0 WBR=2.500000 WBB=3 GAIN=120 GAINDB=4.00 BIN=1 FLIP=2 BIT_DEPTH=8
Apr  7 05:21:22 allsky allsky.sh[31824]: STARTING EXPOSURE at: 2022-04-07 05:21:22   @ 60.0 sec
Apr  7 05:21:22 allsky allsky.sh[31824]:  > Kill command: pgrep 'libcamera-still' | xargs kill -9 2> /dev/null
Apr  7 05:21:22 allsky allsky.sh[31824]:   > Capture command: libcamera-still --output '/home/pi/allsky/tmp/image-20220407052122.jpg' --timeout 10 --nopreview --width 4056 --height 3040 --shutter 60000000 --gain 4 --awbgains 2.5,3 --vflip --saturation 1 --brightness -0.01 --quality 95 > /tmp/capture_RPiHQ_debug.txt 2>&1
Apr  7 05:21:24 allsky allsky.sh[24171]: saveImage.sh: Stretching '/home/pi/allsky/tmp/image-20220407052019.jpg' by 10
Apr  7 05:21:27 allsky allsky.sh[24228]: upload.sh: FTP '/home/pi/allsky/tmp/image-20220407052019.jpg' to 'allsky/image.jpg', TEMP_NAME=SaveImage-28203
Apr  7 05:21:32 allsky allsky.sh[24228]: upload.sh: FTP '/home/pi/allsky/tmp/image-20220407052019.jpg' finished
Apr  7 05:22:25 allsky allsky.sh[31824]:   > Saving NIGHT image 'image-20220407052122.jpg'
Apr  7 05:22:25 allsky allsky.sh[31824]: Sleeping 0.0 seconds...

I also notice that the stars in the timelapse video are not as clear as in any individual frame. Are there any parameters that I can add that may affect that?

Regards
Pat

EricClaeys, 2022-04-09:
@patdev1   Pat, my bad.  Your entries in CAPTURE_EXTRA_PARAMETERS will NOT be applied or appear in the log file.  Allsky doesn't (yet) know about the "contrast" option.  Sorry about that.

What I suggest is to stop Allsky, then manually run the `libcamera-still` command with various arguments.  For example:
~~~
libcamera-still --output '/tmp/test.jpg' --timeout 10 --nopreview --width 4056 --height 3040 --shutter 60000000 --gain 4 --awbgains 2.5,3 --vflip --saturation 1 --brightness -0.01 --quality 95 > /tmp/capture_RPiHQ_debug.txt 2>&1
~~~
You can then add "--contrast 1.5" or any other argument that `libcamera-still` takes.
To make it easy to make changes, do the following:

1. create a file called, for example, `expose.sh`
2. `chmod 755 expose.sh`
3. open expose.sh in an editor
4. add `#!/bin/bash` as the first line
5. add the libcamera-still line above as the 2nd line.
6. run the command:  `./expose.sh`
7. look at the image in `/tmp/test.jpg`
8. in the editor, change arguments
9. rerun the command:  `./expose.sh`
10. look at the picture again
11. repeat steps 8-10

patdev1, 2022-04-10:
Thanks for the script suggestion.
I have tried several combinations. Varying contrast does not seem to make a big impact. At the moment the setting below seem to get the best results:
Saturation = 0.000000
Contrast = 5.000000
Brightness = 0.000000
ColourGains = [ 1.500000, 2.000000 ]
AwbMode = 0
Sharpness = 1.000000
AnalogueGain = 4.000000
ExposureTime = 30000000

I will continue to adjust settings over the next few clear nights, I would like to be able to get closer to what I had and a little more colour.

EricClaeys, 2022-04-10:
@patdev1   I'm glad you are making progress. I wish there was more I could do to help.  libcamera apparently is somewhat different than raspistill, and I haven't taken the time to play around with settings (too busy making code changes).

Since your stars are mostly back and I will be of limited use going forward, are you ok if we close this issue?

patdev1, 2022-04-10:
Yes fine to close.

EricClaeys, 2022-04-11:
Closing with agreement of user.  His libcamera images are close to the older raspistill images, but he's still playing with settings.

closed 2022-04-11 by EricClaeys

patdev1, 2022-10-11:
Hi

Cloud free nights have been in rather short supply over the last week or so and as a result testing has been limited.

I have set parameters in the "capture_extra_parameters" variable in config.sh.  The effect seems limited.  I do not seem to see these reflected in the allsky.log file (example below) - debug level was set to 4. Is this expected?

***@***.***
Apr  7 05:21:22 allsky allsky.sh[24171]: DEBUG: saveImage.sh NIGHT /home/pi/allsky/tmp/image-20220407052019.jpg EXPOSURE_US=60000000 AUTOEXPOSURE=0 AUTOGAIN=0 AUTOWB=0 WBR=2.500000 WBB=3 GAIN=120 GAINDB=4.00 BIN=1 FLIP=2 BIT_DEPTH=8
Apr  7 05:21:22 allsky allsky.sh[31824]: STARTING EXPOSURE at: 2022-04-07 05:21:22   @ 60.0 sec
Apr  7 05:21:22 allsky allsky.sh[31824]:  > Kill command: pgrep 'libcamera-still' | xargs kill -9 2> /dev/null
Apr  7 05:21:22 allsky allsky.sh[31824]:   > Capture command: libcamera-still --output '/home/pi/allsky/tmp/image-20220407052122.jpg' --timeout 10 --nopreview --width 4056 --height 3040 --shutter 60000000 --gain 4 --awbgains 2.5,3 --vflip --saturation 1 --brightness -0.01 --quality 95 > /tmp/capture_RPiHQ_debug.txt 2>&1
Apr  7 05:21:24 allsky allsky.sh[24171]: saveImage.sh: Stretching '/home/pi/allsky/tmp/image-20220407052019.jpg' by 10
Apr  7 05:21:27 allsky allsky.sh[24228]: upload.sh: FTP '/home/pi/allsky/tmp/image-20220407052019.jpg' to 'allsky/image.jpg', TEMP_NAME=SaveImage-28203
Apr  7 05:21:32 allsky allsky.sh[24228]: upload.sh: FTP '/home/pi/allsky/tmp/image-20220407052019.jpg' finished
Apr  7 05:22:25 allsky allsky.sh[31824]:   > Saving NIGHT image 'image-20220407052122.jpg'
Apr  7 05:22:25 allsky allsky.sh[31824]: Sleeping 0.0 seconds...

Regards
Pat