AllskyTeam/allsky #1100: Keogram by part compressed or expanded

si6pack, 2022-03-26:
Hi,
First of all I want to thank all contributors for this amazing piece of work - very much appreciated!

What's my problem:
Allsky is running on two pi's - Pi3 ModelB Rev.1.2 V10 buster and Pi4 ModelB Rev.1.1 V10 buster
Startrails and timelapse work flawlessly, however, keogram gets truncated in a way that parts are compressed,, while others are expanded - pls see attached .jpg
[image]
 This holds true for both pi's, which are located at different sides.
I tried to exclude on one pi bad pictures while leaving this property untouched on the other - no difference.
As requested by Wiki, I generated an allsky.log file (debug level 4) during manual keogram generation, however, there is no mention about keogram execution Which was successful, but truncated/expended)...
[allsky.log.txt](https://github.com/thomasjacquin/allsky/files/8355190/allsky.log.txt)
I also visited the image overlay settings and changed the font size and color - no effect...
Any ideas what the problem might be?
Additional info: My pi3 was running for 2 years with the "old" software - no issues with the keogram. However, this problem popped up, when I upgraded according the wiki...
I have no clue where to start to get rid of this issue...
Regards,
Christian

AndreasLMeg, 2022-03-26:
@si6pack 
Hello Christian, 
I think the problem is fixed with this pull request: https://github.com/thomasjacquin/allsky/pull/1091
This was only a problem with the annotation (gridline) 

Do you know how to get the latest version of keogram.cpp, and how the compile a new executeable ?

regards
Andreas

si6pack, 2022-03-26:
@Andreas
Hi Andreas,
That‘s great news, however, I must admit, I am more or less a dump user…I would very much appreciate if you could guide me or post a link which I could use as a how-to…
Thank you in advance!
Regards,
Christian

AndreasLMeg, 2022-03-26:
Hello Christian,
Please do everything step by step and if you have difficulties, stop the procedure and get in touch with me.
To avoid complications clone allsky again in the directory allsky_christian:
```
cd
git clone --recursive https://github.com/thomasjacquin/allsky.git allsky_christian
```
Then switch to src and create keogram
```
cd allsky_christian/src/
make clean keogram

``` 
then we do a test run of keogram
` ./keogram --directory ~/allsky/images/20220324/ --extension jpg -o keogram.jpg -c -x -v -v`

This will take a while and you will see how all the individual images are processed one after the other...
```
pi@allsky:~/allsky_christian/src $ ./keogram --directory ~/allsky/images/20220324/ --extension jpg -o keogram.jpg -c -x -v -v
Getting nchan and/or size from: '/home/pi/allsky/images/20220324//image-20220324171540.jpg'
        nchan = 3
        size = 2028 x 1520
[  1/788] /home/pi/allsky/images/20220324//image-20220324171540.jpg, channels=3
[395/788] /home/pi/allsky/images/20220324//image-20220325000956.jpg, channels=3
[592/788] /home/pi/allsky/images/20220324//image-20220325033738.jpg, channels=3
[198/788] /home/pi/allsky/images/20220324//image-20220324204157.jpg, channels=3
[  2/788] /home/pi/allsky/images/20220324//image-20220324171643.jpg, channels=3
[396/788] /home/pi/allsky/images/20220324//image-20220325001059.jpg, channels=3
[593/788] /home/pi/allsky/images/20220324//image-20220325033841.jpg, channels=3
[199/788] /home/pi/allsky/images/20220324//image-20220324204301.jpg, channels=3
...
```
In the directory "allsky christian/src/" you will now find the new Keogram: keogram.jpg
Please load this file onto a Windows computer and check the image file

if everything is ok, we still have to make sure that the right files are in the allsky directory:
```
cp keogram.cpp ~/allsky/src/
cp keogram ~/allsky/

```
now we switch to the allsky directory and generate the "old" keograms
```
cd ~/allsky/
./scripts/generateForDay.sh -k 20220324
./scripts/uploadForDay.sh -k 20220324  # if you want
```
You can now control the keograms in the browser.
At the end we delete the directory "allsky_christian"
`rm -rf ~/allsky_christian/`

EricClaeys, 2022-03-26:
@si6pack. Thanks for looking at the Wiki.  If you have any suggestions for improving it, please let me know.

The allsky.log file only contains output from when Allsky is run automatically.  If you run any commands from a terminal window, their output goes to the terminal window, which is why you didn't see anything in the log file when you manually created the keogram.

The image overlay settings in the WebUI only apply to images created directly by the camera, i.e., the ones constantly being created.  The text on the keogram is controlled by the KEOGRAM_EXTRA_PARAMETERS setting in `config/config.sh`.  I know that is confusing but we hope to fix that by putting all settings for everything in one place.

Let us know if @AndreasLMeg' fix worked.  If so, would you please close this issue?
Thanks - Eric

si6pack, 2022-03-27:
Hi Andreas,

 

Thank you very much for your efforts guiding me through the process of cloning, generating and testing – very much appreciated!

 

It worked like a charme! Perfect with both pi’s! 8-)

 

Thank you again!

Regards,

Christian (si6pack)

si6pack, 2022-03-27:
Hi Eric,

 

Thank you for explaining and letting me know  very much appreciated!

 

If there is a spark in my brain for improvement, I will come back!

 

Will do

 

Regards,

Christian

(Issue #1100)

 

<https://github.com/thomasjacquin/allsky/issues/1100#issuecomment-1079796977
6QFTANCNFSM5RW3ZIGA> .
<https://github.com/notifications/beacon/ANRA5ILIUHFDOBR2FDZT2LDVB6QFTA5CNFS
M5RW3ZIGKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOIBOGJ4I

si6pack, 2022-03-27:
Issue solved! Thank you!

closed 2022-03-27 by si6pack