AllskyTeam/allsky #1101: Fresh git pull 3.26 @ 1700EST

maserowik, 2022-03-26:
Having problem with the allsky Admin panel.

when I go to the camera settings I do not see any options.

[image]
```
Mar 26 17:02:54 Pi-AllSky allsky.sh[662]: === next exposure=525 (multiply by 0.883) [last_exposure_us=595, maxAcceptable=134, mean=138]
Mar 26 17:02:54 Pi-AllSky allsky.sh[662]:  >xxx mean was 121 and went from 1 below min of 122 to 4 above max of 134, is now at 138; OK to set temp max to current_exposure_                                                                  us of 595
Mar 26 17:02:55 Pi-AllSky allsky.sh[662]: === next exposure=593 (multiply by 1.109) [last_exposure_us=535, minAcceptable=122, mean=121]
Mar 26 17:02:55 Pi-AllSky allsky.sh[662]:  >xxx mean was 138 and went from 4 above max of 134 to 1 below min of 122, is now at 121; should NOT set temp min to current_expo                                                                  sure_us of 535
Mar 26 17:02:56 Pi-AllSky allsky.sh[662]: === next exposure=525 (multiply by 0.883) [last_exposure_us=595, maxAcceptable=134, mean=138]
Mar 26 17:02:56 Pi-AllSky allsky.sh[662]:  >xxx mean was 121 and went from 1 below min of 122 to 4 above max of 134, is now at 138; OK to set temp max to current_exposure_                                                                  us of 595
Mar 26 17:02:56 Pi-AllSky allsky.sh[662]: === next exposure=593 (multiply by 1.109) [last_exposure_us=535, minAcceptable=122, mean=121]
Mar 26 17:02:56 Pi-AllSky allsky.sh[662]:  >xxx mean was 138 and went from 4 above max of 134 to 1 below min of 122, is now at 121; should NOT set temp min to current_expo                                                                  sure_us of 535
Mar 26 17:02:57 Pi-AllSky allsky.sh[662]: === next exposure=525 (multiply by 0.883) [last_exposure_us=595, maxAcceptable=134, mean=138]
Mar 26 17:02:57 Pi-AllSky allsky.sh[662]:  >xxx mean was 121 and went from 1 below min of 122 to 4 above max of 134, is now at 138; OK to set temp max to current_exposure_                                                                  us of 595
Mar 26 17:02:58 Pi-AllSky allsky.sh[662]: === next exposure=593 (multiply by 1.109) [last_exposure_us=535, minAcceptable=122, mean=121]
Mar 26 17:02:58 Pi-AllSky allsky.sh[662]:  >xxx mean was 138 and went from 4 above max of 134 to 1 below min of 122, is now at 121; should NOT set temp min to current_expo                                                                  sure_us of 535
Mar 26 17:02:58 Pi-AllSky allsky.sh[662]:   > Sleeping 5.0 sec from histogram exposure
Mar 26 17:02:58 Pi-AllSky allsky.sh[662]:   > Saving DAY image 'image-20220326170247.jpg'
Mar 26 17:03:03 Pi-AllSky allsky.sh[662]: STARTING EXPOSURE at: 2022-03-26 17:03:03   @ 595 us (0.595 ms)
Mar 26 17:03:04 Pi-AllSky allsky.sh[662]: === next exposure=529 (multiply by 0.889) [last_exposure_us=595, maxAcceptable=134, mean=137]
Mar 26 17:03:05 Pi-AllSky allsky.sh[662]: === next exposure=592 (multiply by 1.118) [last_exposure_us=530, minAcceptable=122, mean=120]
Mar 26 17:03:05 Pi-AllSky allsky.sh[662]:   > Sleeping 5.0 sec from histogram exposure
Mar 26 17:03:05 Pi-AllSky allsky.sh[662]:   > Saving DAY image 'image-20220326170303.jpg'
Mar 26 17:03:10 Pi-AllSky allsky.sh[662]: STARTING EXPOSURE at: 2022-03-26 17:03:10   @ 590 us (0.590 ms)
Mar 26 17:03:11 Pi-AllSky allsky.sh[662]:   > Sleeping 5.0 sec from histogram exposure
Mar 26 17:03:11 Pi-AllSky allsky.sh[662]:   > Saving DAY image 'image-20220326170310.jpg'
Mar 26 17:03:16 Pi-AllSky allsky.sh[662]: STARTING EXPOSURE at: 2022-03-26 17:03:16   @ 590 us (0.590 ms)
Mar 26 17:03:17 Pi-AllSky allsky.sh[662]: === next exposure=532 (multiply by 0.902) [last_exposure_us=590, maxAcceptable=134, mean=135]
Mar 26 17:03:17 Pi-AllSky allsky.sh[662]:   > Sleeping 5.0 sec from histogram exposure
Mar 26 17:03:17 Pi-AllSky allsky.sh[662]:   > Saving DAY image 'image-20220326170316.jpg'
Mar 26 17:03:22 Pi-AllSky allsky.sh[662]: STARTING EXPOSURE at: 2022-03-26 17:03:22   @ 530 us (0.530 ms)
Mar 26 17:03:23 Pi-AllSky allsky.sh[662]:   > Sleeping 5.0 sec from histogram exposure
Mar 26 17:03:23 Pi-AllSky allsky.sh[662]:   > Saving DAY image 'image-20220326170322.jpg'
Mar 26 17:03:28 Pi-AllSky allsky.sh[662]: STARTING EXPOSURE at: 2022-03-26 17:03:28   @ 530 us (0.530 ms)
Mar 26 17:03:29 Pi-AllSky allsky.sh[662]: === next exposure=457 (multiply by 0.864) [last_exposure_us=530, maxAcceptable=134, mean=141]
Mar 26 17:03:30 Pi-AllSky allsky.sh[662]:   > Sleeping 5.0 sec from histogram exposure
Mar 26 17:03:30 Pi-AllSky allsky.sh[662]:   > Saving DAY image 'image-20220326170328.jpg'
Mar 26 17:03:35 Pi-AllSky allsky.sh[662]: STARTING EXPOSURE at: 2022-03-26 17:03:35   @ 455 us
Mar 26 17:03:36 Pi-AllSky allsky.sh[662]:   > Sleeping 5.0 sec from histogram exposure
Mar 26 17:03:36 Pi-AllSky allsky.sh[662]:   > Saving DAY image 'image-20220326170335.jpg'
Mar 26 17:03:41 Pi-AllSky allsky.sh[662]: STARTING EXPOSURE at: 2022-03-26 17:03:41   @ 455 us
Mar 26 17:03:42 Pi-AllSky allsky.sh[662]: === next exposure=401 (multiply by 0.883) [last_exposure_us=455, maxAcceptable=134, mean=138]
Mar 26 17:03:42 Pi-AllSky allsky.sh[662]: === next exposure=458 (multiply by 1.147) [last_exposure_us=400, minAcceptable=122, mean=117]
Mar 26 17:03:43 Pi-AllSky allsky.sh[662]: === next exposure=395 (multiply by 0.870) [last_exposure_us=455, maxAcceptable=134, mean=140]
Mar 26 17:03:43 Pi-AllSky allsky.sh[662]:  >xxx mean was 117 and went from 5 below min of 122 to 6 above max of 134, is now at 140; OK to set temp max to current_exposure_                                                                  us of 455
Mar 26 17:03:43 Pi-AllSky allsky.sh[662]: === next exposure=454 (multiply by 1.137) [last_exposure_us=400, minAcceptable=122, mean=118]
Mar 26 17:03:43 Pi-AllSky allsky.sh[662]:  >xxx mean was 140 and went from 6 above max of 134 to 4 below min of 122, is now at 118; should NOT set temp min to current_expo                                                                  sure_us of 400
Mar 26 17:03:44 Pi-AllSky allsky.sh[662]: === next exposure=390 (multiply by 0.858) [last_exposure_us=455, maxAcceptable=134, mean=142]
Mar 26 17:03:44 Pi-AllSky allsky.sh[662]:  >xxx mean was 118 and went from 4 below min of 122 to 8 above max of 134, is now at 142; OK to set temp max to current_exposure_                                                                  us of 455
Mar 26 17:03:45 Pi-AllSky allsky.sh[662]: === next exposure=447 (multiply by 1.118) [last_exposure_us=400, minAcceptable=122, mean=120]
Mar 26 17:03:45 Pi-AllSky allsky.sh[662]:  >xxx mean was 142 and went from 8 above max of 134 to 2 below min of 122, is now at 120; should NOT set temp min to current_expo                                                                  sure_us of 400
Mar 26 17:03:45 Pi-AllSky allsky.sh[662]: === next exposure=376 (multiply by 0.846) [last_exposure_us=445, maxAcceptable=134, mean=144]
Mar 26 17:03:45 Pi-AllSky allsky.sh[662]:  >xxx mean was 120 and went from 2 below min of 122 to 10 above max of 134, is now at 144; OK to set temp max to current_exposure                                                                  _us of 445
Mar 26 17:03:46 Pi-AllSky allsky.sh[662]:   > Saving DAY image 'image-20220326170341.jpg'
Mar 26 17:03:46 Pi-AllSky allsky.sh[662]:   > Sleeping 5.0 sec from histogram exposure
pi@Pi-AllSky:/var/log $
```
getting this error in the follow log file
/var/log/lightttpd
```2022-03-26 17:00:16: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: display in /var/www/html/includes/camera_settings.php on line 164
2022-03-26 17:00:16: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: display in /var/www/html/includes/camera_settings.php on line 164
2022-03-26 17:00:16: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: display in /var/www/html/includes/camera_settings.php on line 164
2022-03-26 17:00:16: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: display in /var/www/html/includes/camera_settings.php on line 164
2022-03-26 17:00:16: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: display in /var/www/html/includes/camera_settings.php on line 164
2022-03-26 17:00:16: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: display in /var/www/html/includes/camera_settings.php on line 164
2022-03-26 17:00:16: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: display in /var/www/html/includes/camera_settings.
[... cut ...]
2022-03-26 17:00:16: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: display in /var/www/html/includes/camera_settings.php on line 164
2022-03-26 17:00:16: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: display in /var/www/html/includes/camera_settings.php on line 164
2022-03-26 17:00:16: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: display in /var/www/html/includes/camera_settings.php on line 164
2022-03-26 17:00:16: (mod_fastcgi.c.421) FastCGI-stderr: PHP Notice:  Undefined index: display in /var/www/html/includes/camera_settings.php on line 164
```

EricClaeys, 2022-03-26:
@maserowik  Mike, we decided to **greatly simplify** the WebUI.  I take it you don't like the new way????
Now that I'm done attempting humor, are you using the camera_settings.php file from a day or two ago?  If so, did you update the whole WebUI via `gui/install.sh`, or did you only update the one file?

EricClaeys, 2022-03-27:
@maserowik   Mike, the "Undefined index: display ....." bug above is fixed.  Please download the `includes/camera_settings.php` file from allsky-port, or reinstall the WebUI (`cd ~/allsky; sudo gui/install.sh`).
Sorry about the bug, but I'm glad you found it since you've been very good about reporting issues.

If this fixes the issue, please close this Issue.
Thanks.

maserowik, 2022-03-27:
> @maserowik Mike, we decided to **greatly simplify** the WebUI. I take it you don't like the new way???? Now that I'm done attempting humor, are you using the camera_settings.php file from a day or [...]

this is how i install the software
cd ~
git clone --recursive https://github.com/thomasjacquin/allsky.git
cd allsky
./install.sh

cd ~
cd allsky
sudo gui/install.sh
reboot

maserowik, 2022-03-27:
i did a fresh install of the gui and everything is working
Thanks Eric

maserowik, 2022-03-27:
I spoke to soon.  Now in var/log/lightttppd/error.log  ... I am seeing this
pi@Pi-AllSky:/var/log/lighttpd $ cat error.log
2022-03-27 00:00:02: (server.c.1759) logfiles cycled UID = 0 PID = 26112
2022-03-27 06:32:31: (server.c.2059) server stopped by UID = 0 PID = 1
2022-03-27 06:32:32: (server.c.1464) server started (lighttpd/1.4.53)
2022-03-27 06:32:32: (server.c.1493) WARNING: unknown config-key: compress.cache-dir (ignored)
2022-03-27 06:32:32: (server.c.1493) WARNING: unknown config-key: compress.filetype (ignored)
2022-03-27 06:33:29: (server.c.2059) server stopped by UID = 0 PID = 1
2022-03-27 06:33:49: (server.c.1464) server started (lighttpd/1.4.53)
2022-03-27 06:33:49: (server.c.1493) WARNING: unknown config-key: compress.cache-dir (ignored)
2022-03-27 06:33:49: (server.c.1493) WARNING: unknown config-key: compress.filetype (ignored)

EricClaeys, 2022-03-28:
@maserowik Would you please attach your `/etc/lighttpd/lighttpd.conf`.  You'll need to append ".txt" to the name.
Did you restart the lighttpd service at 4:32:31 or 6:33:29?  Strange that it would restart twice in a minute.

@linuxkidd Michael, any ideas?  `compress-cache-dir` and `compress.filetype` have been in lighttpd.conf forever.  What version of lighttpd do you have?

maserowik, 2022-03-28:
> @maserowik Would you please attach your `/etc/lighttpd/lighttpd.conf`. You'll need to append ".txt" to the name. Did you restart the lighttpd service at 4:32:31 or 6:33:29? Strange that it would [...]

here is the file.

[lighttpd.conf.txt](https://github.com/thomasjacquin/allsky/files/8362391/lighttpd.conf.txt)

EricClaeys, 2022-03-29:
@maserowik Mike, these errors in /var/log/lighttpd/error.log are harmless:
> 022-03-27 06:32:32: (server.c.1493) WARNING: unknown config-key: compress.cache-dir (ignored)
2022-03-27 06:32:32: (server.c.1493) WARNING: unknown config-key: compress.filetype (ignored)

but to get rid of them:
~~~
sudo truncate -s 0 /var/log/lighttpd/error.log
cd /etc/lighttpd
sudo YOUR_FAVORITE_EDITOR lighttpd.conf
# comment out the "compress.cache-dir" and "compress.filetype" lines around line 53.
sudo systemctl restart lighttpd
~~~
The `/var/log/lighttpd/error.log` file should now only have a couple entries about stopping / starting the service.

I put in a change request to fix this.

If the problems above are fixed, please close this Issue.  Thanks - Eric

maserowik, 2022-04-03:
did a new git pull and now it is working 
Thanks

closed 2022-04-03 by maserowik