AllskyTeam/allsky #1102: Can't install via portainer

OrionA51, 2022-03-26:
I have had my rpi running pihole for a long time and when I first installed allysky it overwrote the lighttpd.conf file and I could no longer access pihole through the webui. I reinstalled pihole and then could not access the webui for allsky. I am a newb to this stuff so I have installed pihole via portainer and is now running in a container using host port 1010 and port 80 on the other side. I would like to install allsky using portainer, but my attempts have failed due to errors when I would try to deploy the image after creating the custom template from the files. It did not matter if I used the zip file or the pull from github. One of the errors I got was because of the compose path docker-compose.yml (in portainer) as I just left that at default since I don't know what to input there.

EricClaeys, 2022-03-27:
@OrionA51. I never heard of Pihole or portainer.  What do they do?  Do you have, or can you recreate the lighttpd.conf file from before you installed Allsky' WebUI?  I think the Allsky changes to it are fairly straightforward so I might be able to create you a merged version.

Us developers talked about whether or not the WebUI installation script should check if there is already a web server on the Pi.  This is probably a good example of why we should check.

OrionA51, 2022-03-27:
Pi-hole is an ad blocker. Very common. Portainer is a webui that has something to do with docker and running programs in containers. Portainer also makes it easy to install programs (like Pi-hole) with one click. Now that Pi-hole runs in the container, it no longer uses the lighttpd.conf file that it used before. It launches either through another file location or through Portainer. Im not really sure how it all works. I was just hoping to run both programs with a container to make it easier. I can try to install allsky again and see what happens. I let you know. Thanks

EricClaeys, 2022-03-27:
@OrionA51   Thanks. Please try installing allsky.  If it runs on the Pi or in a separate container from Pi-hole, I'd think it would be fine.

OrionA51, 2022-03-27:
After installing I did the reboot. It now is locked up. I had to change the dns on my pc to get back online. I think it's because Pi-hole uses port 80. Ill probably need to change the port that allsky uses. Just for reference, this is what I would like to do except with allsky - https://rsw.io/how-to-install-pi-hole-in-a-docker-container-on-a-raspberry-pi/. Id like to create a docker-compose.yml to place it in a container. I might then be able to use the docker-compose path that portainer is looking for when creating a new template in portainer.

EricClaeys, 2022-03-27:
@OrionA51 What locked up?  The Pi, or the WebUI or Pi-hole?
It should be easy to change Allsky's port - it's one line in the lighttpd.conf file, then "sudo systemctl restart lighttpd".  The hardest part will be to remember to use that port in the URL.

OrionA51, 2022-03-27:
Sorry. The Pi booted in recovery mode. The data lights on the ethernet port on the pi are now solid on even without the cord plugged to the port. I tried changing the port in the lighttpd file and the running the install again and nothing happened. So I just did an uninstall and the pi still boots to recovery mode without any network. I can access portainer still and access Pi-hole via 0.0.0.0:1010/admin. Those programs are still running. I don't know how to get the Pi out of recovery mode now.

EricClaeys, 2022-03-27:
@OrionA51 Sorry, I have no idea how to get out of recovery mode, but I'm sure Google does.

OrionA51, 2022-03-27:
I’ve spent the last 3 days straight trying to figure all this out. I don’t have a clue now how to get my network back on the pi after installing allsky. At the bottom of the terminal where the network icon is it says connection to dhcpcd lost. Googling this issue is hard because I know what caused it, but looking for that specific reference in a google search just brings up basic issues related to no network. In the terminal I was able to install the GUI for allsky with cmd promt and I changed the port to 81, but cannot open the webui. I tried ….local:81 and 0.0.0.0:81 and neither work. My other programs still work and can open them using their respective ports in the webui. I will continue to try to get my network connection working again on the pi, but I’m still trying to figure out how to access the webui for allsky with that new port. 

Ryan C.

EricClaeys, 2022-03-27:
@OrionA51 What is the Pi' IP address?  It's probably not 0.0.0.0.
You can try http://IP_ADDRESS:81 where IP_ADDRESS is the IP ADDRESS, or http://127.0.0.1:81

OrionA51, 2022-03-27:
From within the Pi I have to use 0.0.0.0:PORT right now. I can’t use the IP address of the Pi anymore because of the lost network. Networking is currently disabled which is what I am trying to fix. I can’t access the Pi from another location so I am working directly in the Pi with a monitor connected. 

Ryan C.

EricClaeys, 2022-03-27:
@OrionA51 Interesting, I never saw 0.0.0.0 before.  Allsky doesn't do anything to the network other than set the lighttpd port number.

OrionA51, 2022-03-27:
I’m going to reinstall the OS since I had no luck recreating the eth0 file to get the network drivers to work. I can change the port in the GUI/lighttpd.conf before running the install and that should work, right? I figure I can just make it port 8080.

Ryan C.

OrionA51, 2022-03-27:
UPDATE: I reinstalled the Rasp OS. I then pulled the allsky image and before installing I changed the port in the /gui/lighttpd.conf file to 8080. I then installed allsky and then the webgui. It currently works now with the port I assigned. I don't know if installing Pi-hole in a container will still work, but I have cloned my drive in case something gets messed up again. I will update if it works. It seems though that it might be important to first install Allsky before anything else because it places an lighttdp file in the etc folder. If Pi-hole is installed into a container does not alter the lighttpd file in the etc folder.

EricClaeys, 2022-03-27:
@OrionA51   Glad you got the Pi working again.  Allsky doesn't have a way to change the port except for what you did.  We figure if anyone's going to change the port they know what they're doing since that's a fairly advanced task.

I'd like the gui/install.sh script to be smart enough to only install lighttpd if it's not already installed.  In the meantime, if you ever update the gui you'll need to reset the port.

If you issue is resolved, would you please close this Issue ?  Thanks.

OrionA51, 2022-03-27:
LAST UPDATE: I was able to install Pi-hole in Docker with Docker-Compose and It did not mess with Allsky. I able able to access both programs via Webui after installing Pi-hole. For anyone reading this...(read above post first) Installing Pi-hole this way will pose a challenge because you can follow the online guide (link at end), but it will give an error when you get to /$ sudo docker-compose up -d. It will say "Port 53 in use." You have to kill what is running on port 53 using /$ kill $(lsof -t -i:53) and then input /$ docker-compose up --detach. You can can continue with the next steps in the guide.  This is the guide I used to install Pi-hole - https://www.geeksforgeeks.org/create-your-own-secure-home-network-using-pi-hole-and-docker/  *Note- make sure you have nothing else important running on 53. I did this with a fresh install of the OS and only installed Allsky so nothing else was using it.

EricClaeys, 2022-03-27:
@OrionA51  What do you mean by being able to access both programs in WebUI?  Did you add something to the Allsky code to allow the WebUI to access Pi-hole?

OrionA51, 2022-03-28:
No they just run on different ports now. Installing Pi-hole not in a container would normally overwrite the lighttpd.conf that Allsky creates in the /etc/lighttpd folder. Installing Pi-hole via docker container will not change that file. If Installing Pi-hole the regular way (no container) after Allsky WebUI is already installed, the Pi-hole install will overwrite that file and the WebUI for allsky won't launch.

closed 2022-03-28 by OrionA51