AllskyTeam/allsky #1104: installation of software

maserowik, 2022-03-27:
OK new issue.  I did a git pull of the newer version of allsky.  I follow what I have done in the past but i believe something has changed some where.  I when I go to my website.  n3bsq.com/allsky and select either video/startrails or keograms I am get the following message
[image]

should i run the website/install.sh -- update even thought the site is hosted?

Mike

OrionA51, 2022-03-27:
I believe you have to run the install separately. First the allsky install and then the website/gui install.

AndreasLMeg, 2022-03-27:
I always follow the instructions on the wiki page https://github.com/thomasjacquin/allsky/wiki/How-to-update-the-software
maybe https://github.com/thomasjacquin/allsky/wiki/Upgrade-from-0.8.2-or-prior-versions would help

maserowik, 2022-03-27:
@OrionA51 @AndreasLMeg   I have done the steps in the wiki numerous time still getting the message listed above.  no error in the error.log file either.

OrionA51, 2022-03-27:
I won’t be of very much help because I am currently having my own problems. Not the same as yours though. The only thing I can suggest is uninstalling everything using the command prompt. Every trace. I am not quite sure I remember exactly how to do that but I did Google it once. 

Ryan C.

EricClaeys, 2022-03-28:
@maserowik  Mike,  We are finding it very difficult to determine if the website is running on a Pi or another machine.  Seems like it should be simple.  ARG!!!

Would you please rename the `functions.php` file on your web server (it should be in the same directory as image.jpg) to `functions-OLD.php`, then copy the attached file (with out the ".txt") to the server, and try again?  This file tries a different mechanism for determining if it's running on a Pi or not.  The message you are seeing should ONLY appear if on a Pi, which your website isn't.

Thanks

[functions.php.txt](https://github.com/thomasjacquin/allsky/files/8358991/functions.php.txt)

EricClaeys, 2022-03-28:
@OrionA51 
> I believe you have to run the install separately. First the allsky install and then the website/gui install.
FYI, It's not possible to run both at the same time, and if you don't have Allsky installed you can't run the website or gui install since there is no install.sh script.  And if you run the website install before the gui install, it'll give you a warning, but let you continue.

maserowik, 2022-03-28:
@EricClaeys 
> @maserowik Mike, We are finding it very difficult to determine if the website is running on a Pi or another machine. Seems like it should be simple. ARG!!!
>
> Would you please rename the [...]

i uploaded the newer file to my website.  it appears to have fix one issue but another issue is resent.
startrails and keograms appear to work but when i got to videos i get this.
[image]

but when i got to my host files i have files in there.
[image]

Thanks
Mike

maserowik, 2022-03-28:
not sure what changes on my end but have a new issue 
on the web gui i lost the live view now.
[image]

ugh!!!!!!

AndreasLMeg, 2022-03-28:
Hello Mike,

I'm not a web/html developer but the delepopment tools in the browser (firefox: ctl+shift+i) often are helpful
[image]

In some cases you can find problems in the loggings, inspectors, ....
Unfortunately that's all I can do to troubleshoot

maserowik, 2022-03-28:
> Hello Mike,
> 
> I'm not a web/html developer but the delepopment tools in the browser (firefox: ctl+shift+i) often are helpful [image]
> 
> In some cases you can find problems in the loggings, inspectors, .... Unfortunately that's all I can do to troubleshoot
[image]

on hosting site the file is there
[image]

AndreasLMeg, 2022-03-28:
In my case.: GET 	http://10.0.0.59/current/tmp/image.jpg?_ts=1648492889233

and using "http://10.0.0.59/current/tmp/image.jpg" in the browser I get the correct image:
[image]

maserowik, 2022-03-29:
@EricClaeys 
> @maserowik Mike, are you running the WebUI on your remote server and not on the Pi? I didn't think that would work.
> 
> I will look into the website issue, but I suspect another file needs updating.

Everything is running from the pi.  the only thing remote is the website.  I thinking i may want to re-flash the sd card and start over from the beginning.  
maybe one of the raspberry pi updates messed something up.

EricClaeys, 2022-03-29:
@maserowik Mike,
In the image you included with all the red "404 (Not Found)" messages, the URL it was trying to get looks like a combination of your remote site's directory structure and your Pi's URL.  What is your "IMG_DIR" set to in config.sh?

For the "No allsky images" message from your server, would you please attach the videos/index.php file from the server (appending ".txt")?  I'm pretty sure I know what the issue is.

maserowik, 2022-03-29:
@EricClaeys 

> @maserowik Mike, In the image you included with all the red "404 (Not Found)" messages, the URL it was trying to get looks like a combination of your remote site's directory structure and your Pi's URL. What is your "IMG_DIR" set to in config.sh?

[image]

> For the "No allsky images" message from your server, would you please attach the videos/index.php file from the server (appending ".txt")? I'm pretty sure I know what the issue is.

[index.php.txt](https://github.com/thomasjacquin/allsky/files/8375613/index.php.txt)

Thanks
Mike

maserowik, 2022-04-03:
@EricClaeys 
Not sure if you had a chance to look at the index.php.txt on the above message,  Did a fresh install and have attached the index.php.txt again.
[index.php.txt](https://github.com/thomasjacquin/allsky/files/8405342/index.php.txt)

maserowik, 2022-04-04:
@EricClaeys 
I updated the allsky web site today and it broke what the website again.
[image]

EricClaeys, 2022-04-05:
@maserowik   I'm 99% sure you installed a non-official fixed version of the functions.php file, and the update put the old, buggy version back. I'm hoping the fixed version will be approved within a day.  In the meantime, here it is:

[functions.php.txt](https://github.com/thomasjacquin/allsky/files/8414941/functions.php.txt)

On a positive note, the update today should fix you "No allsky images".

Sorry for the hassle and for taking so long to fix the message above in red.

maserowik, 2022-04-05:
@EricClaeys 

Something is still not right.
1) i renamed the original file that is on my hosted site to functions.php.405 and then uploaded the one to my website and called it functions.php

2) i then did a shift f5 to clear out my browser

3) went to n3bsq.com/allsky and I do see my live image
 
4) went to the side bar for video/startrail and keograms and received the following screen.
[image]

received the generic error of http 500.

thanks
mike

EricClaeys, 2022-04-05:
@maserowik Mike, does your keograms/index.php file look like this?
~~~html
<?php include '../functions.php'; disableBuffering(); // must be first line ?>
<!DOCTYPE html>
<html lang="en">
	<head>
		<meta http-equiv="Content-Type" content="text/html" />
		<meta name="viewport" content="width=device-width, initial-scale=1">
		<link rel="shortcut icon" type="image/png" href="../allsky-favicon.png">
		<title>Keograms</title>

		<script src="https://ajax.googleapis.com/ajax/libs/jquery/2.2.0/jquery.min.js"></script>
	<script src="../analyticsTracking.js"></script>
	<link rel="stylesheet" href="https://maxcdn.bootstrapcdn.com/font-awesome/4.5.0/css/font-awesome.min.css">
		<link href="../allsky.css" rel="stylesheet">
	</head>
	<body>
		<?php display_thumbnails("Keogram"); ?>
	</body>
</html>
~~~
If so, put "//" in front of `disableBuffering()` so the first line looks like:
~~~html
<?php include '../functions.php'; //disableBuffering(); // must be first line ?>
~~~

One of the things we added was to disable buffering so you'd see each thumbnail as it was created, rather than seeing nothing until all of them were created.  I'm wondering if your server doesn't support that and is throwing an error.

Do you have access to the server log file?

EricClaeys, 2022-04-05:
@maserowik Mike,
Would you please do me a favor and 
* rename your functions.php to functions.php-good
* move the attached file in, calling it functions.php
* REMOVE the "//" from the startrails/index.php file
* go to the startrails page

You should see a list of numbers and names at the top of the page, then the thumbnails.
Copy/paste the values to your reply.

I'm trying to determine what function doesn't work on your server that caused the error the first time.
Thanks much!

[f.php.txt](https://github.com/thomasjacquin/allsky/files/8420619/f.php.txt)

maserowik, 2022-04-05:
@EricClaeys 
Array
(
    [0] => system
    [1] =>  exec
    [2] =>  shell_exec
    [3] =>  passthru
    [4] =>  mysql_list_dbs
    [5] =>  ini_alter
    [6] =>  dl
    [7] =>  symlink
    [8] =>  link
    [9] =>  chgrp
    [10] =>  leak
    [11] =>  popen
    [12] =>  apache_child_terminate
    [13] =>  virtual
    [14] =>  mb_send_mail
)

maserowik, 2022-04-05:
@EricClaeys 
I did a shift f5 and now this is the display i am getting.
i am getting the number mentioned above and now images.
[image]

maserowik, 2022-04-05:
@EricClaeys 

I removed the // in all three index.php .  I am only getting this error on the videos
[image]

EricClaeys, 2022-04-05:
@maserowik The image with the numbers/names and thumbnails is what I expected, although I didn't know what names would be there.  The names are PHP functions which have been disabled by your service provider since they can be security issues.  What's interesting is that allsky isn't trying to use any of those functions, and yet you got the "This page isn't working" error message earlier, which is because the PHP program crashed somewhere.  I assumed it crashed when it tried to use a function that would appear in the list.

I'm not sure why you're getting the "Warning: exec() has been disabled", since we check if the "exec" function is disabled, and if so, don't try to use it.  And you can see from the disabled list that "exec" is in it.

Would you please try the attached file?

Do you have access to the server log file?

[f2.php.txt](https://github.com/thomasjacquin/allsky/files/8421526/f2.php.txt)

EricClaeys, 2022-04-05:
@maserowik    I found a way to display PHP errors instead of "This page isn't working".
Please restore the functions.php to your original one, then use the attached file for the `keograms/index.php` file (rename it first).  Then display the keograms.  I'm hoping you'll get a "Parse error".

[keogram_index.php.txt](https://github.com/thomasjacquin/allsky/files/8421682/keogram_index.php.txt)

maserowik, 2022-04-06:
@EricClaeys 

> @maserowik I found a way to display PHP errors instead of "This page isn't working". Please restore the functions.php to your original one, then use the attached file for the `keograms/index.php` [...]

Warning: Unknown: Invalid date.timezone value 'EDT', we selected the timezone 'UTC' for now. in Unknown on line 0

Fatal error: Uncaught Error: Call to undefined function disableBuffering() in /home/u919658081/domains/n3bsq.com/public_html/allsky/keograms/index.php:24 Stack trace: #0 {main} thrown in /home/u919658081/domains/n3bsq.com/public_html/allsky/keograms/index.php on line 24

maserowik, 2022-04-06:
@EricClaeys 
 I add the// to the disableBuffering() the page load fine.

EricClaeys, 2022-04-06:
@maserowik   Mike, please try with this functions.php file and the "keogram_index.php.txt" file I posted yesterday.

Thanks
[fixed_functions.php.txt](https://github.com/thomasjacquin/allsky/files/8432752/fixed_functions.php.txt)

maserowik, 2022-04-06:
@EricClaeys 
> @maserowik Mike, please try with this functions.php file and the "keogram_index.php.txt" file I posted yesterday.
> 
> Thanks [fixed_functions.php.txt](https://github.com/thomasjacquin/allsky/files/8432752/fixed_functions.php.txt)

not sure what I should see but it appears to be working for the keograms.  startrails and video not working 

[image]

maserowik, 2022-04-06:
@EricClaeys 
so I asked my hosting company if there was any logging file.  They turned on error logging on my site and i looked in the startrails and keogram there are error log file there.  But when I got to  my video dir there is a log file there. it is attached to this email
[error_log.txt](https://github.com/thomasjacquin/allsky/files/8437128/error_log.txt)

EricClaeys, 2022-04-07:
@maserowik    It's great that they enabled logging, it helps a lot.
The code checks if the "exec()" function is disabled, which according to the numbers/names above, it is on your server, and yet our check obviously isn't working.  I'll add some debugging code and post another file to try.

For the keograms/index.php file can you restore it to the version that had `disablebuffering(); ...` as the first line, and remove the `//`, the view the keograms?  I think last time you tried you got the "This page isn't working" message.  If you DO get that message, please look in the error_log.txt file.

Thanks Mike, I appreciate your support.

maserowik, 2022-04-07:
@EricClaeys 
I did the switch this morning and it worked.  no error message at all.
[image]

EricClaeys, 2022-04-07:
@maserowik   Is it correct that your `functions.php` is the `fixed_functions.php.txt` I uploaded yesterday, and everything else is back to what you upgraded to a few days ago?

Your keograms are all short and wide.  Is that the shape of the sensor?

maserowik, 2022-04-07:
@EricClaeys 
I believe I am using the fixed_fucttions code you sent the other day .  correct all code is back to the upgrade a few days ago.  

Keograms and Stratrails work now
no error log in the Keogram folder
no error log in the Startrails folder

the timelapse still produces the following error.
[image]
error in the video log folder error is attached to here
[error_log (2).txt]

(https://github.com/thomasjacquin/allsky/files/8446156/error_log.2.txt)
e

EricClaeys, 2022-04-08:
@maserowik   Mike, I'll have to create a debugging version of functions.php.  Our check for exec() being disabled apparently isn't working on your server, even though the list of numbers/names said it was.

FYI, the "fixed" functions.php was just made official.

maserowik, 2022-04-19:
@EricClaeys 
> @maserowik Mike, I'll have to create a debugging version of functions.php. Our check for exec() being disabled apparently isn't working on your server, even though the list of numbers/names said it was.
> 
> FYI, the "fixed" functions.php was just made official.

I did a fresh install on the website and not sure if the there was a fix for the video in that build but I still am getting this on my video page.

[image]

EricClaeys, 2022-04-22:
@maserowik    Mike, I don't have a fix yet; I'm still not sure what the problem is since the code only calls exec() if it is **en**abled, but your system says it's disabled.
I still need to give you a version with some debugging code.

EricClaeys, 2022-05-04:
@maserowik    Mike,  Try the attached file.  It won't fix the problem but hopefully will give some additional info.
Make sure you save your current `/var/www/html/allsky/functions.php` file.

This file _should_ only display the "Failed to make" message once.

[functions.php.txt](https://github.com/thomasjacquin/allsky/files/8618093/functions.php.txt)

maserowik, 2022-05-09:
@EricClaeys 
sorry for the delay   here is a screen capture of the error.
[image]

maserowik, 2022-05-09:
[image]
error log from hosting site

EricClaeys, 2022-05-11:
@maserowik   Darn, the functions.php file I attached is too new for your website.  Can you update the website, then re-apply the new functions.php.txt file?

maserowik, 2022-05-11:
@EricClaeys 

> @maserowik Darn, the functions.php file I attached is too new for your website. Can you update the website, then re-apply the new functions.php.txt file?

Was this directed to me or did this get cross threaded.

Mike

maserowik, 2022-05-11:
@EricClaeys    I did a fresh download of the website today 5/11/2022 @ 11:00 AM and then redownloaded the function.php.txt file from what you post 7 days ago.  I did a ctrl f5 to refresh my browser and here is the message for the video.
[image] funny thing is that the file are arriving on my hosting site.
[image]
  and here is the error.log
[error_log (3).txt](https://github.com/thomasjacquin/allsky/files/8671076/error_log.3.txt)

EricClaeys, 2022-05-11:
@maserowik Mike, I must have made more changes to the file than I remembered. If you upload your original `functions.php` I will add the debugging code to it.

maserowik, 2022-05-12:
@EricClaeys   Eric well i guess I really have it mucked up now.  I am not sure what is happening now.  I did a fresh pull of the website and thought I put the the original functions.php back but I have not clue if that was the correct one.  not sure how to proceed with this again.

EricClaeys, 2022-05-13:
@maserowik   Mike, would you please double check that you copied all updated files in the Website to your server?  The message above mentions line 16 of videos/index.php, but the current version of that file only has 6 lines.  The prior version had at least 16.

A lot of people have the Website installed on remote servers like you, so I've been wanting to come up with a way to make it easier to install and maintain it.  One thought is to have a copy of the configuration files on the Pi where people can easily change them, and then have them automatically copied to the remote server.  And have the ability to create the thumbnails on the Pi and upload them.  Anything else you can think of to make it easier to maintain and update a remote Website?

maserowik, 2022-05-13:
@EricClaeys How was Thomas doing it before? Any way to go back to it?

 
still fighting with the website.

EricClaeys, 2022-05-13:
@maserowik Mike, prior to last August,  video thumbnails weren't made.  After that, they were made the same way we make them now, but we now check beforehand if they can be made, and display error messages if it fails.
The error you are getting about "invalid date.timezone" I have never seen and don't know what causes it since Allsky doesn't use the timezone.  It might be a configuration issue on your server.  And I don't know what causes the "exec()" message you were getting.  I am hoping the newest website software will tell me.

maserowik, 2022-05-26:
@EricClaeys Eric, I redid everything and getting a message on my website.  
[image]
I believe i have all the files we talked about updated.

EricClaeys, 2022-05-26:
@maserowik Mike, `disableBuffering` Is defined around line 50 in the current `functions.php`, and also in the functions.php several posts ago.
It might be easier for you to give me access to your website like you did a few months ago.  Feel free to email me.

maserowik, 2022-05-27:
@EricClaeys I just set you the information.  Let me know if you have questions.

maserowik, 2022-05-28:
@EricClaeys Thanks for solving the issue.  I am closing this now.
Mike

closed 2022-05-28 by maserowik