AllskyTeam/allsky #1105: CET -> CEST +2h?

si6pack, 2022-03-27:
Hi,
tonight  (26.03. -> 27.03) we changed from CET to CEST (= +1h). 
While image saving started 19:46 yesterday, it started 21:46 today...
I checked "date" on both my Rpi's - returned me CEST (date and time were correct). 
Since I did not touch the camera settings, I suppose another "problem"...according to my camera settings, imaging starts when sun altitude is -9° below the horizon.
Thank you for considering...
Regards,
si6pack (Christian)

EricClaeys, 2022-03-27:
@si6pack Run the following and see what it returns for daytime and nighttime for today:
~~~
(source ~/allsky/config/config.sh; sunwait list angle $(jq -r ".angle,.latitude,.longitude" ${CAMERA_SETTINGS}))
~~~
and for before the time change:
~~~
(source ~/allsky/config/config.sh; sunwait list d 25 m 3 y 2022 angle $(jq -r ".angle,.latitude,.longitude" ${CAMERA_SETTINGS}))
~~~

This will help us narrow down where the issue is.

EricClaeys, 2022-03-28:
Test
[image][image]

si6pack, 2022-03-28:
@EricClaeys - this is what was returned after executing the commands:

Command 1 from above:
06:14, 20:26

Command 2 from above:
Error: "Year" must be between 0 and 99: **2022** -> changed to **22**

Command 2 modified
05:21, 19:21

si6pack, 2022-03-28:
@EricClaeys - I just checked the images section of the WebUI of both of my Raspis and had to learn, that today Mar 28th, image saving started 20:26 and 20:27, respectively. In a way back to normal...
Hope this helps for the further analysis?!

EricClaeys, 2022-03-30:
@si6pack,  the `sunwait` commands returned times as expected - one hour difference.  Allsky runs the same command to determine when daytime/nighttime begin, so I'm not sure what the issue was.  I have a feeling it won't happen again until daylight savings changes again in 6 months.

Interesting that `sunwait` doesn't take a 4-digit year.  If you type `sunwait -?` it shows "YYYY".  Oh well, glad you figured that out.

**Are you ok if we close this Issue?**

si6pack, 2022-03-30:
@EricClaeys, I am under the same impression - issue will take place in 6 month from now when switching back to CET...I keep an eye on it.
Ans, yes I am fine with closing this issue!
All the best,
C

EricClaeys, 2022-03-30:
Closing per agreement with submitter.

closed 2022-03-30 by EricClaeys