AllskyTeam/allsky #1112: certification problem again

Pedritoprint, 2022-03-30:
After updating my allsky yesterday to latest version, to put my camera to the map, the camera couldn't load the necessary files to my HomePage. 
--> **Fataler Fehler: Certificate verification: Not trusted** (xxxxxxxxx)
After reading the WIKI and changing parameters to corresponding files like in my older Version of the Allsky-SW, the failure is still present. Not possible to upload via ftp like before. I have nothing changed on my own WebServer. 
Manually upload the "test.txt"-File like written in WIKI is also not working, same ERROR.
Any news or somthing to change on WebServer?
Reinstalling the older version: no problem to upload the files to WebServer. mhhh??
greetings PA

Pedritoprint, 2022-03-30:
OK dr EricClaeys. 
- I installed the new SW exactly like written in WIKI [9]
- checked all parameters in config.sh and ftp-settings.sh (Editor in WebUi) and Camera Settings. Lot of the parameters are automatically taked over from the prior version of allsky.
- Then, after Problem by uploading to WebServer, i changed, corresponding to WIKI [12.iii], the Parameter "LFTP_COMMANDS" with "_set ssl:verify-certificate no_" like in Version before. 
- I also tested the Version with an lftprc-file. 
- Then i replaced the **ftp-settings.sh** File with the "old" one i hat in prior version, no success.
- then i renamed ne new "allsky"-**folder** to _ori and renamed the old one to **allsky**.
- rebooted the Raspi and all was working fine, also if i took the new ftp-settings.sh to the "old" allsky-Version. btw, i think the new and old ftp-settings file are exact the same, so the "failure" has to be in an other file i'm looking for.

EricClaeys, 2022-03-31:
@Pedritoprint In the Wiki page on Troubleshooting uploads, try the steps in the last section "Still having problems uploading?"  If that doesn't help, attach your `allsky/tmp/lftp_cmds.txt` file, but be sure to cross out your username and password.

Pedritoprint, 2022-03-31:
i think you mean the part with the manually uploaded "test.txt"-File. I made this as written in my first post. Here my log attached.....
[lftp_cmds.txt](https://github.com/thomasjacquin/allsky/files/8386606/lftp_cmds.txt)
And as wrote in my last post, the only file who could be uploaded has a wrong timestamp despite the fact that the Raspi and my WebServer has the correct time set. The time of the uploaded file, to the Server, has +1h.
Peter

EricClaeys, 2022-03-31:
@Pedritoprint. Peter, I just found the cause of this bug and submitted a fix.  I will let you know when it is ready.  In the meantime you can copy your old `allsky-OLD/scripts/upload.sh` to `allsky/scripts`.
Sorry about that.

Are your Pi and the server both using daylight savings?

Pedritoprint, 2022-03-31:
Yes EricClaeys, both works with daylight savings and thanks for infos about the bug.
good luck and i'm waiting for news about your intervention.
Peter

EricClaeys, 2022-03-31:
@Pedritoprint. Peter,
Please grab the `scripts/upload.sh` file from GitHub.  It has the fix as well as other improvements. 
If this resolves the problem,  please close this Issue.
Thanks

Pedritoprint, 2022-04-01:
Thanks EricClaeys for your effort. well done.
Peter

closed 2022-04-01 by Pedritoprint