AllskyTeam/allsky #1118: Time to take image differs from requested exposure time.

kalalaa, 2022-04-02:


EricClaeys, 2022-04-02:
@kalalaa You have two unrelated thins going on. 
When did you take your darks?  If it was more than about 9 months ago, you'll want to retake them.  The algorithm to take them and to subtract them has improved.

For the second issue, I don't know what causes some ZWO cameras to apparently take exposures shorter than requested.  That check is new.  If you look at the timestamps in the log you posted, you'll see the software requested 20 second exposures but returned in significantly less than that.
When that happens, do the pictures look OK?  Try turning on `Show Mean Brightness` in the WebUI and see if it's jumping all over the place.

kalalaa, 2022-04-02:
My darks are new. Maybe  the big difference in darks came from frame splitting. Now i have some good darks, with 
acceptable final noise. When i take new darks with different temperature, i check them. If the new darks have different pattern, i delete them, and capture it again, so i now can handle it.
But you can see, sometimes the dark is only 2.2sec and sometimes it is 8.6 sec long. 
When the subtraction is on, with correct darks, there is a little noise on the final image. I think this come from the different dark, and light timings. The final problem is this. I found some ASI forums, with long exposure issues, so the problem is maybe in the asi firmware.
The final image is acceptable, with some noise, and i can use it, so not a big problem, but in older version, i didn't have this.

Thank you!

kalalaa, 2022-04-02:
I found in other forums, there are problems with ASI120 long exposures, with some softwares.
https://bbs.astronomy-imaging-camera.com/d/6968-zwo-asi120mc-raw16-long-exposure-linux
https://www.cloudynights.com/topic/487175-zwo-asi120mc-for-dso-photography/

kalalaa, 2022-04-02:
Oh and i found this. I will try it.
https://github.com/rkaczorek/astroberry-server/issues/109

kalalaa, 2022-04-02:
I installed debian buster, and the allsky. Timing has the same behavior. 
I found an interesting thing: If i turn off 0.8 exposure, the time has negative difference (at 20 sec the exposure is about 8 sec), and if i use 0.8 exposure, the difference is positive. At 20 sec the exposure is around 38 sec. BUT the final image has around the same brightness, so the real exposure is 20 sec in both cases.
The real difference, when i use 0.8 expo, the noise is lower, and stable, when 0.8 off, the noise pattern sometimes different from other darks, and the noise is much bigger.
I think the problem is not solved, but i can use the allsky camera as earlier, now with good results with the dark subtraction.

Thank you!

EricClaeys, 2022-04-05:
Closing Issue, per user request.  Problem solved to his satisfaction.

closed 2022-04-05 by EricClaeys