AllskyTeam/allsky #1120: Not capturing during night

OrionA51, 2022-04-02:
If I enable daytime capture I can get a live preview. If I turn daytime capture off, it never captures at night. I get the yellow text screen in live view saying daytime capture is off. Says it all night.

EricClaeys, 2022-04-02:
@OrionA51 Please follow the instructions in the Reporting Issues Wiki page and attach the log file.  Start Allsky about 10 minutes before your nighttime begins, and let it runs for about 10 minutes after nighttime begins.  I will need to see what happens at the transition.

OrionA51, 2022-04-02:
I’ve been leaving it run and it is now imaging. Daytime is set to false. So it looks to me like it’s a time issue. Or time zone issue. Under camera settings - other settings, I do have the coordinates input. The Rpi time is correct too. I’ll look into getting the log file and also see if it stops imaging at or near sunset.

Ryan C.

OrionA51, 2022-04-03:
Ok so right at sunset the capture stopped and I get the screen that says capture is off during the day. I assume that should be easy to fix? Just a matter a reversing something…

Ryan C.

OrionA51, 2022-04-03:
@EricClaeys Ok I sort of feel stupid here. I had my longitude set as -81W. Which is correct, but the program as I understand cannot use a "-" sign. I understand now that "W' or "E" is the determining factor for "-" or "+". Using a -+ sign is mainly for GPS systems. I would suggest to make that clear in the settings or simply disable to ability to use -+ to make it idiot proof. Sorry to about opening an issue too soon.

EricClaeys, 2022-04-03:
@OrionA51 Glad you figured it out.
I will add a check for a minus sign with a N, S, E, or W, which should be illegal.
I modified a test version of the website to allow either the +/- or N,S,E,W version.  I think the WebUI should allow either form.

If your problem is resolved,  please close the Issue.

closed 2022-04-03 by OrionA51