AllskyTeam/allsky #1121: White images during the day

nacho3g, 2022-04-02:
Hello everyone, I use a Raspberry pi 4 with a HQ camera. I have done a clean installation and during the day I do not get images, I think the system deletes them by being over exposed, during the night, I do get images.

Mi log:

> Apr  2 20:42:33 allsky allsky.sh[690]:      ***** Starting AllSky *****
Apr  2 20:42:34 allsky allsky.sh[690]: CAMERA: RPiHQ
Apr  2 20:42:52 allsky allsky.sh[846]: *** WARNING: Could not set locale to en_US.UTF-8 ***
Apr  2 20:42:52 allsky allsky.sh[846]:  ************************************************
Apr  2 20:42:52 allsky allsky.sh[846]:  *** Allsky Camera Software v0.8.3.3  |  2022 ***
Apr  2 20:42:52 allsky allsky.sh[846]:  ************************************************
Apr  2 20:42:52 allsky allsky.sh[846]: Capture images of the sky with a Raspberry Pi and a RPi HQ camera
Apr  2 20:42:52 allsky allsky.sh[846]: Add -h or --help for available options
Apr  2 20:42:52 allsky allsky.sh[846]: Author: Thomas Jacquin - <jacquin.thomas@gmail.com>
Apr  2 20:42:52 allsky allsky.sh[846]: Contributors:
Apr  2 20:42:52 allsky allsky.sh[846]: -Knut Olav Klo
Apr  2 20:42:52 allsky allsky.sh[846]: -Daniel Johnsen
Apr  2 20:42:52 allsky allsky.sh[846]: -Robert Wagner
Apr  2 20:42:52 allsky allsky.sh[846]: -Michael J. Kidd - <linuxkidd@gmail.com>
Apr  2 20:42:52 allsky allsky.sh[846]: -Rob Musquetier
Apr  2 20:42:52 allsky allsky.sh[846]: -Eric Claeys
Apr  2 20:42:52 allsky allsky.sh[846]: -Andreas Lindinger
Apr  2 20:42:52 allsky allsky.sh[846]:  Camera: Raspberry Pi HQ camera
Apr  2 20:42:52 allsky allsky.sh[846]:   - Resolution: 4096x3040
Apr  2 20:42:52 allsky allsky.sh[846]:   - Pixel Size: 1.55microns
Apr  2 20:42:52 allsky allsky.sh[846]:   - Supported Bins: 1x, 2x and 3x
Apr  2 20:42:52 allsky allsky.sh[846]: Capture Settings:
Apr  2 20:42:52 allsky allsky.sh[846]:  Command: raspistill
Apr  2 20:42:52 allsky allsky.sh[846]:  Image Type: RGB24
Apr  2 20:42:52 allsky allsky.sh[846]:  Resolution (before any binning): 4096x3040
Apr  2 20:42:52 allsky allsky.sh[846]:  Quality: 95
Apr  2 20:42:52 allsky allsky.sh[846]:  Daytime capture: Yes
Apr  2 20:42:52 allsky allsky.sh[846]:  Exposure (night): 10000ms
Apr  2 20:42:52 allsky allsky.sh[846]:  Auto Exposure (night): Yes
Apr  2 20:42:52 allsky allsky.sh[846]:  Gain (day): 1.00
Apr  2 20:42:52 allsky allsky.sh[846]:  Gain (night): 1.00
Apr  2 20:42:52 allsky allsky.sh[846]:  Auto Gain (night): Yes
Apr  2 20:42:52 allsky allsky.sh[846]:  Brightness (day): 100
Apr  2 20:42:52 allsky allsky.sh[846]:  Brightness (night): 50
Apr  2 20:42:52 allsky allsky.sh[846]:  Saturation: 1.0
Apr  2 20:42:52 allsky allsky.sh[846]:  Auto White Balance: No
Apr  2 20:42:52 allsky allsky.sh[846]:  WB Red: 2.50
Apr  2 20:42:52 allsky allsky.sh[846]:  WB Blue: 2.00
Apr  2 20:42:52 allsky allsky.sh[846]:  Binning (day): 1
Apr  2 20:42:52 allsky allsky.sh[846]:  Binning (night): 1
Apr  2 20:42:52 allsky allsky.sh[846]:  Delay (day): 15000ms
Apr  2 20:42:52 allsky allsky.sh[846]:  Delay (night): 10ms
Apr  2 20:42:52 allsky allsky.sh[846]:  Text Overlay:
Apr  2 20:42:52 allsky allsky.sh[846]:  Text Position: 15px left, 35px top
Apr  2 20:42:52 allsky allsky.sh[846]:  Font Name: 0
Apr  2 20:42:52 allsky allsky.sh[846]:  Font Color: 255, 255, 255
Apr  2 20:42:52 allsky allsky.sh[846]:  Small Font Color: 0, 0, 255
Apr  2 20:42:52 allsky allsky.sh[846]:  Font Line Type: 16
Apr  2 20:42:52 allsky allsky.sh[846]:  Font Size: 7.0
Apr  2 20:42:52 allsky allsky.sh[846]:  Font Line: 1
Apr  2 20:42:52 allsky allsky.sh[846]:  Outline Font : No
Apr  2 20:42:52 allsky allsky.sh[846]:  Rotation: 0
Apr  2 20:42:52 allsky allsky.sh[846]:  Flip Image: 0
Apr  2 20:42:52 allsky allsky.sh[846]:  Filename: image.jpg
Apr  2 20:42:52 allsky allsky.sh[846]:  Filename Save Directory: /home/pi/allsky/tmp
Apr  2 20:42:52 allsky allsky.sh[846]:  Latitude: 42.84N
Apr  2 20:42:52 allsky allsky.sh[846]:  Longitude: 2.67W
Apr  2 20:42:52 allsky allsky.sh[846]:  Sun Elevation: -6
Apr  2 20:42:52 allsky allsky.sh[846]:  Locale: es_ES.UTF-8
Apr  2 20:42:52 allsky allsky.sh[846]:  Notification Images: Yes
Apr  2 20:42:52 allsky allsky.sh[846]:  Show Time: Yes (format: %Y%m%d %H:%M:%S)
Apr  2 20:42:52 allsky allsky.sh[846]:  Show Exposure: No
Apr  2 20:42:52 allsky allsky.sh[846]:  Show Gain: No
Apr  2 20:42:52 allsky allsky.sh[846]:  Show Brightness: No
Apr  2 20:42:52 allsky allsky.sh[846]:  Show Focus Metric: No
Apr  2 20:42:52 allsky allsky.sh[846]:  Preview: No
Apr  2 20:42:52 allsky allsky.sh[846]:  Taking Dark Frames: No
Apr  2 20:42:52 allsky allsky.sh[846]:  Debug Level: 0
Apr  2 20:42:52 allsky allsky.sh[846]:  On TTY: No
Apr  2 20:42:52 allsky allsky.sh[846]:  Mode Mean: Yes
Apr  2 20:42:52 allsky allsky.sh[846]:     Mean Value (night): 0.200
Apr  2 20:42:52 allsky allsky.sh[846]:     Mean Value (day): 0.500
Apr  2 20:42:52 allsky allsky.sh[846]:     Threshold: 0.010
Apr  2 20:42:52 allsky allsky.sh[846]:     p0: 5.000
Apr  2 20:42:52 allsky allsky.sh[846]:     p1: 20.000
Apr  2 20:42:52 allsky allsky.sh[846]:     p2: 45.000
Apr  2 20:42:52 allsky allsky.sh[846]: ==========
Apr  2 20:42:52 allsky allsky.sh[846]: === Starting daytime capture ===
Apr  2 20:42:52 allsky allsky.sh[846]: ==========
Apr  2 20:42:52 allsky allsky.sh[846]: Stop the allsky service to end this process.
Apr  2 20:42:52 allsky allsky.sh[846]: STARTING EXPOSURE at: 2022-04-02 20:42:52   @ 500 us (0.500 ms)
Apr  2 20:42:55 allsky allsky.sh[846]: Sleeping 1.0 seconds...
Apr  2 20:42:56 allsky allsky.sh[846]: STARTING EXPOSURE at: 2022-04-02 20:42:56   @ 190 us
Apr  2 20:42:57 allsky allsky.sh[935]: removeBadImages.sh: File is bad: removed 'image-20220402204252.jpg' (above threshold: MEAN=99.848600, threshold = 90)
Apr  2 20:42:58 allsky allsky.sh[846]: Sleeping 1.0 seconds...
Apr  2 20:42:59 allsky allsky.sh[846]: STARTING EXPOSURE at: 2022-04-02 20:42:59   @ 72 us
Apr  2 20:43:00 allsky allsky.sh[987]: removeBadImages.sh: File is bad: removed 'image-20220402204256.jpg' (above threshold: MEAN=99.850500, threshold = 90)
Apr  2 20:43:01 allsky allsky.sh[846]: Sleeping 1.0 seconds...
Apr  2 20:43:02 allsky allsky.sh[846]: STARTING EXPOSURE at: 2022-04-02 20:43:02   @ 27 us
Apr  2 20:43:03 allsky allsky.sh[1055]: removeBadImages.sh: File is bad: removed 'image-20220402204259.jpg' (above threshold: MEAN=99.846800, threshold = 90)
Apr  2 20:43:04 allsky allsky.sh[846]: Sleeping 1.0 seconds...
Apr  2 20:43:05 allsky allsky.sh[846]: STARTING EXPOSURE at: 2022-04-02 20:43:05   @ 10 us
Apr  2 20:43:06 allsky allsky.sh[1105]: removeBadImages.sh: File is bad: removed 'image-20220402204302.jpg' (above threshold: MEAN=99.848100, threshold = 90)
Apr  2 20:43:08 allsky allsky.sh[846]: Sleeping 1.0 seconds...
Apr  2 20:43:09 allsky allsky.sh[846]: STARTING EXPOSURE at: 2022-04-02 20:43:09   @ 4 us
Apr  2 20:43:09 allsky allsky.sh[1156]: removeBadImages.sh: File is bad: removed 'image-20220402204305.jpg' (above threshold: MEAN=99.847500, threshold = 90)

EricClaeys, 2022-04-03:
@nacho3g Try decreasing your daytime brightness to 50.  In config.sh you can set REMOVE_BAD_IMAGES to false to see the images, which might give you an idea of why they are so bright.

In the future,  when reporting problems please follow the instructions in the Wiki page on Reporting Issues. Reading a log file in a comment is difficult, and at Debug Level 0 it doesn't give enough information to fix anything.

nacho3g, 2022-04-03:
Sorry for the bother, corrected the first post with the brightness changed to 50, same result.

nacho3g, 2022-04-04:
Thx solved.

EricClaeys, 2022-04-05:
Closing Issue - problem is solved per @nacho3g .

closed 2022-04-05 by EricClaeys