AllskyTeam/allsky #1123: WebPage Problems

Pedritoprint, 2022-04-03:
please see [here ](https://github.com/thomasjacquin/allsky-website/issues)
I think, this issues would't be read nor answered so often or is it better to write Web-Problems here in this issues?

EricClaeys, 2022-04-03:
@Pedritoprint Sorry, I don't understand your question. 
Problems related to the website (either on the Pi or not) should be reported in the allsky-website repository, not this one, so the fix can be associated to the Issue.
I realize not everyone understands that or knows where the problem is, but you were correct in creating the Issue in the other location.

I monitor all 3 repositories. 

Assuming this comment is related to the allsky-website Issue, would you please close this Issue since it's a duplicate?

Pedritoprint, 2022-04-03:
Thanks EricClaeys for your answer. I only saw in the WebPage issues, that there are some issues not answered, one since 3 days. So i asked here, because i feel, that this Issues are frequently visited. No Problem, i close this question and thanks.

closed 2022-04-03 by Pedritoprint