AllskyTeam/allsky #1124: Allsky log location

maserowik, 2022-04-03:
I have the log set to 4 
[image]

but on the pi i do not see a allsky.log 
[image]

did the allsky.log file move locations?

EricClaeys, 2022-04-03:
@linuxkidd   Michael,  any ideas?  This happened to me a long time ago and I don't remember what I did to get it back.

@maserowik There are no older log files.  Did you delete them, and if so, any chance you deleted allsky.log?   Have you tried restarting the service or rebooting?

maserowik, 2022-04-04:
> @linuxkidd   Michael,  any ideas?  This happened to me a long time ago and I don't remember what I did to get it back.
> 
> 
> 
> @maserowik There are no older log files.  Did you delete them, and if so, any chance you deleted allsky.log?   Have you tried restarting the service or rebooting?

@EricClaeys 
Yes I had to redo my the SD card due to user error.  I have restarted the pi No log file appears.  I then restarted the service still no log.  I am hoping that once midnight happen at my location the file will appear.  Hopefully

maserowik, 2022-04-04:
@EricClaeys 
the log file is now there.

EricClaeys, 2022-04-05:
@maserowik    What, if anything, did you do to get the log file to appear?

maserowik, 2022-04-05:
@EricClaeys 

It appeared at midnight. Maybe a process kicked off to create it. Not sure what made the log file appear.