AllskyTeam/allsky #1132: RETCODE=100

lumdiniz, 2022-04-08:
Alguém pode me ajudar, eu fui atualizar para a ultima versão e agora fica dando esse erro não consigo iniciar a camera. @thomasjacquin @EricClaeys 

  ***** Starting AllSky *****
CAMERA: ZWO
*** WARNING: Could not set locale to en_US.UTF-8 ***

**********************************************
*** Allsky Camera Software v0.8.3.3 |  2022 ***
**********************************************

Capture images of the sky with a Raspberry Pi and an ASI Camera

Add -h or --help for available options

Author: Thomas Jacquin - <jacquin.thomas@gmail.com>

Contributors:
 -Knut Olav Klo
 -Daniel Johnsen
 -Yang and Sam from ZWO
 -Robert Wagner
 -Michael J. Kidd - <linuxkidd@gmail.com>
 -Chris Kuethe
 -Eric Claeys

ZWO ASI224MC Information:
  - Native Resolution: 1304x976
  - Pixel Size: 3.8microns
  - Supported Bins: 1 2 
  - Color Camera: bayer pattern: RG

  - Camera ID: [none]
  - Camera Serial Number: 091a4b0013090900
- Sensor temperature: 46.00
  - Current camera mode: NORMAL (0)

Capture Settings:
 Image Type: RGB24
 Resolution (before any binning): 1304x976
 Quality: 95
 Daytime capture: No
 Exposure (day): 0.500ms, Auto: Yes
 Exposure (night): 10000ms, Auto: Yes
 Max Auto-Exposure (day): 10000ms (10.0s)
 Max Auto-Exposure (night): 23000ms (23.0s)
 Delay (day): 15000ms
 Delay (night): 10ms
 Gain (night only): 300, Auto: No, max: 200
 Gain Transition Time: 15.0 minutes
 Brightness (day): 100
 Brightness (night): 100
 Skip Frames (day): 5
 Skip Frames (night): 1
 Aggression: 75%
 Gamma: 50
 WB Red: 53, Blue: 90, Auto: No
 Binning (day): 1
 Binning (night): 1
 USB Speed: 80, auto: Yes
 Text Overlay: CAMERA BRAZIL
 Text Extra File: [none], Age: 0 seconds
 Text Line Height 30px
 Text Position: 15px from left, 35px from top
 Font Name: SIMPLEX (0)
 Font Color: 255, 255, 255
 Small Font Color: 0, 0, 255
 Font Line Type: 16
 Font Size: 7.0
 Font Line Width: 1
 Outline Font : No
 Flip Image: 0
 Filename: image.jpg
 Filename Save Directory: /home/pi/allsky/tmp
 Latitude: 23.11S, Longitude: 49.83W
 Sun Elevation: -6
 Locale: pt_BR.UTF-8
 Notification Images: No
 Histogram Box: 500 500 50 50, center: 652x488, upper left: 402x238, lower right: 902x738
 Show Histogram Box: No
 Show Histogram Mean: No
 Show Time: Yes (format: %Y%m%d %H:%M:%S)
 Show Temperature: Yes, type: C
 Show Exposure: Yes
 Show Gain: Yes
 Show Brightness: Yes
 Preview: No
 Taking Dark Frames: No
 Debug Level: 0
 On TTY: Yes
 Video OFF Between Images: No
 ZWO SDK version 1, 20

*** Press Ctrl+C to stop ***

*** ERROR: it's not DAY or NIGHT, it's ''
     ***** Stopping AllSky *****
'capture' exited with RETCODE=100
***
*** After fixing, restart allsky.sh. ***
***
     ***** AllSky Stopped *****

lumdiniz, 2022-04-10:
@EricClaeys o comando sunwait poll angle -6 23.11S 49.83W não me retornou nada, usei o comando sunwait poll exit angle -6 23.11S 49.83W me retornou NIGHT.
Executei a allsky pelo ./allsky.sh, e também pelo automático não inicia da o mesmo erro.

EricClaeys, 2022-04-10:
@lumdiniz   That explains why the error message didn't say what sunwait returned.
It works fine for me on two machines.
What operating system are you running?  Buster or Bullseye?
Please do the following: `type sunwait`.  It will return a path name,  do an "ls -l PATH_NAME`, and do an `ls -l /home/pi/allsky/src/sunwait` and post the results.

The sunwait command isn't working as expected, so we need to find out why.

lumdiniz, 2022-04-10:
@EricClaeys segue os resultados

pi@allsky:~ $ type sunwait
sunwait está na tabela hash (/usr/local/bin/sunwait)

pi@allsky:~ $ ls -l /usr/local/bin/sunwait
-rwxr-xr-x 1 root root 40848 mar  4  2021 /usr/local/bin/sunwait

EricClaeys, 2022-04-11:
@lumdiniz   Please do the following and report the results:
~~~shell
/usr/bin/sunwait poll angle -6 23.11S 49.83W
/usr/local/bin/sunwait poll angle -6 23.11S 49.83W
/home/pi/allsky/src/sunwait poll angle -6 23.11S 49.83W
~~~
You should be able to copy the commands above and paste into your Pi.
I'm not sure what the issue is, but am thinking it's using the wrong version of `sunwait`.

Here is what I get:
~~~shell
pi:~ $ /usr/bin/sunwait poll angle -6 23.11S 49.83W
NIGHT
pi:~ $ /usr/local/bin/sunwait poll angle -6 23.11S 49.83W
NIGHT
pi:~ $ /home/pi/allsky/src/sunwait poll angle -6 23.11S 49.83W
NIGHT
pi:~ $ 
~~~

EricClaeys, 2022-04-15:
@lumdiniz    Any update?

lumdiniz, 2022-04-15:
> 

@EricClaeys @thomasjacquin  está rodando com a versão antiga, segui o passo a passo para atualizar porém não está funcionando o dia e noite, está com esse erro, atualizei o raspibian porém volta a pasta da versão antiga funciona normal vc coloca a nova versão não inicia com o erro do tópico

EricClaeys, 2022-04-15:
@lumdiniz   What do you get if you run the commands I suggested?
~~~
/usr/bin/sunwait poll angle -6 23.11S 49.83W
/usr/local/bin/sunwait poll angle -6 23.11S 49.83W
/home/pi/allsky/src/sunwait poll angle -6 23.11S 49.83W
~~~

RLWagner, 2022-04-17:
I just ran into this same problem - newly installed Bullseye, git clone of allsky yesterday morning.  I used the solution from Issue 778: Raspi OS full-upgrade, rebuild sunwait and copy the result to /usr/local/bin.  This got sunwait working and the RETCODE=100 disappeared.  However, allsky still isn't working - see my next issue....

EricClaeys, 2022-04-17:
@RLWagner By any chance did you execute the 3 command I listed above before you made the change?

EricClaeys, 2022-04-17:
@RLWagner Would you please execute `echo $PATH` and let me know what it says?
I have a feeling it's different on Bullseye than Buster, so allsky is picking up a different version of `sunwait` that behaves differently.

EricClaeys, 2022-04-18:
I could still use the output of `echo $PATH` on a Bullseye system, so I can determine the best fix.

It's interesting that no one else on Bullseye is having this problem.

RLWagner, 2022-04-18:
/home/wagnerr/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games

lumdiniz, 2022-04-18:
@EricClaeys

echo $PATH

/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games:/snap/bin

EricClaeys, 2022-04-19:
Thanks guys

EricClaeys, 2022-05-12:
@lumdiniz, @RLWagner, you should be able to temporarily resolve this by executing `sudo rm /usr/local/sbin/sunwait`.  The next version of Allsky will resolve this.

I'm closing this Issue.

closed 2022-05-12 by EricClaeys