AllskyTeam/allsky #1147: data.json file not found

OrionA51, 2022-04-11:
When opening the index.html file to view the website, everything seems to work, but I get this message. I follow what it says, but it doesn't fix it. It does the same thing if I install the website on another computer and open the index.html file. In the Rpi folder, the data.json file is there and it has the current date and time.
[image]

EricClaeys, 2022-04-11:
@OrionA51. What is the URL to the website on the Internet?

OrionA51, 2022-04-11:
@EricClaeys It's not on the internet. I am only opening it within my local network.

EricClaeys, 2022-04-11:
@OrionA51 Press F12, then go to the "Console" tab (you may need to press the `>>` first).  Then Reload the page and see if there are any messages that help explain it.

The website giving this error is on your Pi, correct?

EricClaeys, 2022-04-11:
@OrionA51 The Allsky software will create the data.json file every morning in `allsky/tmp`, then if you have things set up correctly tly, will copy it to `/var/www/html/allsky`, which is where the website looks for it.

OrionA51, 2022-04-11:
@EricClaeys Yes I was trying to open it on the Pi first and then waited until the morning to try again. I then downloaded the website .zip file from github to another PC on the network to see if I got the same message. Last I checked, the data.json file appeared to be updated with the current date and time. My pi lost power recently though and I have to sort that out to continue with this issue.

EricClaeys, 2022-04-11:
@OrionA51   The Allsky software will only upload the data.json file to the machine(s) listed in `ftp-settings.sh`, just like any other file.
You can force it to be uploaded:
~~~
cd ~/allsky/scripts
./postData.sh
~~~

OrionA51, 2022-04-11:
@EricClaeys I'm currently looking into it, but here is what I get in the browser. I'm still learning all this
[image]
.

EricClaeys, 2022-04-11:
@OrionA51   It looks like you are trying to access the website by entering the file name, `/var/www/html/allsky/index.html`?  If so, that won't work, as you have seen.

You need to access it via a URL like `http://192.168.0.123/allsky` or `http://localhost/allsky` or something like that, depending on what your Pi is called.  When you installed the WebUI, it installed a web server on your Pi - it's what gets requests from URLs and returns pages.

OrionA51, 2022-04-11:
@EricClaeys I can access the webUI fine. I was trying to open the index.html file that is used for making a website available in the internet because I plan to make a website so other people can see it.

EricClaeys, 2022-04-12:
@OrionA51   Right, and you have to open it via a URL, not a file name.  First try to get it working locally - add "/allsky" to the end of the URL you use for the WebUI.  For example, from my PC, which is on the same network as the Pi, I get to the WebUI with the URL `http://allsky`.  I get to the website via `http://allsky/allsky`.

Once you get it working internally, you'll need to determine what the Internet URL will be.

OrionA51, 2022-04-12:
> @OrionA51. Ryan, GitHub can't help your Pi get on the Internet - it's just a repository for code and documentation
>
> You will need to get a static IP address from your Internet provider, and ideally a name associated with the IP address, e.g., http://myallsky.com. You will then need to open up your firewall to allow incoming traffic to the Pi.
>
> Those are both outside the scope of Allsky and hence what I can help with. Your Internet provider should be able to help open the firewall if you are using their hardware.
>
> You said "yes, that works". Does that mean you can now get to the website internally, and things look OK? If so, I'd like to close this Issue.

I had something odd happen with the schedule start last night. I’m going to wait until tonight to see if it happens again and then I’ll close it. 

Ryan C.

OrionA51, 2022-04-13:
@EricClaeys Thank you for your help. Excellent support.

closed 2022-04-13 by OrionA51

EricClaeys, 2022-04-13:
@OrionA51   My pleasure.  I'm always happy when I get help someone overcome a problem and enjoy Allsky!

skyviewobs, 2022-07-17:
I have installed the website to my Web Hosting site @ thetynans.co.uk/allsky and am getting the same data.json file not found message.

data.json is loaded every night to the Allsky base directory (\htdocs\allsky\) and is overwritten each morning with the end of night script as are the Video \ Keogram and Stairtrail latest images to their respective directories off the allsky base directory. The video length recorded match the data in the data.json file for elapsed time so I have to assume data.json is being read correctly but the data.json message persists. Post end of night is set and postData.sh runs correctly and no errors are generated in the Allsky log on the Pi.

This error has only occurred since setting up a new clean install of my Pi (both clean installs of Raspian and Allsky software - wiped the SD card) and a complete clean reinstall of the Allsky-Website on the remote server.

Can anyone advise a solution, thanks.

EricClaeys, 2022-07-17:
@skyviewobs in your browser, press F12 and go to the "Network" tab.  You may need to click on `>>>` first.  Reload the page.  On the tight side you'll see a list of files the browser is trying to access.  data.json will probably be in red, which means the browser couldn't find the file.  Click on it to see where it thinks the file is.

skyviewobs, 2022-07-17:
Hi Eric,

It appears it is looking for the file in https://www.thetynans.co.uk/allsky/data.json which is where the file is located on the server but as below there is a 404 error as if the file is missing. All user data on my hosting is off the /htdocs/folder so in reality /htdocs/allsky I have previously being running the website with Ver 7 for the last 18 months without any issues so not sure what is up. 

usingNewVirtualSky=true
controller.js:371 Read data.json at 07-17 5:50:21 pm:
angular.js:10661          GET https://www.thetynans.co.uk/allsky/data.json?_ts=1658076621989 404
(anonymous) @ angular.js:10661
q @ angular.js:10480
g @ angular.js:10187
(anonymous) @ angular.js:14634
$eval @ angular.js:15878
$digest @ angular.js:15689
$apply @ angular.js:15986
(anonymous) @ angular.js:1658
e @ angular.js:4473
d @ angular.js:1656
yc @ angular.js:1676
Zd @ angular.js:1570
(anonymous) @ angular.js:28683
i @ jquery.min.js:2
fireWith @ jquery.min.js:2
ready @ jquery.min.js:2
J @ jquery.min.js:2
controller.js:446   *** Unable to read file
controller.js:359   * streamDaytime == true
controller.js:360   * sunrise = 2022-07-17 06:00:00 (default)
controller.js:361   * sunset = 2022-07-17 18:00:00 (default)
angular.js:12416 TypeError: Cannot read properties of null (reading 'format')
    at writeSunriseSunsetToConsole (controller.js:362:71)
    at controller.js:447:5
    at angular.js:14634:28
    at m.$eval (angular.js:15878:16)
    at m.$digest (angular.js:15689:15)
    at m.$apply (angular.js:15986:13)
    at g (angular.js:10511:36)
    at L (angular.js:10683:7)
    at A.onload (angular.js:10624:9)
(anonymous) @ angular.js:12416
(anonymous) @ angular.js:9203
(anonymous) @ angular.js:14642
$eval @ angular.js:15878
$digest @ angular.js:15689
$apply @ angular.js:15986
g @ angular.js:10511
L @ angular.js:10683
A.onload @ angular.js:10624
load (async)
(anonymous) @ angular.js:10607
q @ angular.js:10480
g @ angular.js:10187
(anonymous) @ angular.js:14634
$eval @ angular.js:15878
$digest @ angular.js:15689
$apply @ angular.js:15986
(anonymous) @ angular.js:1658
e @ angular.js:4473
d @ angular.js:1656
yc @ angular.js:1676
Zd @ angular.js:1570
(anonymous) @ angular.js:28683
i @ jquery.min.js:2
fireWith @ jquery.min.js:2
ready @ jquery.min.js:2
J @ jquery.min.js:2
www.thetynans.co.uk/:1 Access to XMLHttpRequest at 'https://ajax.googleapis.com/ajax/libs/jquery/2.2.0/jquery.min.js' from origin 'https://www.thetynans.co.uk' has been blocked by CORS policy: Request header field access-control-allow-origin is not allowed by Access-Control-Allow-Headers in preflight response.
content_script.js:207 Request Error when loading :  https://ajax.googleapis.com/ajax/libs/jquery/2.2.0/jquery.min.js ProgressEvent {isTrusted: true, lengthComputable: false, loaded: 0, total: 0, type: 'error', …}
content_script.js:207          GET https://ajax.googleapis.com/ajax/libs/jquery/2.2.0/jquery.min.js net::ERR_FAILED
(anonymous) @ content_script.js:207
a.get @ content_script.js:207
[... 55 lines cut ...]
G_cp_7 @ content_script.js:208
a.Ea @ content_script.js:205
(anonymous) @ content_script.js:212
d @ content_script.js:210
(anonymous) @ content_script.js:209
(anonymous) @ content_script.js:209
G_cp_7 @ content_script.js:208
b.Jb @ content_script.js:212
(anonymous) @ content_script.js:195
Promise.then (async)
(anonymous) @ content_script.js:195
Promise.then (async)
a @ content_script.js:195
b @ content_script.js:212
(anonymous) @ content_script.js:212
controller.js:247 === Day Time streaming starting at 5:50 pm
controller.js:314   m_now = 2022-07-17 17:50:26
controller.js:317   Times:
controller.js:318   m_now=17:50, m_sunrise=06:00, m_sunset=18:00
controller.js:319   beforeSunriseTime = false
controller.js:320   afterSunsetTime = false
controller.js:340 XXX Not rereading data.json, numImagesRead=1
image.jpg:1          GET https://www.thetynans.co.uk/allsky/image.jpg?_ts=1658076751523 404

[image]

EricClaeys, 2022-07-18:
@skyviewobs  Your web server doesn't appear to be handling .json file, or at least not handling them correctly.

In FTP on the server, rename the file to `data.json.txt` then in a browser go to `https://www.thetynans.co.uk/allsky/data.json.txt`.  If that works, then contact the administrator of your web server.

skyviewobs, 2022-07-18:
Hi Eric,

Thanks for your response, I can confirm the file is read with the .txt extension correctly so you are almost certainly correct it is a server side issue. Looking further it appears as if it might be related to a MIME type for .json files not being enabled on the Webserver so I will need to contact my Hosting provider to resolve, will update thread if get a resolution.

Thanks for your assistance, Brendan.

skyviewobs, 2022-07-18:
Hi Eric,

The issue has been resolved by switching my server operating system from Windows to Linux, all errors now cleared.

Thanks for your help.
Brendan.