AllskyTeam/allsky #1156: Only "Loading" on Allsky Website

AndreHellmuth, 2022-04-13:
I need help please
The current image.jpg is loaded correctly on the server. But I can't see a picture on my "Allsky website". Always only "Loading". What could be the problem?
[image]
[image]

EricClaeys, 2022-04-14:
@AndreHellmuth   Did you change the `imageName` setting in `config.js` to point to "image.jpg"?  If not, see [this Wiki page](https://github.com/thomasjacquin/allsky/wiki/allsky-website-Settings) for a description of the various settings.
If you press F12 then go to the Network link (you may need to click on the `>>` to get there), then refresh your browser, you'll see `image.jpg` in red.  Click on it to see where it's looking for the file.

What is the `image_dir` folder for?

AndreHellmuth, 2022-04-14:
Hi, i change the config.js 
Sorry, what do you mean with "F12"? (Sorry I am German an my English is not the best") 
[image]
[image]

EricClaeys, 2022-04-14:
@AndreHellmuth   Meine Deutsch ist sehr schlect.  Im America ist es sehr _difficult_ (don't know the German word) zu ubrung.

Where is your Allsky Website?  On your Pi, or on a remote server? Your first screenshot looks like an FTP program which implies your website is on a remote server.  If so, you need `imageName="image.jpg"` in the `config.js` file.

Keyboards, at least English ones, have function keys along the top, from F1 to F12.  On my laptop I have to press the "fn" (function) key then one of the function keys at the top.  If I don't press the "fn" key, I can adjust the screen brightness, volume, etc.  Anyhow, pressing F12 brings up "developer mode" in browsers where you can see the code, see all files the web page is requesting, and a bunch of other stuff.  The "Network" page is where you see the files requested. It highlights files that don't exist in red, with a "404" error code which you will often see if you go to a page that doesn't exist.  `image.jpg` should appear in red since I'm 99% sure the web page is requesting it, but can't find it.  If you click on `image.jpg` it will give you the full URL it's trying to access, which can be used to help determine the problem.

Does your keyboard have function keys?  If not you can probably display them via the settings page. On Edge I click on the three "..." in the upper right, then click on "More tools", then "Developer tools".

AndreHellmuth, 2022-04-14:
Hi Eric, many thanks for your help. Some things can be so easy. 
imageName="image.jpg"  was the final step. 

Have a nice Day...
Greets... André

EricClaeys, 2022-04-14:
@AndreHellmuth It sounds like your problem was resolved; if so, would you please close this Issue?

Thanks

closed 2022-04-14 by AndreHellmuth