AllskyTeam/allsky #1163: Website works on local network, but only sits at "Loading" screen when accessing externally.

OrionA51, 2022-04-16:
Everything works almost as it should. At the end of the night, the splash screen that should say, "Camera is off during the day" isn't displayed at the day start. It just sits on the last night image taken. "Camera is off during the day" will display if I restart the service during the day but will not display again unless I restart the service. This is not a big issue, but I would like to fix it. The main issue I have though is not being able to see the live feed on the allsky website unless I am on my local network. I can however see the live feed in the WebUI outside my local network. For reference, I am accessing my local web server outside my network because I opened up port 80 on my gateway and use my public IP to access the WebUI and Website. I would think that If the live feed works in the WebUI when outside my local network, then the allsky website should work the same.

EricClaeys, 2022-04-16:
@OrionA51   I assume you are using the latest website?  If it's more than 2 weeks old, please update it since I change the code that displays that message to (hopefully) fix the issue.
Assuming you ARE using the latest software, please press F12 when the "Camera is off during the day" does not appear when it should, then go to the `Console` link on the right.  You may need to click on the `>>` first.  Then copy/paste the message in the console.

For your second problem, just for clarification, are you talking about the WebUI (the one that lets you change settings), or the Allsky Website that can show archived timelapse, keograms, and startrails?  Unless your Pi is accessible via the Internet, the WebUI is only viewable on your Pi.  Many people put the Allsky Website on a separate server on the Internet, then have the Pi copy files to it.  I'll be able to help with the issue once you clarify this.  Thanks

OrionA51, 2022-04-16:
The first issue I’ll have to get back to you on tomorrow but for the second issue, I can access the web UI and the all sky website from the Internet. I simply opened port 80 on my router and host port 8080 (all sky on my pi). I just use my public IP from any device with an internet connection and I can get into the web UI and change settings. I can also get into the all sky website the same way, but it will not show a live feed. It just sits on the loading screen. The live feed works in the “webui” when connecting externally, but not the “website.” Both the webui and website show a live feed locally. 

Ryan C.

EricClaeys, 2022-04-16:
@OrionA51   Are you willing to share your URL with me?  If you prefer to do it privately, you can email me at AstroEric then the at sign, then a dot then com.
Eric

OrionA51, 2022-04-16:
Yes, but that’s an incomplete email address. 

Ryan C.

EricClaeys, 2022-04-16:
@OrionA51    Duh.  I was trying so hard to be clever with hiding my address from spam/hackers, that I didn't even bother to check that it was correct.    e c c s s w . CoM

OrionA51, 2022-04-16:
Sent. If you want access to the WebUI also, then I’ll have to give you the login obviously.

Ryan C.

OrionA51, 2022-04-16:
Email got returned. Att e c C S s w dot com dot Eric? 

Ryan C.

OrionA51, 2022-04-16:
@EricClaeys Ok I sent it. When I saw the info you gave me it had a different format from what it looks like on the site here.

EricClaeys, 2022-04-17:
Closing Issue.  Problem was resolved via private email.

closed 2022-04-17 by EricClaeys