AllskyTeam/allsky #1164: Constellation Overlay wrong (Time?)

AndreHellmuth, 2022-04-16:
Hi, I miss the sun with the planets. In the evening when the moon comes, the sun can be seen with the planets. ;) The time and coordinates are correct. Overnight recording is fine too. Am I missing something? I didn't find anything in "config.sh".
The cardinal points are also correct according to the picture.

[image]

EricClaeys, 2022-04-16:
@AndreHellmuth. Is your website accessible from the Internet so I can look?
The top and bottom of the picture is cut off so I can't tell if it thinks North is up or down.  You can temporarily shrink the `overlaySize` in the `config.js` file to see.
To make it easier to see things I suggest temporarily turning off the Messier objects (change the word `objects` to `xobjects`) and `showstarlabels`.
If you can post a picture from night that would help.

AndreHellmuth, 2022-04-16:
Hi Eric, this is my website: 
https://www.andres-astroseite.de/allsky/
I hope this night is CS ;)

I changed the options in "config.js". 
My problem is, that the sun comes hours later from the other site.
I think it is a coordinates or time problem? 
Time in PI is correct.

AndreHellmuth, 2022-04-16:
[image]

EricClaeys, 2022-04-16:
@AndreHellmuth   I will check into it later today.

In the meantime, how many allsky cameras do you have?  There are 8 map pins for you, with slightly different locations.  And all of them have an incorrect `Image URL` - you are using an incorrect version of Thomas' image.  Would you please change it in the WebUI to your image?  It should be `https://www.andres-astroseite.de/allsky/image.jpg`.

I have asked Thomas about the 8 pins - assuming your have one Pi, when you update the map data the server should recognize the Pi's ID and update that entry in the database; there appears to be a bug in the server code or you have a LOT of Pi's.

AndreHellmuth, 2022-04-16:
Ups.. i have only 1 Allskycam.

EricClaeys, 2022-04-16:
@AndreHellmuth   
Please remove the `latitude` and `longitude` lines from `virtualsky.json` - they belong in the `config.js` file, not this one.
Then make a copy of `virtualsky.json` and call it `virtualsky.json-ORIGINAL`.  This will allow you to restore your settings after we have fixed the problem.
Now, make the following changes to `virtualsky.json`:
* Change the following settings to **true**: `mouse`, `keyboard`, `showdate`, `constellationlabels` and `showposition`.  This will allow me to use the mouse to rotate the overlay and use the keyboard to check on things.  It will also hide a lot of the text and lines which make it hard to see the actual picture.
* Change the word `objects` to `xobjects` (that will cause the Messier objects to disappear).
* Add a line:  `"showstarlabels": false,`

In `config.js`, please remove the double quotes around the numbers in the `latitude` and longitude` lines.  Double quotes are for strings of text.  Numbers don't need quotes.

Are you POSITIVE your latitude is 52 degrees NORTH and longitude is 10 EAST?  I'm 99.999% that's correct since you are in Germany.

Is North up or down in the picture?

Thanks for making the changes - I'm confident we'll be able to resolve this with the changes above and answers to the questions.

AndreHellmuth, 2022-04-16:
Thanks...
I will do it tomorrow. Now I must work :(

EricClaeys, 2022-04-16:
@AndreHellmuth   I can see the Big and Little Dippers.  Now the fun begins.  It's a pain in the butt to get the constellation overlay correct.  I have spent hours.  See [the Wiki page](https://github.com/thomasjacquin/allsky/wiki/allsky-website-Settings#adjusting-image-size-and-sky-overlay) on adjusting the overlay.  It will still take time, but less time.  If you figure out a quicker way to adjust the overlay, PLEASE let me know.  You will be a hero.

On the image at 22:30 your time, the Big Dipper is just a little below the center.
On the overlay, the Big Dipper is half off the bottom, just right of center.
That implies you need to rotate the overlay count-clockwise (a negative `az`, maybe -80 ?).  You will probably need to then move the overlay up (decrease `overlayOffsetTop`).  Then will almost certainly have to adjust rotation, size, and offset.  And then do it again.  Setting `"mouse":true,` in `virtualsky.json` allows you to use the mouse to rotate the overlay.  That's a quick way to see if rotating will help, and what impact it will have.  Turning off most of the other things in the overlay (per my suggestions above) makes it easier to see the stars underneath.  Right now it's almost impossible to see them under the overlay since the overlay is so cluttered.

FYI
On your images, you can get the text on the upper left to be closer together by decreasing the **Line Height** in the WebUI.  The text is pretty hard to read so you may want to consider increasing the **Font Size**.

AndreHellmuth, 2022-04-17:
Hi Eric, thanks for responding.
I changed the points you're wrote. 
There are three things...
1. Sun comes 8 hours later in the Overlay
2. The Coordinates are wrong 
3. No Time in Overlay. ( I changed all points in "virtual sky.json" )

Yes, you're right. I come from Germany. 
My Coordinates are correct.  52.0 N / 10.08 E

[image]

EricClaeys, 2022-04-17:
@AndreHellmuth I just went to your website and the coordinates in the upper left are correct.  Possibly removing latitude and longitude from `virtualsky.json` fixed it.
The time is not appearing because you didn't make all the changes I requested in my post from 23 hours ago.

Now that the overlay has the correct coordinates you can set `az` back to 0, then start adjusting values as needed.

AndreHellmuth, 2022-04-18:
@EricClaeys I changed all. az is back to "0". 
Must I change the following settings back to false: mouse, keyboard, showdate, constellationlabels and showposition ?

EricClaeys, 2022-04-18:
@AndreHellmuth. That's up to you.  There is no right or wrong with these settings.
Personally, I don't like mouse on except when trying to align the overlay since it makes it hard to zoom in when using a phone.  showdate and showposition aren't very useful except when testing.  constellationlabels is a personal preference.

EricClaeys, 2022-04-18:
@AndreHellmuth Since the problem is resolved, would you please close the Issue?

AndreHellmuth, 2022-04-19:
@EricClaeys, thanks for your help.

closed 2022-04-19 by AndreHellmuth